rwaldron/johnny-five #1090: Add support for winch/multi-turn servo

beriberikix, 2016-04-09:
[Winch servos](http://hitecrcd.com/products/servos/boat-servos/hs-785hb-3.5-turn-winch-servo/product) are basically normal servos with a multi-turn pot. Surprisingly, not many people know about them but they can be super handy for pan-tilt things.

To control them using the standard Servo arduino library, one needs to scale the values passws to the [`write()` call](http://forum.arduino.cc/index.php?topic=120884.msg910323#msg910323). 

I could see it added as a new `type` in the `Servo` component class.

dtex, 2016-04-09:
I think this is a good addition, but I don't think we need a new type. Maybe just a ```servoRange``` parameter... i.e.

````
new Five.Servo({
  pin: 11,
  servoRange: [0, 1260] // 3.5 * 360
});
````

Then just map the servoRange down to [0,180] before calling servoWrite. The size of each step will be pretty big (~5°) on a microcontroller with 8-bit PWM but it is what it is.

beriberikix, 2016-04-09:
Ah, that sounds like a scalable solution!

rwaldron, 2016-04-09:
That model is $117 :( 

http://www.amazon.com/Hitec-RCD-33785S-HS-785HB-Winch/dp/B000BOGI7E

Something about "servoRange" rubs me the wrong way, can we think about this a bit more?

dtex, 2016-04-09:
I love thinking about API's! I wish it were my full time job (seriously)

dtex, 2016-04-10:
Talking out loud...

We already have ```range``` which sets software limits on what we can ask of a 180° servo, but now we are talking about the physical abilities of the servo varying. Having ```range``` and ```servoRange``` could be confusing if a user is only aware of one of the two. Now I kinda' wish ```range``` had been ```limits``` but that horse left the barn long ago.

The primary difference between this and a standard servo is an encoder that is configured to work with multiple rotations and it do not have the physical tabs that limit their range. 

This property would also be useful for 90° and 360° servos.

This feature could also be used to compensate for servos that don't map perfectly to our pre-defined PWM range as long as those required PWM values are both within our default pwmRange... but that's confusing and I should probably never mention it again.

beriberikix, 2016-04-10:
@rwaldron ya, that's 'spensive - when they're in stock they're [half that](https://www.servocity.com/html/hs-785hb_3_5_rotations.html#.Vwmm-xMrKRs) at Servo City. But the real deal is [HobbyKing](http://www.hobbyking.com/hobbyking/store/uh_listCategoriesAndProducts.asp?cwhl=XX&idCategory=1669&v=&sortlist=P&LiPoConfig=&CatSortOrder=asc) @ $9.

dtex, 2017-08-17:
Managed to score one for $14. It has to drop ship from china so I won't have it for a month and a half, but hey we're already 15 months in on this issue so what's a couple more right?

dtex, 2017-09-06:
@rwaldron wrote:
> Something about "servoRange" rubs me the wrong way, can we think about this a bit more?

So the distinction we need to make is between our desired limits (i.e. to keep a robot from literally tripping over its own feet) vs. the capabilities of the servo. I wish ```range``` had been ```limits``` instead but it's too late for that. The options I see are:

* ```endpoints```
* ```throw```
* ```deviceRange```
* ```servoRange```
* ```physicalRange```

BTW, my servo from Hobby King arrived last night.

dtex, 2017-09-10:
I'm going with deviceRange for now since "device" represents the physical component in our device/controller world.

rwaldron, 2017-09-11:
> I wish range had been limits instead but it's too late for that. 
...
> I'm going with deviceRange for now since "device" represents the physical component in our device/controller world.

Sounds good. In the meantime, you might be able to ease into `limits` by making it an alias first and then just update all the docs that use `range` to `limits` and then we can reclaim `range` in some future release?

dtex, 2018-01-11:
Hi @beriberikix ,

I've made pretty good headway on this (I have it working great with Firmata), but there are changes that need to land in the other IO plus-ins so we don't break servo for non-firmata boards like the Tessel, Beaglebone and so on.

Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for "non-180°" servos there.

closed 2018-01-11 by dtex

dtex, 2018-05-10:
@beriberikix,

It only took changes to 9 repos owned by 6 different maintainers and new code in 4 different programming languages, but this is done.

Go build something really cool!

BTW, I'm really looking forward to Mesh. I've got an Argon and six Xenon's on the way. My plan is to use an Argon for high level instructions and a Xenon in each limb of a walking robot. That way the grunt work like Inverse Kinematics can be offloaded to the Xenons. Should be fun.

beriberikix, 2018-05-12:
Wow, heroic effort! 💪 

Glad you're excited about Particle Mesh! Sounds fun :)