rwaldron/johnny-five #1093: Multiple Board objects for single Arduino board (Not Working)

anno143, 2016-04-10:
Hi there!
I am working on my final year project  related to home automation. I adapted [heimcontrol.js](https://github.com/ni-c/heimcontrol.js) as base structure for my project. Heimcontrol.js uses [duino](https://github.com/ecto/duino) to communicate with Arduino. 
Duino is not good enough to meet my requirements so i decided to use johnny-five. 
Now I have developed three plugins for heimcontrol.js. Each plugin consists upon two view (jade) files and one index.js (node.js) file.
In each index.js file i am creating a five.Board() object.
My program works only when i  use one plugin at a time i.e. one five.Board() object.
If I use more than one plugins at a time it gave me following error:

"A timeout occurred while connecting to the Board"

Can I use multiple Board objects for single Arduino connected?

dtex, 2016-04-10:
You only want to connect to the Arduino once. Johnny-Five has a cool, little known feature wherein it keeps a list of boards that have been instantiated. If you instantiate a single board in your "main" file, then require Johnny-Five in a separate file, and then instantiate a device (pin, servo, relay, whatever) without having instantiated a board in that other file, it'll just work. Johnny-Five will look at the list of boards that have been isntantiated and use the first one in the list.

Just make sure you don't instantiate your device in the plug-in file until after the board has been initialized in your main file. For example:

index.js
````js
var Five = require("johnny-five");
var plugOne = require("./plugOne.js");

var board = new Five.Board();

board.on("ready", function() {
  plugOne.init();
});
````

plugOne.js
````js
var Five = require("johnny-five");
var servo;

var plugOne = {
  init: function() {
    // servo will default to the board instantiated in index.js
    servo = new Five.servo(11);
  }
};

module.exports = plugOne;
````

dtex, 2016-04-10:
If you'd like something that's doesn't rely on a magical, undocumented feature (thereby confusing anyone who looks at your code) you could just pass the original board instance into the init function.

mwittig, 2016-04-11:
index.js
```
var Five = require("johnny-five");
var plugOne = require("./plugOne.js");

var board = new Five.Board();

board.on("ready", function() {
  plugOne.init(board);
});
```
plugOne.js

```
var Five = require("johnny-five");
var servo;

var plugOne = {
  init: function(board) {
    // servo will default to the board instantiated in index.js
    servo = new Five.servo({
        pin: 11,
        board: board
    });
  }
};

module.exports = plugOne;
```

anno143, 2016-04-12:
Thank you everyone. I really appreciate your help. 
Mwitting! Your method works.
Again thank you everyone

closed 2016-04-12 by dtex

rwaldron, 2016-04-12:
@mwittig 👏 nice one :)