rwaldron/johnny-five #1094: How to get back value from digitalRead callback ?

SinLok, 2016-04-10:
```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
	var tmpValue;
	this.digitalRead(3, function(value){
		tmpValue = value;
	});
});
```

I try to console.log tmpValue and display undefined.
So, how to get back value from digitalRead callback easily?

Thank you!

rwaldron, 2016-04-10:
Do you mean that you want to do something with the value outside of `digitalRead` callback? It's not clear from your example what you're trying to accomplish. Let's start there and work towards solving your problem: what is it that you'd like do with the program?

SinLok, 2016-04-10:
Yes, I mean I need to get back the value and then do something in the outside.

For example, this is the program I using C language to implement. Now, I want to translate it to JavaScript.
```
while(digitalRead(pin) == LOW)
	if (loopCnt-- == 0) return -2;
```
But I don't know how to get back the value from digitalRead callback and then compare with Low

rwaldron, 2016-04-12:
On the surface, the first problem I see is that most of the C IO paradigms don't translate directly, so you need to change the way you approach the problem you're trying to solve. For example, this is how you would asynchronously handle digital read values: 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  function doSomethingOnLow() {
    console.log("pin 3 went low");
  }

  board.pinMode(3, board.io.MODES.INPUT);
  board.digitalRead(3, function(value) {
    if (value === board.io.LOW) {
      doSomethingOnLow(); 
    }
  });
});
```

I think I could be more helpful if I better understood what exactly you were trying to accomplish, from that knowledge I could then share from my experience.

reconbot, 2016-05-25:
I'm going to close this issue for now, but if you'd like to continue with it feel free to reopen.

closed 2016-05-25 by reconbot