rwaldron/johnny-five #1096: I'm using Meteor with johnny-five.io and they have duplicated class name "collection"

pacozaa, 2016-04-11:
> lib/mixins/collection.js 

is duplicate to 

> Meteor.collection

1. When my code doesn't include `Meteor.collection` involve everything works fine
2. After I include `Meteor.collection` involve it cause this error

[image]

**How to use both awesome things together?**

reconbot, 2016-04-11:
Can you provide some sample code that's causing this issue? Module names usually don't conflict with nodejs, not because they don't have the same name, but because they end up in different scopes.

pacozaa, 2016-04-11:
@reconbot 

There is two files related 
and here is the whole code of each file.

1. model.js
```javascript
Status = new Mongo.Collection('status');
```
    
  which use for get a mongo collection.
2. board.js
```javascript
const five = Meteor.npmRequire('johnny-five');
const board = new five.Board();
```

If I comment all code in model.js, everything works fine!

pacozaa, 2016-04-11:
@reconbot Oh I just use some reservation variable which I assign name `Map` then I change a variable name to `MapPosition` and everything is working right now :+1: 

Thank you

closed 2016-04-11 by pacozaa