rwaldron/johnny-five #1098: Can johnny-five communicate with an arduino board in real time?

iszufu, 2016-04-12:
As I know, johnny-five is based on firmata protocol when connecting with an arduino board. Now, I hope that I can control a board in real time, rather than uploading a sketch after finishing coding. Cause every time I change my mind to control a board I should reconnect the board, which waste a lot of time and experience badly. The most commonly example may use johnny-five to control an arduino smart car. I am not familiar with this framework, can johnny-five achieve this, just like what Snap4Arduino has done?

rwaldron, 2016-04-14:
Thanks for filing. 

> Now, I hope that I can control a board in real time, rather than uploading a sketch after finishing coding. 

The former is indeed how Johnny-Five works.

> can johnny-five achieve this, just like what Snap4Arduino has done?

According to that page, the project uses StandardFirmata, which means that all of the existing controls could be replaced with Johnny-Five.

Please re-open if you have any issues or encounter any bugs along the way!

closed 2016-04-14 by rwaldron

iszufu, 2016-05-13:
Have you ever tried to build a PhoneGap or Cordova app for mobile with johnny-five to control an arduino borad by bluetooth serial? If you do have, please send me the link for toturial. But if you haven't, can you please think about it and give me some solutions to achieve this? I would like to use [blockly](https://developers.google.com/blockly/) from google to build an mobile app for controlling an arduino based car. 
By the way, I know that [cylon.js](https://cylonjs.com/) have done this. But I prefer to use johnny-five, cause I am more familiar with it and I think it should achieve this as well.