rwaldron/johnny-five #1099: Stepper lib

taha, 2016-04-12:
I'm trying to control a set of stepper motors by driving them a single step at a time. Let's say I'm controlling 3 stepper motors simultaneously. Each stepper may require a different motion at the same iteration (possible motions are: 1 step CW, 1 step CCW, or no step). So for example, steppers 1 and 3 may need 1 step CW, but stepper 2 would need 1 step CCW or no step at all. Consider the following sequence of steps for all three motors where `0` means no step, `1` means a CW step, and `-1` means a CCW step.

```
stepper 1 | stepper 2 | stepper 3
----------|-----------|-----------
    1     |     1     |     1
   -1     |     0     |     0
    1     |     1     |     0
    0     |    -1     |     0
    0     |     0     |     1
   ...    |    ...    |    ...
```

I attempted to execute the driving of the sequence above using something very similar to the following. However, the code generated many memory leaks warnings, since I'm not waiting for the event listener to consume the callback event before I initialise another one (which is bound to exceed node's limit since I'm going through a list of 2000 items).

```js
// sequence.length => 2000

sequence.forEach((row) => {
  if(row[0] !== 0) {
    stepper1.step({
      steps: 1,
      direction: row[0] > 0 ? five.Stepper.DIRECTION.CW : five.Stepper.DIRECTION.CCW
    }, () => { /* noop */ });
  }

  if(row[1] !== 0) {
    stepper2.step({
      steps: 1,
      direction: row[1] > 0 ? five.Stepper.DIRECTION.CW : five.Stepper.DIRECTION.CCW
    }, () => { /* noop */ });
  }

  if(row[2] !== 0) {
    stepper3.step({
      steps: 1,
      direction: row[2] > 0 ? five.Stepper.DIRECTION.CW : five.Stepper.DIRECTION.CCW
    }, () => { /* noop */ });
  }
});
```

I did then attempt to use lodash's `_.after` function to make sure all callbacks are executed before I move on to the next sequence row. However, waiting for the callbacks to execute the next stepper motions affected the smoothness of the motors' motions due to the latency generated by the events emitter, so the motors produce a very "jittery/shakey" motion. The callback param is also required by J5's stepper lib. And the events emitter is traced back all the way to firmata's `stepperStep` method.

My goal is to be able to execute the example sequence above without any memory leaks warnings, or jitteriness on the steppers.

taha, 2016-04-12:
### UPDATE 1

I've noticed that the jittery effect happens when I attempt to control more than one stepper motor at a time using the method explained above (the step(1) in a for loop). My guess would be that the jittery effect happens since J5 and firmata sends 1 command at a time, so there will be some latency between step commands for the same motor (since it has to finish executing the commands for the other motors that followed it before reaching the next command for the first motor).

Can someone confirm that this is the issue, and if so, how to compensate for it?

rwaldron, 2016-04-12:
That is likely the cause for some amount of jitter. The Stepper protocol was designed for more broad step instructions. I think there is a strong interest in redesigning control from the protocol up. cc @dtex @soundanalogous

Sorry I'm only just now getting to this, I'm super swamped with Tessel 2 work right now :(

taha, 2016-04-12:
Yeah I agree. I've looked into it more since I've posted this, I think the issue lies in the StepperFirmata from the ConfigurableFirmata library. It basically treats all stepper instances as independent from one another, i.e. the steps are not aggregated (if that's the right word to use) from all the available instances. It does the following for each stepper instance:

```c++
    digitalWrite(dir_pin, direction);
    delayMicroseconds(this->stepDelay);
    digitalWrite(step_pin, LOW);
    delayMicroseconds(this->stepDelay);
    digitalWrite(step_pin, HIGH);
```

When it should be doing something like this (I think):

```c++
    digitalWrite(stepper1_dir_pin, direction);
    digitalWrite(stepper2_dir_pin, direction);
    digitalWrite(stepper3_dir_pin, direction);
    delayMicroseconds(this->stepDelay);
    digitalWrite(stepper1_step_pin, LOW);
    digitalWrite(stepper2_step_pin, LOW);
    digitalWrite(stepper3_step_pin, LOW);
    delayMicroseconds(this->stepDelay);
    digitalWrite(stepper1_step_pin, HIGH);
    digitalWrite(stepper2_step_pin, HIGH);
    digitalWrite(stepper3_step_pin, HIGH);
```

I'll play around with the StepperFirmata to make sure that this in fact the issue.

And no worries! I just thought I'd check I wasn't missing anything :)

soundanalogous, 2016-04-12:
I'm considering switching to [AccelStepper](http://www.airspayce.com/mikem/arduino/AccelStepper/classAccelStepper.html) as the underlying stepper library for Firmata instead of StepperFirmata. Can you try AccelStepper with Arduino and see if that  meets your needs? You'd have to write your program in the Arduino language since AccelStepper won't work with Firmata yet.

soundanalogous, 2016-04-12:
StepperFirmata also doesn't currently support micro stepping, which if enabled would help reduce jitter.

taha, 2016-04-12:
@soundanalogous yeah good shout. I'll give AccelStepper a go. Although I need the JS side of it, since my program is controlled via WebSockets. I think the code in [StepperFirmata.cpp](https://github.com/firmata/ConfigurableFirmata/blob/master/src/StepperFirmata.cpp) is straight forward enough that I can hook up the basic 1 step commands through Firmata. I'm using the EasyDriver, I believe that handles the micro stepping out of the box which makes life much much easier (1600 micro steps instead of 200 for a full rotation).

soundanalogous, 2016-04-12:
correct, EasyDriver handles micro stepping for you.

soundanalogous, 2016-04-12:
If you have ideas for improvement at the Stepper API level, please chime in here: https://github.com/firmata/protocol/issues/42

soundanalogous, 2016-04-12:
Isn't trying to move one step at a time over a serial protocol extremely inefficient? You're limited by the baud rate and establishing accurate timing would be nearly impossible. That's why in Firmata I didn't expose a way to move by a single step from the host application.

taha, 2016-04-12:
You are right regarding the baud rate limitation. That may have contributed to the steppers jitter, but it's also due to the way the StepperFirmata lib handles the commands. For example, I tested the following which didn't result in any jitter.

```c++
digitalWrite(MOTOR1_STEP, HIGH);
digitalWrite(MOTOR2_STEP, HIGH);
digitalWrite(MOTOR3_STEP, HIGH);
delayMicroseconds(1000);

digitalWrite(MOTOR1_STEP, LOW);
digitalWrite(MOTOR2_STEP, LOW);
digitalWrite(MOTOR3_STEP, LOW);
delayMicroseconds(1000);
```
However, the following code produces the same jitter as when sending one step at a time from J5.

```c++
digitalWrite(MOTOR1_STEP, HIGH);
delayMicroseconds(1000);
digitalWrite(MOTOR1_STEP, LOW);
delayMicroseconds(1000);

digitalWrite(MOTOR2_STEP, HIGH);
delayMicroseconds(1000);
digitalWrite(MOTOR2_STEP, LOW);
delayMicroseconds(1000);

digitalWrite(MOTOR3_STEP, HIGH);
delayMicroseconds(1000);
digitalWrite(MOTOR3_STEP, LOW);
delayMicroseconds(1000);
```

taha, 2016-04-12:
I'm in the process of writing a simple firmata stepper handler tailored to my needs for now. I'd basically only control the single steps and the step delay. However, I'm not quite sure what the best solution would be for overcoming the baud rate limitation. Would some sort of command polling work in this case? I'm not familiar SYSEX based commands, so not quite sure how I'd go about implementing such thing (or simply sending the steps queue in chunks to the Arduino).

soundanalogous, 2016-04-12:
What is your use case for single step control from J5, vs sending a specified number of steps? There may be other ways to solve the problem such as adding limits ([see this modification for an example](https://github.com/brinnLabs/Robust-Firmata/blob/master/RobustFirmata/RobustFirmata.ino#L833-L1002))

taha, 2016-04-12:
@soundanalogous The steppers are used for a CNC experiment I'm working on. I have a planner which works out what each stepper needs to do at every loop iteration. That's mainly why I opted in for the single step method, since I have the planner doing all the work. In theory, all I'd need to do is move each stepper to it's allocated command: CW|CCW step or no step at all. But even in the case of no step, it has to align with the other motors that have step commands.

soundanalogous, 2016-04-12:
Why not write a stepper controller in c++ (Arduino) then expose an interface to it that you can control over the web instead of trying to rely on Firmata?

taha, 2016-04-13:
Yeah I guess you're right. I went for Firmata to try and save time for an initial proof of concept prototype, which turned out to be counterproductive for my particular application.

Thanks for your help @soundanalogous!

soundanalogous, 2016-04-13:
Yeah unfortunately until I figure out a good general purpose method for delegating timing and synchronization control, Firmata does not work very well for timing-critical operations. It's better to handle the timing / sync on the MCU and expose a higher level interface to control via an external host application.

dtex, 2016-04-13:
In the future if we do look at incorporating accelstepper or something similar it would be really cool if we could abstract out the accel (and easing?) functions so that they can be used for other things (servo, stepper, DC motor, LED, etc.). I know sketch size is a concern so this is wishful thinking, but if we could offload that to the microcontroller it could be pretty amazing.

soundanalogous, 2016-04-13:
hummm that's an interesting way to think about it. Basically a few easing utilities that are reusable. Probably wouldn't consume much memory unless they end up requiring lookup tables for efficiency

dtex, 2016-04-13:
@scottgonzalez has some awesome optimizations for shrinking easing functions and I don't think you'd have to rely on lookup tables because the update rate doesn't have to be super high. Most devices can only respond so fast (i.e. servos are usually around 0.2 seconds to move 60°) and others can look smooth with update rates as low as 30hz (led.pulse() for example).

dtex, 2017-07-03:
We're making progress on this:

https://github.com/firmata/protocol/pull/90
https://github.com/firmata/ConfigurableFirmata/pull/72
https://github.com/firmata/firmata.js/pull/176

Once these land we can add support in J5

dtex, 2018-08-21:
Let's leave this one open since the new version of the Stepper class is coming soon and we can close it properly then.

dtex, 2019-08-16:
Tracking Stepper progress in #1133. Closing this.

closed 2019-08-16 by dtex