rwaldron/johnny-five #1100: Adding SHT31 sensor support

burgo4, 2016-04-12:
Hi all, thanks for the great framework !!!

It would be great to have SHT31 -D humidity and temperature sensor support. If I know the SHT31 uses 16bit read and write. I was trying to implement controller for that sensor, but unsuccessfully. Any plans for the future ?

rwaldron, 2016-04-12:
Sure, we can work on this!

Do you have a branch with your controller progress?

rwaldron, 2016-04-12:
http://www.amazon.com/Adafruit-Sensiron-Temperature-Humidity-Breakout/dp/B019MGIC3Y ?

rwaldron, 2016-04-12:
A cursory look at this: https://github.com/adafruit/Adafruit_SHT31/blob/master/Adafruit_SHT31.cpp and the datasheet (https://learn.adafruit.com/downloads/pdf/adafruit-sht31-d-temperature-and-humidity-sensor-breakout.pdf) and I think we shouldn't have too much trouble with this.

burgo4, 2016-04-12:
Yep, it is SHT31 (Amazon link). For now I don't have any branch. I was just trying to implement controller locally. Should I create branch ? Btw. I found very simple but nice example wrote in JAVA - [https://raw.githubusercontent.com/ControlEverythingCommunity/SHT31/master/Java/SHT31.java](https://raw.githubusercontent.com/ControlEverythingCommunity/SHT31/master/Java/SHT31.java). Maybe it can be our inspiration.

burgo4, 2016-04-14:
Hey! 
I've found my source code when I was experimenting with the controller for SHT31 -D. Can you check that? I've tried to read values, but unsuccessfully. Do you see there some bugs ? I was inspirated by JAVA source code from https://github.com/ControlEverythingCommunity/SHT31/blob/master/Java/SHT31.java.

My source code you can find here: http://is.gd/oIv5GO

rwaldron, 2016-04-14:
@burgo4 I'm not sure if their are any bugs, but as far as implementing the baseline driver, you nailed it :)

- Drop that into lib/imu.js's Drivers (you probably already did this)
- Add necessary controllers in:
  - thermometer.js (example: https://github.com/rwaldron/johnny-five/blob/master/lib/thermometer.js#L473-L490)
  - hygrometer.js (example: https://github.com/rwaldron/johnny-five/blob/master/lib/hygrometer.js#L8-L25)
  - imu.js (example: https://github.com/rwaldron/johnny-five/blob/master/lib/imu.js#L1731-L1772)

Then you'll need to whip up some examples (take a look at the existing stuff). We also need to find a fritzing component, but if none exists I will just make us a basic one to work with. Ping me if you need anything, or when you're ready for some review! Great work!!

burgo4, 2016-04-15:
@rwaldron Hey man! Thanks for your reaction ;-) I created a new branch [thermo-humidity-meter-SHT31-D](https://github.com/burgo4/johnny-five/tree/thermo-humidity-meter-SHT31-D) and pushed it. I've added the necessary controllers, created some examples. I tried to find a fritzing component but I've found just this image [http://bit.ly/1Sehi7t](http://bit.ly/1Sehi7t), but maybe it will be helpful when you will create a diagram. Can you check that and give me some review ? Thanks

rwaldron, 2016-04-15:
Ok, I can whip up a component for you :)

rwaldron, 2016-04-15:
https://dl.dropboxusercontent.com/u/3531958/sht31-d.fzz

rwaldron, 2016-04-15:
WOWOWOWOW

I just looked at your branch and the implementation is _perfect_. Grab that fritzing, make some examples with it and open a PR!

burgo4, 2016-04-15:
Great !!! 
So I'm going to make some examples and then I will open PR! 
Thanks

rwaldron, 2016-04-15:
We also need to have some tests in test/imu.js, but I don't want that to block you, so please let me know if you need help there

burgo4, 2016-04-15:
I created some tests and add diagram to the solution. PR is open so please check all my staff once again. Thanks a lot for your help :-)

burgo4, 2016-04-16:
Thanks for the great collaboration Rick. Soon I will need a support for the TSL2591 sensor (https://www.adafruit.com/products/1980). What do you think about it ?

rwaldron, 2016-04-16:
@burgo4 you're absolutely welcome to implement a controller for TSL2591 in the `Light` class. Let's open a new issue for that

closed 2016-04-16 by rwaldron