rwaldron/johnny-five #1101: Browserify / Webpack support

bcjordan, 2016-04-12:
# Browserify

Minimal repro repository: https://github.com/bcjordan/johnny-five-browserify

Repro steps:

```
git clone https://github.com/bcjordan/johnny-five-browserify.git
npm install
npm run build
```

Expected: `index.html` has a `window.five` object
Actual: error on build

```
Brians-MacBook-Pro:johnny-five-browserify brian$ npm install && npm run build && open index.html
npm WARN package.json firmata-browserify-test@1.0.0 No description
npm WARN package.json firmata-browserify-test@1.0.0 No repository field.
npm WARN package.json firmata-browserify-test@1.0.0 No README data

> firmata-browserify-test@1.0.0 build /Users/brian/code/johnny-five-browserify
> browserify main.js -o bundle.js

Error: Cannot find module 'mock-firmata' from '/Users/brian/code/johnny-five-browserify/node_modules/johnny-five/lib'
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:46:17
    at process (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:173:43)
    at ondir (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:188:17)
    at load (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:69:43)
    at onex (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:92:31)
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:22:47
    at FSReqWrap.oncomplete (fs.js:95:15)

npm ERR! Darwin 14.5.0
npm ERR! argv "node" "/usr/local/bin/npm" "run" "build"
npm ERR! node v0.12.7
npm ERR! npm  v2.9.1
npm ERR! code ELIFECYCLE
npm ERR! firmata-browserify-test@1.0.0 build: `browserify main.js -o bundle.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the firmata-browserify-test@1.0.0 build script 'browserify main.js -o bundle.js'.
npm ERR! This is most likely a problem with the firmata-browserify-test package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     browserify main.js -o bundle.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls firmata-browserify-test
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/brian/code/johnny-five-browserify/npm-debug.log
```

# Webpack

Minimal repro repository: https://github.com/bcjordan/johnny-five-webpack

Repro steps:

```
git clone https://github.com/bcjordan/johnny-five-webpack.git
npm install
npm run build
```

Expected: `index.html` has a `window.five` object
Actual: error on build (below), error in console: `bundle.js:9885 Uncaught Error: Cannot find module "repl"`

```
Brians-MacBook-Pro:johnny-five-webpack brian$ ni && nrb && open index.html
npm WARN package.json johnny-five-webpack@1.0.0 No description
npm WARN package.json johnny-five-webpack@1.0.0 No README data
|
> fsevents@1.0.11 install /Users/brian/code/johnny-five-webpack/node_modules/webpack/node_modules/watchpack/node_modules/chokidar/node_modules/fsevents
> node-pre-gyp install --fallback-to-build

[fsevents] Success: "/Users/brian/code/johnny-five-webpack/node_modules/webpack/node_modules/watchpack/node_modules/chokidar/node_modules/fsevents/lib/binding/Release/node-v14-darwin-x64/fse.node" is installed via remote
\
> serialport@2.1.0 install /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node" is installed via remote
webpack@1.12.15 node_modules/webpack
├── interpret@0.6.6
├── tapable@0.1.10
├── clone@1.0.2
├── supports-color@3.1.2 (has-flag@1.0.0)
├── async@1.5.2
├── esprima@2.7.2
├── enhanced-resolve@0.9.1 (graceful-fs@4.1.3, memory-fs@0.2.0)
├── loader-utils@0.2.14 (object-assign@4.0.1, big.js@3.1.3, json5@0.5.0, emojis-list@1.0.0)
├── mkdirp@0.5.1 (minimist@0.0.8)
├── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)
├── memory-fs@0.3.0 (errno@0.1.4, readable-stream@2.0.6)
├── webpack-core@0.6.8 (source-list-map@0.1.6, source-map@0.4.4)
├── node-libs-browser@0.5.3 (https-browserify@0.0.0, tty-browserify@0.0.0, constants-browserify@0.0.1, path-browserify@0.0.0, os-browserify@0.1.2, string_decoder@0.10.31, process@0.11.2, punycode@1.4.1, querystring-es3@0.2.1, assert@1.3.0, timers-browserify@1.4.2, domain-browser@1.1.7, events@1.1.0, stream-browserify@1.0.0, util@0.10.3, console-browserify@1.1.0, vm-browserify@0.0.4, readable-stream@1.1.13, http-browserify@1.7.0, url@0.10.3, buffer@3.6.0, browserify-zlib@0.1.4, crypto-browserify@3.2.8)
├── uglify-js@2.6.2 (uglify-to-browserify@1.0.2, async@0.2.10, source-map@0.5.3, yargs@3.10.0)
└── watchpack@0.2.9 (graceful-fs@4.1.3, async@0.9.2, chokidar@1.4.3)

johnny-five@0.9.39 node_modules/johnny-five
├── lodash.debounce@4.0.5
├── ease-component@1.0.0
├── color-convert@1.2.2
├── browser-serialport@2.0.3
├── temporal@0.5.0
├── nanotimer@0.3.10
├── lodash.clonedeep@4.3.2 (lodash._baseclone@4.5.5)
├── chalk@1.1.3 (supports-color@2.0.0, escape-string-regexp@1.0.5, ansi-styles@2.2.1, strip-ansi@3.0.1, has-ansi@2.0.0)
├── firmata@0.11.3 (es6-shim@0.33.13)
├── es6-shim@0.35.0
└── serialport@2.1.0 (bindings@1.2.1, sf@0.1.7, async@0.9.0, optimist@0.6.1, nan@2.2.1, debug@2.2.0, object.assign@4.0.3)

> johnny-five-webpack@1.0.0 build /Users/brian/code/johnny-five-webpack
> webpack main.js bundle.js

Hash: a99ec70128630264c2ca
Version: webpack 1.12.15
Time: 1623ms
    Asset     Size  Chunks             Chunk Names
bundle.js  1.28 MB       0  [emitted]  main
   [0] ./main.js 38 bytes {0} [built]
    + 137 hidden modules

WARNING in ./~/johnny-five/~/serialport/~/bindings/bindings.js
Critical dependencies:
76:22-40 the request of a dependency is an expression
76:43-53 the request of a dependency is an expression
 @ ./~/johnny-five/~/serialport/~/bindings/bindings.js 76:22-40 76:43-53

WARNING in ./~/johnny-five/~/serialport/~/bindings/package.json
Module parse failed: /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/package.json Line 2: Unexpected token :
You may need an appropriate loader to handle this file type.
| {
|   "name": "bindings",
|   "description": "Helper module for loading your native module's .node file",
|   "keywords": [
 @ ./~/johnny-five/~/serialport/~/bindings ^\.\/.*$

WARNING in ./~/johnny-five/~/serialport/~/bindings/README.md
Module parse failed: /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/README.md Line 2: Unexpected token ===
You may need an appropriate loader to handle this file type.
| node-bindings
| =============
| ### Helper module for loading your native module's .node file
|
 @ ./~/johnny-five/~/serialport/~/bindings ^\.\/.*$

ERROR in ./~/johnny-five/lib/board.js
Module not found: Error: Cannot resolve module 'mock-firmata' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/lib
 @ ./~/johnny-five/lib/board.js 75:19-42 145:6-29

ERROR in ./~/johnny-five/lib/repl.js
Module not found: Error: Cannot resolve module 'repl' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/lib
 @ ./~/johnny-five/lib/repl.js 1:11-26

ERROR in ./~/johnny-five/~/serialport/serialport.js
Module not found: Error: Cannot resolve module 'fs' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport
 @ ./~/johnny-five/~/serialport/serialport.js 22:9-22

ERROR in ./~/johnny-five/~/serialport/lib/list-unix.js
Module not found: Error: Cannot resolve module 'fs' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/lib
 @ ./~/johnny-five/~/serialport/lib/list-unix.js 4:9-22

ERROR in ./~/johnny-five/~/serialport/lib/list-unix.js
Module not found: Error: Cannot resolve module 'child_process' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/lib
 @ ./~/johnny-five/~/serialport/lib/list-unix.js 3:11-35

ERROR in ./~/johnny-five/~/serialport/~/bindings/bindings.js
Module not found: Error: Cannot resolve module 'fs' in /Users/brian/code/johnny-five-webpack/node_modules/johnny-five/node_modules/serialport/node_modules/bindings
 @ ./~/johnny-five/~/serialport/~/bindings/bindings.js 6:9-22
Brians-MacBook-Pro:johnny-five-webpack brian$
```

rwaldron, 2016-04-13:
Thanks for the report. 

> Browserify seems to be mostly working, just with the exception of missing mock-serialport—if users add that to devDependencies, it requires OK. Fix might just be to add mock-serialport to optionalDependencies, though I'm not sure what implications of that change might be.

I think you meant `mock-firmata`—I just pushed a refactoring of the tests such that the source never needs to require mock-firmata directly.

rwaldron, 2016-04-13:
All of the webpack errors are related to webpack settings, aren't they? It looks like webpack is trying to include things that it either shouldn't or that need special instructions. Someone that has better understanding of webpack should advise

bcjordan, 2016-04-18:
Updated johnny-five-browserify repo to use `v0.9.41`, I think browserify might still be choking on this remaining `require("mock-firmata")` https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L143 — repro steps:

git clone https://github.com/bcjordan/johnny-five-browserify.git
npm install
npm run build

Error:

```
Brians-MacBook-Pro:johnny-five-browserify brian$ npm run build

> firmata-browserify-test@1.0.0 build /Users/brian/code/johnny-five-browserify
> browserify main.js -o bundle.js

Error: Cannot find module 'mock-firmata' from '/Users/brian/code/johnny-five-browserify/node_modules/johnny-five/lib'
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:46:17
    at process (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:173:43)
    at ondir (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:188:17)
    at load (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:69:43)
    at onex (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:92:31)
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:22:47
    at FSReqWrap.oncomplete (fs.js:95:15)

npm ERR! Darwin 14.5.0
npm ERR! argv "node" "/usr/local/bin/npm" "run" "build"
npm ERR! node v0.12.7
npm ERR! npm  v2.9.1
npm ERR! code ELIFECYCLE
npm ERR! firmata-browserify-test@1.0.0 build: `browserify main.js -o bundle.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the firmata-browserify-test@1.0.0 build script 'browserify main.js -o bundle.js'.
npm ERR! This is most likely a problem with the firmata-browserify-test package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     browserify main.js -o bundle.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls firmata-browserify-test
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/brian/code/johnny-five-browserify/npm-debug.log
Brians-MacBook-Pro:johnny-five-browserify brian$
```

henricavalcante, 2016-04-22:
https://github.com/rwaldron/johnny-five/commit/624dc7af1d3f9fbbef95dfe76d8b1b8dd75fcb3d

are you using latest push of johnny-five?

brodo, 2016-04-22:
I just ran into this issue. I'm using the latest repository head:
```bash
johnny-five@0.9.43 (git://github.com/rwaldron/johnny-five.git#dd8b1694817e9aa8818c6b2a97c2a5d8d622661f)
```

The errors I get are:

```
WARNING in ./~/bindings/bindings.js
Critical dependencies:
76:22-40 the request of a dependency is an expression
76:43-53 the request of a dependency is an expression
 @ ./~/bindings/bindings.js 76:22-40 76:43-53

WARNING in ./~/bindings/README.md
Module parse failed: /Users/brodo/Development/remotino/node_modules/bindings/README.md Unexpected token (2:3)
You may need an appropriate loader to handle this file type.
SyntaxError: Unexpected token (2:3)
    at Parser.pp.raise (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:920:13)
    at Parser.pp.unexpected (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:1483:8)
    at Parser.pp.parseExprAtom (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:330:12)
    at Parser.pp.parseExprSubscripts (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:225:19)
    at Parser.pp.parseMaybeUnary (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:204:17)
    at Parser.pp.parseExprOp (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:171:41)
    at Parser.pp.parseExprOp (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:173:19)
    at Parser.pp.parseExprOps (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:153:15)
    at Parser.pp.parseMaybeConditional (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:133:19)
    at Parser.pp.parseMaybeAssign (/Users/brodo/Development/remotino/node_modules/webpack/node_modules/acorn/dist/acorn.js:110:19)
 @ ./~/bindings ^\.\/.*$

ERROR in ./~/johnny-five/lib/board.js
Module not found: Error: Cannot resolve module 'mock-firmata' in /Users/brodo/Development/remotino/node_modules/johnny-five/lib
 @ ./~/johnny-five/lib/board.js 143:6-29
```

rwaldron, 2016-04-22:
Sorry, I completely forgot to get rid of that from the source :\

Just waiting for a commit on master to go green then I'll do a release with that removed.

rwaldron, 2016-04-22:
Ok, please try johnny-five@0.9.44

brodo, 2016-04-22:
Thanks a lot, that fixes the error for me!

rwaldron, 2016-05-04:
@bcjordan have you tried this yet?

bcjordan, 2016-05-04:
Excellent, Browserify works too! Closing out. Thanks @rwaldron!

rwaldron, 2016-05-04:
Awesome! Thanks for the patience and great reports everyone!!

closed 2016-05-04 by rwaldron

akrueger, 2016-08-30:
It wasn't quite clear to me what the resolution was on this. Did you end up externally loading bindings.js outside of the bundle? I have managed to calm webpack down, but not to successfully load bindings.js as a dependency for other modules inside the bundle.

Twistedst, 2017-02-10:
To fix the fs error from being thrown as well I had to add "fs" to the externals array in the Webpack config. I also needed to put child_process in the IgnorePlugin instead of mock_firma.
`module.exports = {
  plugins: [
    new webpack.IgnorePlugin(/^child_process$/),
    new webpack.ContextReplacementPlugin(/bindings$/, /^$/)
  ],
  externals: ["bindings", "fs"]
}`

kslhunter, 2017-04-17:
**Add `NormalModuleReplacementPlugin` to `webpack.config.js`:**
```javascript
plugins: [
    new webpack.NormalModuleReplacementPlugin(
        /^bindings$/,
        require.resolve("./bindings")
    )
]
```

**Create `binding.js`:**
```javascript
module.exports = exports = function (str) {
    if(str === "binding"){
        let result = require("../node_modules/ref/build/Release/binding.node");
        result.path = "../node_modules/ref/build/Release/binding.node";
        return result;
    }
    else if(str === "ffi_bindings.node"){
        let result = require("../node_modules/ffi/build/Release/ffi_bindings.node");
        result.path = "../node_modules/ffi/build/Release/ffi_bindings.node";
        return result;

    }
};
```

**Install `node_loader`:**
```
npm install node-loader
```

**Configure `rules -> node_loader` on `webpack.config.js`:**
```javascript
rules: [
    {
        test: /\.node$/,
        use: 'node-loader'
    }
]
```

**Configure `extensions -> .node` on `webpack.config.js`:**
```javascript
extensions: ['.ts', '.js', ..., '.node']
```

sylhero, 2017-08-15:
@kslhunter do u still need
`module.exports = {
  plugins: [
    new webpack.IgnorePlugin(/^mock-firmata$/),
    new webpack.ContextReplacementPlugin(/bindings$/, /^$/)
  ],
  externals: ["bindings"]
}`
I did what you posted above but I got another error:
`Uncaught TypeError: __webpack_require__(...) is not a function
    at eval (ref.js:5)`
do u have an idea?

kslhunter, 2018-07-02:
@sylhero I do not need it. I solved using my solution only.

simonbuchan, 2018-10-04:
If anyone's interested, I've got a very early version of a `bindings-loader` I'm using: https://github.com/simonbuchan/bindings-loader

Let me know if it helps or there are any changes that would help!