rwaldron/johnny-five #1102: IMU class with MPU6050 is not working

pacozaa, 2016-04-13:
- I have a problem using MPU6050 with johnny-five current version. I tested MPU6050 with johnny-five older version and it worked. 
I tried this 3 examples [accelerometer-mpu6050](http://johnny-five.io/examples/accelerometer-mpu6050/),[gyro-mpu6050](http://johnny-five.io/examples/gyro-mpu6050/),[imu-mpu6050](http://johnny-five.io/examples/imu-mpu6050/) and they not fire any event out except accelerometer-mpu6050 but any value printed is 0 and it only fire once.

- I use this [Arduino I2CScanner](https://gist.githubusercontent.com/kinathru/e2609f70a430e52ec3de/raw/778c7bd0b4a7e94ef3a1d40a0929136dc30daeac/scanner.c) code which print 0x68 and this [Arduino MPU6050](https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_raw/MPU6050_raw.ino) code which print things out.
I pretty sure about my wiring because I use hmc5883l and bmp085 which they work well with johnny-five or Arduino code in the same bus and wiring.

rwaldron, 2016-04-15:
What are you running this code with, Arduino or something else?

pacozaa, 2016-04-15:
@rwaldron I run the code with johnny-five with nodejs on my laptop and it's not work then I test and recheck my wiring by using Arduino IDE and its code to confirm that the hardware still ok and identify the problem.

rwaldron, 2016-04-15:
Yeah, I'm using the same setup and it's working correctly. Can you post a picture of your hardware setup?

pacozaa, 2016-04-22:
1. This is my exact code I'm using

``` javascript
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var imu = new five.IMU({
    controller: "MPU6050"
  });
  imu.on("change", function() {
      console.log("temperature");
      console.log("  celsius      : ", this.temperature.celsius);
      console.log("  fahrenheit   : ", this.temperature.fahrenheit);
      console.log("  kelvin       : ", this.temperature.kelvin);
      console.log("--------------------------------------");

      console.log("accelerometer");
      console.log("  x            : ", this.accelerometer.x);
      console.log("  y            : ", this.accelerometer.y);
      console.log("  z            : ", this.accelerometer.z);
      console.log("  pitch        : ", this.accelerometer.pitch);
      console.log("  roll         : ", this.accelerometer.roll);
      console.log("  acceleration : ", this.accelerometer.acceleration);
      console.log("  inclination  : ", this.accelerometer.inclination);
      console.log("  orientation  : ", this.accelerometer.orientation);
      console.log("--------------------------------------");

      console.log("gyro");
      console.log("  x            : ", this.gyro.x);
      console.log("  y            : ", this.gyro.y);
      console.log("  z            : ", this.gyro.z);
      console.log("  pitch        : ", this.gyro.pitch);
      console.log("  roll         : ", this.gyro.roll);
      console.log("  yaw          : ", this.gyro.yaw);
      console.log("  rate         : ", this.gyro.rate);
      console.log("  isCalibrated : ", this.gyro.isCalibrated);
      console.log("--------------------------------------");
  });

});

```

2. My node.js version is 4.4.2

3.johnny-five version is 0.9.43

4. My exact hardware is [GY88](http://www.hotmcu.com/imu-10dof-mpu6050hmc5883lbmp085-p-188.html) and this is [schematic](http://g04.s.alicdn.com/kf/HT1fdGMFPRcXXagOFbXr/202407509/HT1fdGMFPRcXXagOFbXr.jpg)

5. My setup picture here.
[image]
[image]
[image]

6. As No.5 my picture don't useful but I think my setup is ok because i can use compass and barometer which use the same bus according to schematic in No.4

rwaldron, 2016-05-04:
I'm still waiting on the GY-88

pacozaa, 2016-05-06:
Than you

rwaldron, 2016-05-06:
Still waiting :(

reconbot, 2016-08-26:
I'm going to close this due to age. If you're still having this issue or if you figured it out let us know and we'll reopen. Hopefully the test hardware ever showed up. Sorry someone couldn't get back to you sooner.

closed 2016-08-26 by reconbot