rwaldron/johnny-five #1106: LED Matrix HT16K33 + Leonardo Clone Issues

den2k, 2016-04-17:
I am having strange errors on a Leo clone + HT16K33 LED 8x8 matrix

The same HT1633 LED 8x8 matrix + J5 code + Uno is working.

But HT1633 LED 8x8 matrix + J5 code + Leo clone (ie;adafruit feather) i not working and give the following error:

C:\Users\den2k\OneDrive\Professional\My Projects\Node.js>node ledmatrix.js
1460916747545 Device(s) COM6
1460916747561 Connected COM6
Writing to COM port (GetOverlappedResult): Unknown error code 121
Writing to COM port (GetOverlappedResult): Unknown error code 121
1460916757568 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:146
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (C:\Users\den2k\OneDrive\Professional\My Projects\Node.js\node_modules\johnny-five\lib\board.js:630:8)
    at Board.(anonymous function) [as error] (C:\Users\den2k\OneDrive\Professional\My Projects\Node.js\node_modules\johnny-five\lib\board.js:641:14)
    at Board.<anonymous> (C:\Users\den2k\OneDrive\Professional\My Projects\Node.js\node_modules\johnny-five\lib\board.js:418:14)
    at Timer.listOnTimeout (timers.js:92:15)

rwaldron, 2016-04-18:
This looks like a failure at node-serialport level. Let's ask @reconbot for help on this one

rwaldron, 2016-04-18:
In the meantime I will order one of these: https://www.adafruit.com/product/2771

den2k, 2016-05-01:
Thanks Rick.  I had some time to look at this again. 

Is this an improvement?  I did an update of serialport. Running serialport 2.15.1

Now running on an LEO clone the LED - Matrix HT16K33 EXAMPLE I get this.

C:\Users\den2k\OneDrive\Professional\My Projects\Node.js>node ledmatrix.js
1462138923284 Connected COM9
1462138928329 Repl Initialized
>> events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: Writing to COM port (GetOverlappedResult): Unknown error code 121
    at Error (native)

Running the same example code on my Arduino Uno from the same computer works fine, so I wonder why?  Hmmm. Thanks.

den2k, 2016-05-02:
I disconnected the radio that was attached to the SDA/SCL lines and the example started to work. The RFM69 radio on the Leo clone (like the Feather but not exactly) is using the SDA/SCL Pin 2 and Pin 3 and it doesn't like sharing on the I2C bus. Not sure but that is what I think is the problem.  Perhaps when you get a Feather you can verify it works.

rwaldron, 2016-05-04:
Does the Feather require a battery? Mine keeps shutting off, even when USB tethered

den2k, 2016-05-08:
No battery required. the feather should work when tethered to USB. I'm not using a feather exactly, but a 32u4 based board like it and the board works fine without a battery.

The radio on the I2C pins on my board didn't like sharing so once I disconnected the radio and the LED matrix started to work. The feather should work with the HT16K33 as well, even without the battery.

By the way, J5 is awesome!

rwaldron, 2016-05-08:
> By the way, J5 is awesome

❤️

Can you link me to the exact board you're using? I would love to learn more

den2k, 2016-05-10:
It's my own design. I've built it to have a radio option, but its a pretty simple ATMega32u4 based Arduino clone. I like it that with a simple StandardFirmataPlus it works fine with J5. Be happy to send you an early sample. It's got one more iteration to fix the problem I mention here on this thread but otherwise seem to work fine with J5, although I haven't tested everything.

reconbot, 2016-05-25:
This is a known Node serialport issue, we don't have a fix quite yet. https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/781

dtex, 2018-01-05:
There has been a lot of movement over at node-serialport. J5 now uses the newer version which might have a fix. @den2k can you advise wether it's okay to close this issue or if you are still having problems.