rwaldron/johnny-five #1107: New proposed example: Tilt switch SW-200D

sayanee, 2016-04-18:
Would it be possible to include a new example on **Tilt switch SW 200D** under the section [Button / Switch](http://johnny-five.io/examples/#button--switch) ? 

If yes, I'm happy to send a pull request with the Fritzing diagram, code, edits to the example page, etc.

It works almost the same as a [push button](http://johnny-five.io/examples/button/) and Arduino. Hence, showing the wiring and slight edits to the code might help another beginner.  

😃

rwaldron, 2016-04-18:
Yes please!

sayanee, 2016-04-23:
Awaiting to add new custom part to Fritzing parts library: https://github.com/fritzing/fritzing-parts/pull/46

rwaldron, 2016-04-25:
Not sure I follow? Did you make that part? You can make the example and submit here without it being part of the Fritzing library. Why you commit the .fzz for the example, it will include the part as well.

sayanee, 2016-04-26:
Thanks @rwaldron - ok I will send the pull request with the part file `*.fzz`. Yes, it is a new part.

sayanee, 2016-04-26:
Please use the pull request #1111. Thanks!

rwaldron, 2016-04-26:
Landed and deployed—thanks again! http://johnny-five.io/examples/switch-tilt-SW_200D/

closed 2016-04-26 by rwaldron