rwaldron/johnny-five #1112: Using J5 and Modbus protocol?

aabm00, 2016-04-26:
Hi.

I have an Arduino Yun taking data from 10 pressure sensors through analog pins every 30 seconds , and now I want to read motor consumption but the device uses the modbus protocol to transmit the information.

I am using now J5 and sockets to transmit the sensors information through node-serialport . 

I have read that to use the modbus I have to use a rs485 port so we have to add the IC MAX485. 
There are some implementations of the modbus for node (in npm) that uses node-serialport. 

So, Can I read simultaniously data from the sensors using J5 and the data consumption using modbus?

Thanks

jesus-acuna, 2016-07-14:
No, you can't. You can't run J5 and Modbus protocol at the same time. At least not with the available libraries. You would have to work the reading of the analog values with a simple Arduino Sketch, then use a Master Modbus sketch to read the data from the slave (the motor in your case)

rwaldron, 2016-07-15:
> You can't run J5 and Modbus protocol at the same time. 

Sure you can, but the modbus protocol would be handled by some other program logic. Here's a hypothetical illustration (using https://www.npmjs.com/package/modbus-serial): 

```
npm install modbus-serial
```

```js
var Modbus = require("modbus-serial");
var five = require("johnny-five");
var board = new five.Board("/dev/ttyUSB0");

board.on("ready", function() {

  var pressureSensors = five.Sensor.Collection({
    pins: ["A0", "A1", "A2"] // ...and so on...
    period: 30000 // 30 second periods
  });

  pressureSensors.on("data", function(sensors) {
    var data = sensors.map(function(sensor) {
      return sensor.value;
    });

    console.log(data);
  });

  var modbus = new Modbus();

  modbus.connectRTU("/dev/ttyUSB1", {baudrate: 9600}, function() {
    // modbus.readHoldingRegisters
    // modbus.readInputRegisters
    // See docs: https://www.npmjs.com/package/modbus-serial
  });
});
```

dtex, 2018-01-05:
This issue appears to have been answered so I'm going to close it. Feel free to re-open if necessary.

closed 2018-01-05 by dtex