rwaldron/johnny-five #1113: array-includes-shim.js considered harmful

mwittig, 2016-04-27:
This shim adds the "includes" method to the Array.prototype (used if Array.includes is undefined) as an enumeratable property. I think this is bad behaviour as sloppy code in other packages might be affected. For example, if one enumerates keys of an array (not a good idea, I know, but a common mistake) and forgets to check for hasOwnProperty() this may be troublesome. Thus, I think it is better to introduce the method as a non-enumerateble property (or use a third-party shim like [array-icnludes](https://www.npmjs.com/package/array-includes)  if you prefer). I'll come up with a PR shortly.

rwaldron, 2016-04-27:
That was admittedly a sloppy, hacked-in-spite, response to the size of the `array-incudes` module. 

>  Thus, I think it is better to introduce the method as a non-enumerateble property 

I agree. 

> (or use a third-party shim like array-includes if you prefer).

Naw, getting away from this was intentional, but you're very right to call out the sloppy-ness of my replacement.

rwaldron, 2016-04-27:
Landed! https://github.com/rwaldron/johnny-five/commit/4e1379778ef7ab1b51b71beaf16af9b9ee18ad9e

mwittig, 2016-04-28:
Thanks for looking into this so quickly

closed 2016-05-03 by mwittig