rwaldron/johnny-five #1115: How To Stop HTU21D Read

reza70, 2016-04-30:
I have a problem using HTU21D .
I use Multi API for read HTU21D .
this is my code  :

`var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  

var multi = new five.Multi({
    controller: "HTU21D"
  });

if($incoming == 1)
{
  multi.on("change", function() {
    console.log("temperature");
    console.log("  celsius           : ", this.temperature.celsius);
    console.log("  fahrenheit        : ", this.temperature.fahrenheit);
    console.log("  kelvin            : ", this.temperature.kelvin);

    console.log("Hygrometer");
    console.log("  relative humidity : ", this.hygrometer.relativeHumidity);

  
  });
}
else
{
  multi.stop();
}

});`

I want disconnect and stop HTU21D and destroy multi variable .
Can in do this ?
thank you

rwaldron, 2016-04-30:
I've been planning for and am currently working on the lower level pieces that are necessary for supporting this capability. I should have something concrete to try out by mid week. 

In the meantime, it would be helpful to me to understand what exactly you're trying to accomplish? In the above program, it's unclear where the `$incoming` variable comes from, or why a non-truthy value would result in "stopping" the sensor. Once this code runs, nothing else will happen. The Board instance will keep its connection open until the program itself is closed.

reza70, 2016-04-30:
Thank you .
I want read data from HTU21D and then close traffic tx  and then run Multi again other time .
I want this run every 1 minute .
Can i disconnect board ?

reza70, 2016-04-30:
really i want use HTU21D and DS18B20 sensor together .
When use Multi and them use Thermometer in code i get error about 1-wire sensor .
so i want create Multi object and read temp and close that and then Create Thermometer object and read data and close that .
i want this process repeat every 1 minute .

rwaldron, 2016-04-30:
What is the error you getting?

reza70, 2016-04-30:
Error: 1-Wire device search timeout - are you running ConfigurableFirmata?
    at Timeout._onTimeout (/usr/lib/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1376:14)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5)

i use ConfigurableFirmata on arduino 
i think conflict address or any things .....

if comment five.Multi , all thing is ok .

i need destroy object and stop communicate between arduino and sensor and call five.Multi or five.Thermometer again

rwaldron, 2016-04-30:
No, that's not the right way to fix this—there should be no conflict at all between the two components, as they're using entirely different protocols. If you see that error it's because the board doesn't have ConfigurableFirmata uploaded, or it does and the 1-wire feature is not enabled. Please make sure that 1-wire is enabled

rwaldron, 2016-04-30:
@briangenisio do you know why 1-Wire would timeout, if the board has ConfigurableFirmata running?

reza70, 2016-04-30:
i sure upload ConfigurableFirmata on arduono and enable 1 wire 
because when use only DS18B20 all thing work and no problem but when use DS18B20 and HTU21D together stop script and show me error

rwaldron, 2016-04-30:
I'm going to try and reproduce, I just need to find my DS18B20 :\

reza70, 2016-05-01:
OK
I found my problem but why ..... ?
I use 2 DS18B20 and connect to same digital port and use address for read data .
if use Multi for HTU21D and Thermometer for DS18B20 on this mode i get error .
but if use only DS18B20 and HTU21D i don't have problem .

while 1-wire function call and Multi in running i get error .

BrianGenisio, 2016-05-01:
IIRC, we were getting that error when we had two MAX31850K units (very similar to the DS18B20) connected simultaneously.  The chip would error out, and the read command would time out, causing the error suggesting that OneWire wasn't installed.  But it was, and everything worked fine when there was one MAX31850K unit. 

I was never able to reproduce the issue, because I only ever had one unit.  But it is what caused this exit condition on that device:
https://github.com/rwaldron/johnny-five/blob/9e06037c1004adaf8c5ec59f597cb709d47b9977/lib/thermometer.js#L102

Not sure if that will help, but perhaps it is something similar?

Just looked at the HTU21D (reminded muself).  That's I2C, not one-wire.  Is it possible that I2C is causing a timing issue with OneWire?  Isn't OneWire timing-based?  Does this error happen immediately?  Or does it happen after a while?  Are the read requests happening at the same time?  I wonder if you can read them in serial... toggle between reading the DS18B20 and HTU21D?  Just a thought.

Hmpfh.  :shrug:

I'll see if I can find my HTU21D but it might be a day or two.

rwaldron, 2016-05-05:
I can reproduce this issue as well. 

While I think these _should_ be able to function at the same time, I suspect there is lower level issue that's unrelated to Johnny-Five specifically. I'm going to try to prove/disprove that later today. In the meantime, I've put together a solution that meets the request in your comment here: 

> I want read data from HTU21D and then close traffic tx and then run Multi again other time .
> I want this run every 1 minute .

1. Set up a project directory: 
  ```
  mkdir temperature-process && cd temperature-process;
  npm init -y && npm install johnny-five && touch index.js && touch child.js
  ```

2. Fill in the two files: 
  - index.js
    ```js
    var cp = require("child_process");
    var avg = null;
    var mode = 0;
    var interval = 1000; // Change to 60000 for a minute

    function read() {
      cp.exec("node child " + mode, function(error, response) {
        response = +response;

        if (avg === null) {
          avg = response;
        }

        avg = (avg + response) / 2;

        console.log("%d: %d", mode, response);
        console.log("Avg. %d°C", avg.toFixed(2));

        if (++mode === 2) {
          mode = 0;
        }

        setTimeout(read, interval);
      });
    }

    read();
    ```

  - child.js
    ```js
    var five = require("johnny-five");
    var board = new five.Board({
      repl: false,
      debug: false,
    });
    var mode = process.argv[2];
    var configs = [
      { controller: "DS18B20", pin: 2 },
      { controller: "HTU21D" },
    ];

    if (mode === undefined) {
      mode = 0;
    }

    board.on("ready", function() {
      var temp = new five.Thermometer(configs[mode]);
      temp.on("data", function() {
        process.stdout.write(String(this.celsius));
        process.exit();
      });
    });
    ```

3. Run it!
  ```
  $ node index.js
  0: 22.1875
  Avg. 22.19°C
  1: 22.11
  Avg. 22.15°C
  0: 22.1875
  Avg. 22.17°C
  1: 22.35
  Avg. 22.26°C
  0: 22.25
  Avg. 22.25°C
  1: 22.38
  Avg. 22.32°C
  0: 22.1875
  Avg. 22.25°C
  1: 22.35
  Avg. 22.3°C
  ```

rwaldron, 2016-05-05:
I just tried reading both with just an Arduino sketch and it works fine. Maybe @soundanalogous can help us understand the timing constraints (if any exist)

jackrosenhauer, 2016-11-26:
Any progress on this? I just ran into this issue as well.

soundanalogous, 2016-11-26:
One possibility is, depending on which board you are using, there is not enough memory remaining for stable operation once you enable both OneWire and I2C in ConfigurableFirmata. What board are you using and how much flash memory and SRAM is remaining after compile the firmware (ConfigurableFirmata) for that board? This will be printed in the bottom of the Arduino IDE.

jackrosenhauer, 2016-11-26:
Sorry deleted my comment because I thought it was off topic but after a second look I'm using the same setup as @reza70. My board is a MEGA 2560 and I'm only using 1 DS18B20 and an HTU21D-F. The output after uploading: 
Sketch uses 26,658 bytes (10%) of program storage space. Maximum is 253,952 bytes.
Global variables use 3,072 bytes (37%) of dynamic memory, leaving 5,120 bytes for local variables. Maximum is 8,192 bytes.

soundanalogous, 2016-11-26:
Ok memory is definitely not an issue then. Have you tried attaching the DS18B20 to different pins?

soundanalogous, 2016-11-26:
Also try changing the sampling interval. It's 19ms by default. Try setting it to 100 or even 1000ms:

```js
var board = new five.Board({
  samplingInterval: 100,
});
```

That will change the rate at which the I2C devices will report their values if continuous reporting is enabled. It could be that the default of 19ms is too frequent when attempting to read from multiple devices.

jackrosenhauer, 2016-11-26:
Mega uses pins 20 and 21 for i2c for the HTU21D and the DS18B20 is hooked up to pin 11. I tried pin 2-12 and did not seem to make a difference.

Here is my stack trace (pretty much the same as above):

events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: 1-Wire device read timeout - are you running ConfigurableFirmata?
    at Timeout._onTimeout (C:\git\raspberrypi\node_modules\firmata\lib\firmata.js:1460:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

jackrosenhauer, 2016-11-26:
Setting the samplingInterval to 1s yields the same results.

soundanalogous, 2016-11-26:
And if you use only the HTU21D or only the DS18B20 everything works without issue, correct?

jackrosenhauer, 2016-11-26:
Yes, using only one of the sensors works without issue.

jackrosenhauer, 2016-11-26:
Also, setting samplingInterval to 1000ms does not seem to change the rate at which the values are being read.

soundanalogous, 2016-11-26:
That's pointing to an issue in the default polling rate set in johnny-five then.

soundanalogous, 2016-11-26:
I'd need to add logging in ConfigurableFirmata and report it back over Firmata sendString in order to gain an understanding of what is happening at the firmware level. I don't have these sensors so I'd have to walk you through the process of setting up debug messages.

jackrosenhauer, 2016-11-26:
Tell me what to do. I'm trying to look into the issue as much as I can. It seems that the samplingInterval value is being set in board.io.settings but is being ignored.

soundanalogous, 2016-11-26:
Are you using a version of ConfigurableFirmata built from firmatabuilder.com or just the default version you get from firmata/ConfigurableFirmata/examples/ConfigurableFirmata.ino?

jackrosenhauer, 2016-11-26:
I've tried both and get the same result. Currently using firmata/ConfigurableFirmata/examples/ConfigurableFirmata.ino.

soundanalogous, 2016-11-26:
Did you make any modifications?

jackrosenhauer, 2016-11-26:
Nope.

jackrosenhauer, 2016-11-26:
Just to be thorough I do get these 2 compile warnigns http://pastebin.com/NRQnCU9K

soundanalogous, 2016-11-26:
Copy this to a new file and compile and upload: https://gist.github.com/soundanalogous/7e7643cd8cf00a5f90de615f5d52b9ec

Also make sure you have ConfigurableFirmata v2.9.1 installed.

jackrosenhauer, 2016-11-26:
Seems to be outputting mostly gibberish besides the sketch name.

soundanalogous, 2016-11-26:
Probably seeing ascii text. If converted to raw bytes that may be helpful info.

soundanalogous, 2016-11-26:
Add this to the end of the setup function:

```c++
// set sampling interval to 1000ms
reporting.setSamplingInterval(1000);
```

jackrosenhauer, 2016-11-26:
How do I get the raw bytes being sent?

soundanalogous, 2016-11-26:
It depends on where the stream is coming from. Where is the gibberish data showing up? Is it part of the error stream or is it from console.log, etc?

soundanalogous, 2016-11-26:
Also please share you full javascript code.

jackrosenhauer, 2016-11-26:
I should just be able to open the serial monitor (COM) and see the output, right? 

code: https://gist.github.com/jackrosenhauer/b4452d2ffbf09409fc4add1ebb4c34e1

soundanalogous, 2016-11-26:
You can try, but I don't think it will work because the serial monitor will own the connection and you won't be able to communicate with your J5 application.

soundanalogous, 2016-11-26:
So you get no output when running that script?

jackrosenhauer, 2016-11-26:
I don't get any additional output while running the J5 app

soundanalogous, 2016-11-26:
Strange. Can you copy and paste the gibberish data. I can probably decipher it if it's ascii.

jackrosenhauer, 2016-11-26:
The gibberish data is not in ascii and was from the serial monitor.

soundanalogous, 2016-11-27:
Add this to your script inside the "ready" function block:

```js
board.on("string", function (message) {
  console.log(message);
});
```

soundanalogous, 2016-11-27:
Then update ConfigurableFirmataDebug.ino with the following: https://gist.github.com/soundanalogous/7e7643cd8cf00a5f90de615f5d52b9ec.

Compile and upload then run your sensors.js. It should hopefully print "reporting..." once a second.

jackrosenhauer, 2016-11-27:
It does indeed print reporting... every second but nothing else.

soundanalogous, 2016-11-27:
That's good. It indicates the firmware isn't crashing. Getting more meaningful data will be difficult as it will require modification of the ConfigurableLibrary.

soundanalogous, 2016-11-27:
Also, just to make sure everything else is working as intended with this firmware, comment out the `thermometer` and let me know if that prints data, then uncomment `thermometer` and comment out `hygrometer` and let me know if that prints data.

jackrosenhauer, 2016-11-27:
Any variation prints "reporting..." but J5 exits on the 1-Wire device read timeout error.

soundanalogous, 2016-11-27:
Go to your Arduino/libraries/ directory and make a copy of ConfigurableFirmata (as a backup) and move that copy out of the libraries directory. Then in the ConfigurableFirmata that remains in Arduino/libraries/, replace the existing /ConfigurableFirmata/src/OneWireFirmata.cpp with this version: https://gist.github.com/soundanalogous/b46593e9fa90ae5846146cc7e6253e16.

If you have the Arduino IDE open, you'll need to close then reopen it after updating that file. The compile and upload ConfigurableFirmataDebug.ino again and let me know what your script prints. It should print a few numbers periodically.

jackrosenhauer, 2016-11-27:
output sample: https://gist.github.com/jackrosenhauer/ddbe376f2212751d86e05ca8e11abd3c

soundanalogous, 2016-11-27:
Now replace ConfigurableFirmata/src/I2CFirmata.h with this file: https://gist.github.com/soundanalogous/af3eba67c19f29dd9a0a1ef5f4d42a31

jackrosenhauer, 2016-11-27:
output#2 sample: https://gist.github.com/jackrosenhauer/1a7b96debcac9f69f7e542c7411b7708

soundanalogous, 2016-11-27:
strange... seeing no strings from I2C, is both thermomenter and hydrometer uncommented?

soundanalogous, 2016-11-27:
Also, here's another update to OneWireFirmata.cpp: https://gist.github.com/soundanalogous/b46593e9fa90ae5846146cc7e6253e16

jackrosenhauer, 2016-11-27:
I had thermometer enabled and hygrometer disabled. hygrometer is now enabled, I've also updated the OneWireFirmata.cpp. 

Output:
I2C write
I2C read
I2C read
...

soundanalogous, 2016-11-27:
That's all you get is those 3 lines now?

jackrosenhauer, 2016-11-27:
No, it continues to output I2C read.

soundanalogous, 2016-11-27:
but no more subcmd values?

[41 more replies]

closed 2018-08-28 by stale[bot]