rwaldron/johnny-five #1117: Accelerometer: update fixed point values to match datasheets for fractional digit accuracy

rwaldron, 2016-05-04:


rwaldron, 2016-05-04:
Sorry for the confusion here, I filed this from a mobile device and I fat fingered a few incorrect labels.

faraazahmad, 2017-03-12:
the tests should be in https://github.com/rwaldron/johnny-five/blob/master/test/accelrometer.js right?

rwaldron, 2017-03-20:
@faraazahmad yes, this is an on going process.

dnlup, 2017-11-24:
@rwaldron Hi, I would like to try to work on this one.
Need to ask you some things before:
*   the method in `accelerometer.js` you need to be fixed is `toGravity`
*   you want the decimals to be fixed as the resolution reported in each device datasheet
*   where should I implement test case in `test/accelerometer.js`? Looking through `test/thermometer.js` I find the `digital.fractional()` test called in different methods, you want to use a custom function or should I use a specific existing one for the accelerometer?

Sorry if the questions could be obvious.

Thanks.

rwaldron, 2017-11-27:
- Yes, each occurrence
- Yes, exactly :)
- ~~Striken~~
- Yes, I believe that would be correct :)

closed 2018-11-29 by stale[bot]