rwaldron/johnny-five #1123: Proximity: update fixed point values to match datasheets for fractional digit accuracy

rwaldron, 2016-05-04:


rwaldron, 2016-05-04:
Sorry for the confusion here, I filed this from a mobile device and I fat fingered a few incorrect labels.

henricavalcante, 2016-05-11:
Can I try this one?

droidenator, 2016-10-05:
I'm looking into this issue and just wanted to be sure it hasn't already been resolved. When I look into proximity.js, I see a lot of `toCm` methods that look like this

```
    toCm: {
      value: function(raw) {
        return toFixed(12343.85 * Math.pow(raw, -1.15), 2);
      }
    }
```

But then I see other ones like

```
    toCm: {
      value: function(raw) {
        // http://www.maxbotix.com/articles/032.htm
        return raw / 2;
      }
    }
```

I'm assuming all return values need to be run through `toFixed`? Also, some are set to return 2 digits after the decimal, others 3 digits. I looked at the datasheets but I didn't see any clear indication why one might return 3 digits instead of 2 or 0.

closed 2018-08-28 by stale[bot]