rwaldron/johnny-five #1124: Problem with IMU yaw.angle using the BNO055 and Arduino uno

inoui, 2016-05-05:
I have a problem using the BNO055 with an arduino uno.
The data returned by yaw.angle is fine if the angle is between -100° and +100° 
but if it's lower or higher, the angle is return by 10° steps, ie 100° / 110° / 120° etc...

johnny-five@0.9.37
node@v5.3.0

I m using basic code 
```
board.on("ready", function() {
    var imu = new five.IMU({
        controller: "BNO055"
    });
    imu.on("change", function() {
        console.log("  yaw    angle      : ", this.gyro.yaw.angle);
    });
});
```

Output is

```
90
91
92
...
99 <--- all good, 1 by 1 step
100
110
120 <--- now 10° by 10°
```

inoui, 2016-05-05:
I found that the problem comes from yaw.angle precision set to 2 in gyro.js

```
line 282 : angle: this.z !== undefined ? ToPrecision(state.z.angle, 2) : 0
```

Is this done on purpose ?

rwaldron, 2016-05-05:
Ugh. I have no idea why precision was used there. Nice find, fixing now.

closed 2016-05-05 by rwaldron