rwaldron/johnny-five #1125: Using ENC28J60 with Jonny Five

alfredoborrero, 2016-05-05:
Hi, is there any way to use the Ethernet Controller "ENC28J60" instead of using the "Arduino Ethernet Shield" ?
I would like to use the "ENC28J60" for my project because is smaller and I don't really need to use the SD feature that comes with the Ethernet Shield and of course is cheaper to use the ENC28J60.

i think the firmware "StandardFirmataEthernet" only works with Ethernet Shields, is that correct?
Is there any other firmware/sketch that i can use?

Regards,

Alfredo Borrero

soundanalogous, 2016-05-05:
You need to use [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata). The easiest way is to use [firmatabuilder.com](firmatabuilder.com), select Ethernet then ENC28J60 as the controller, then the IP of the computer running J5. If you want the equivalent features as StandardFirmata, select the following features:

- DigitalInputFirmata
- DigitalOutputFirmata
- AnalogInputFirmata
- AnalogOutputFirmata
- ServoFirmata
- I2CFirmata

Other features are optional, but don't select all of them unless you're board can handle it. Name you file, click Build and Download and an Arduino Sketch will be generated and download. Compile and upload it using the Arduino IDE. Note that you'll need to install the ConfigurableFirmata library in order to compile the sketch.

A couple of important things to note about using the ENC28J60 with Firmata:

1. You will need a board with more memory than an Arduino Uno or Leonardo provides. An Arduino Mega or Due will work, I'm not sure about any other boards since the UIPEthernet library only runs on AVR or SAM architectures. The [UIPEthernet library](https://github.com/ntruchsess/arduino_uip) (that you'll also need to download and install) consumes too much ram to also run Firmata on an ATmega328p or Atmega32u4 (Uno, Leonardo, etc)
2. You will need to either use an ENC28J60 breakout board or if you use a shield, you need to ensure that the shield bridges the 6 pin ICSP pin on the board to the shield (this is rare among ENC28J60 shields) - if it does not it will not work with the higher RAM boards because they all pass their SPI pins through that port.

So in other words, the WIZNET shields are more expensive, but they are much easier to use and can be run on boards like the Uno since the Arduino Ethernet library requires far less memory than the UIPEthernet library.

alfredoborrero, 2016-05-24:
Thank you so much Jeff. 

I will build the firmware using [firmatabuilder.com](url) 
And for the memory issue I am using Arduino Mega connected to an ENC28J60 breakout board.

dtex, 2016-06-03:
@alfredoborrero It looks like @soundanalogous was able to answer all your questions so I'm going to close this. Feel free to re-open if there is something left unresolved.

closed 2016-06-03 by dtex