rwaldron/johnny-five #1126: DMP processed values for MPU6050

alexborisov, 2016-05-06:
From what I can tell the IMU class only pulls the raw accel/gyro values (as can be observed from the gyro value gradually drifting when sat still). So first up could someone please confirm if this is the case (or it's just something on my end).

If it is the case, It would be a lot more useful if we were provided the calculated fused values supplied by the onboard DMP. This provides much better values than what can be computed manually. It would also make more sense in the context of an _IMU_ since the whole point of one is to provide you with multiple sensors to calculate a better position vs just a convenience wrapper for accel and gyro. I expect the IMU class to provide me data already processed with a complementary/kalman filter or with values returned by the chips onboard processing.

Docs have been published by the manufacturer and there are arduino libraries that use this...

dtex, 2018-01-11:
Hi @alexborisov ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the DMP Processed Values for the MPU6050 (and other IMU devices I would assume) there.

closed 2018-01-11 by dtex

alexborisov, 2018-01-11:
Great thanks! It's a nice cheap chip and goes great with a rpi zero.

- Alex.