rwaldron/johnny-five #1128: How to fix  EIO,i/o error

asit569, 2016-05-10:
Getting this error please resolve

pi@raspberrypi:~ $ cd waterpi-node
pi@raspberrypi:~/waterpi-node $ sudo node index.js
1462815306876 Device(s) RaspberryPi-IO
1462815307277 Connected /dev/ttyACM0
1462815307365 Connected RaspberryPi-IO
connect
1462815311112 Board ID:  pi
1462815311114 Board ID:  uno
1462815311129 Repl Initialized
>> ready
Temperature: 33.00C, humidity: 45.00%
fs.js:613
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^
Error: EIO, i/o error
    at Error (native)
    at Object.fs.writeSync (fs.js:613:20)
    at Bus.i2cWriteSync (/home/pi/waterpi-node/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/i2c-bus.js:340:13)
    at I2C.writeSync (/home/pi/waterpi-node/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:332:34)
    at Raspi.i2cWrite (/home/pi/waterpi-node/node_modules/raspi-io/lib/index.js:559:19)
    at Expander.write (/home/pi/waterpi-node/node_modules/johnny-five/lib/lcd.js:49:11)
    at LCD.Controllers.PCF8574.initialize.value (/home/pi/waterpi-node/node_modules/johnny-five/lib/lcd.js:341:23)
    at new LCD (/home/pi/waterpi-node/node_modules/johnny-five/lib/lcd.js:734:10)
    at Boards.<anonymous> (/home/pi/waterpi-node/index.js:60:9)
    at Boards.emit (events.js:107:17)
    at Boards.<anonymous> (/home/pi/waterpi-node/node_modules/johnny-five/lib/board.js:1197:14)
    at process._tickDomainCallback (node.js:381:11)

nebrius, 2016-05-10:
Looking at the output, it would appear that you are trying to use Raspi IO _and_ an Arduino Uno connected to the serial port, is that correct? Do you need to use both boards?

If so, I think I know what's going on, Raspi IO is trying to claim the serial port for use with `board.io.serial*` and Johnny-Five is also trying to claim the serial port for use with the Uno+Firmata. Yeah, we need to do something about this.

Regardless, I think it would be worthwhile to add a flag to the Raspi IO constructor to tell it to not initialize the serial port.

nebrius, 2016-05-10:
Thinking through this more, this serial port that it's talking about (`/dev/ttyAMA0`) is different than the USB one used by an Uno. @rwaldron does J5 try and open all serial ports it can find and scan them for Firmata devices? If so, is there a way to blacklist certain ports?

fivdi, 2016-05-10:
The stack trace indicates that an attempt is being made to send information to an I2C LCD wired to the GPIO header on a Raspberry Pi but the I2C LCD can't be found.

As mentioned above, there also appears to be an Arduino Uno connected via USB (/dev/ttyACM0) to the Raspberry Pi, is this correct? There's no mention of the serial port broken out onto the GPIO header on the Raspberry Pi (/dev/ttyAMA0) in the OP.

@asit569 It's probably best if you explain what needs to be achieved with the Arduino Uno and the Raspberry Pi.

asit569, 2016-05-10:
@fivdi i am using aws iot and dynamo db for sensor will timely monitor and display it on lcd if everything correct but it ends up with error please see these below what i want to achieve

https://www.hackster.io/demirhanaydin/waterpi-houseplant-remote-watering-and-monitoring-system-340400
https://github.com/demirhanaydin/waterpi-node

fivdi, 2016-05-10:
Lets see if the LCD can be found on the Raspberry Pi.

First run this command:
```
pi@raspberrypi:~ $ ls -l /dev/i2*
```
This should output either this:
```
crw-rw---- 1 root i2c 89, 1 May  8 23:04 /dev/i2c-1
```
or this:
```
crw-rw---- 1 root i2c 89, 1 May  8 23:04 /dev/i2c-0
```
If `/dev/i2c-1` was found, run the following command:
```
i2cdetect -y -r 1
```
If `/dev/i2c0` was found, run the following command:
```
i2cdetect -y -r 0
```
Note that if command i2cdetect can't be found it will need to be installed with the following command first:
```
sudo apt-get update
sudo apt-get install i2c-tools
```
Then post the output of the command `i2cdetect`

nebrius, 2016-05-10:
Good catch, I was misreading the output @fivdi.

fivdi, 2016-05-10:
@nebrius perhaps you were being distracted by work :)

asit569, 2016-05-10:
@fivdi     pi@raspberrypi:~/waterpi-node $ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

asit569, 2016-05-10:
i have  detached all parts lcd,dht 11,relay from pi then also same error exist @nebrius @fivdi

fivdi, 2016-05-10:
Yes, if everything is detached the same error should occur.

Please edit `/home/pi/waterpi-node/index.js`

and change [these lines](https://github.com/demirhanaydin/waterpi-node/blob/master/index.js#L60-L63):
```
  lcd = new five.LCD({ 
    controller: "LCM1602",
    board: this.byId('pi')
  });
```
to:
```
  lcd = new five.LCD({ 
    controller: "PCF8574A",
    board: this.byId('pi')
  });
```
and then check to see if the LCD works.

asit569, 2016-05-10:
Yes finally it solves the issue @fivdi Thankyou

nebrius, 2016-05-10:
> perhaps you were being distracted by work :)

It's what they pay me for after all :) Thanks for the help @fivdi, and I'm glad you got it working @asit569!

fivdi, 2016-05-10:
Excellent!

@asit569 If all is ok now you could hit the `Close issue` button at the bottom of the page :)

closed 2016-05-10 by asit569

yangzai, 2016-05-23:
Is there any way to catch i2c errors like this?

nickcoutsos, 2016-08-05:
Sorry to bring up a closed issue but I got here with the same question as @yangzai and didn't see a solution documented. The i2c error is raised by the board itself and can be handled by listening for its _error_ event:

```
board.on('error', function(err) { console.error(err) })
```

NishantWioc, 2018-03-06:
This problem still exists for me Please Help

`1520356539428 Available RaspberryPi-IO
1520356539906 Connected /dev/ttyACM0
1520356540206 Connected RaspberryPi-IO
connect
1520356544141 Board ID:  pi
1520356544150 Board ID:  uno
1520356544197 Repl Initialized
>> ready
Temperature: 0.00C, humidity: 0.00%
fs.js:753
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EREMOTEIO: remote I/O error, write
    at Object.fs.writeSync (fs.js:753:20)
    at Bus.i2cWriteSync (/home/pi/Desktop/waterpi-node-master/node_modules/i2c-bus/i2c-bus.js:348:13)
    at I2C.writeSync (/home/pi/Desktop/waterpi-node-master/node_modules/raspi-i2c/dist/index.js:294:38)
    at RaspiIOCore.i2cWrite (/home/pi/Desktop/waterpi-node-master/node_modules/raspi-io-core/dist/index.js:744:19)
    at Expander.portWrite (/home/pi/Desktop/waterpi-node-master/node_modules/johnny-five/lib/lcd.js:64:11)
    at LCD.value (/home/pi/Desktop/waterpi-node-master/node_modules/johnny-five/lib/lcd.js:371:23)
    at new LCD (/home/pi/Desktop/waterpi-node-master/node_modules/johnny-five/lib/lcd.js:792:10)
    at Boards.<anonymous> (/home/pi/Desktop/waterpi-node-master/index.js:60:9)
    at Boards.emit (events.js:159:13)
    at Boards.<anonymous> (/home/pi/Desktop/waterpi-node-master/node_modules/johnny-five/lib/board.js:1208:14)
`

I tried the above solutions, My LCD is connected and i'm able to detect it using the method suggested by @fivdi 

I changed 
` lcd = new five.LCD({ 
    controller: "LCM1602",
    board: this.byId('pi')
  });
`
` lcd = new five.LCD({ 
    controller: "PCF8574AT",
    board: this.byId('pi')
  });
`
Beacause my I2c module uses this controller i found it on the backside of it

fivdi, 2018-03-07:
@NishantWioc 

The `"remote I/O error"` means that Johnny-Five / Raspi-IO can't communicate with the I2C LCD. 

Can you post the complete output of `i2cdetect -y -r 1` please?
Please also post the complete content of `/boot/config,txt`.
Can you post a picture that clearly shows how the I2C LCD is connected to the Raspberry Pi? 

>  Is there something related to temperature sensor not working or not connected ?

There appears to be something wrong as `"Temperature: 0.00C, humidity: 0.00%"` is displayed which doesn't seem plausible. I'm not familiar with the [node-dht-sensor](https://github.com/momenso/node-dht-sensor) module which is used to measure the temperature and humidity. You'll need to figure out if the library is being used correctly. It may also be necessary to use the [Raspi-IO](https://github.com/nebrius/raspi-io) excludePins option to tell Raspi-IO not to initialize the pins used by the DHT sensor.

NishantWioc, 2018-03-08:
**Output for `ls -l /dev/i2*`**
`crw-rw---- 1 root i2c 89, 1 Mar  6 18:02 /dev/i2c-1`

**Output For `i2cdetect -y -r 1`**

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

**Content of  Boot/config.txt**
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# NOOBS Auto-generated Settings:

#hdmi_force_hotplug=1

dtparam=i2c_arm=on
dtparam=i2c=on
dtparam=i2c_arm_baudrate=100000

Image for Raspberry to LCD connection
[image]

**`Image for i2cdetect -y -r 1`**
[image]

NishantWioc, 2018-03-08:
## MORE PICTURES 
[image]
[image]
[image]
[image]
[image]

NishantWioc, 2018-03-08:
# Now I'm getting this New Error 

[image]

fivdi, 2018-03-08:
Nice pictures. Everything and in particular the LCD looks good to me.

The "TypeError: Cannot read property forEach of undefined" is occurring on [this line of code](https://github.com/rwaldron/johnny-five/blob/master/lib/lcd.js#L651) which is for a parallel LCD not an I2C LCD.

Has the code been modified since yesterday?
Is the following line of code not being use to create an LCD instance anymore?
```js
 lcd = new five.LCD({ controller: "PCF8574AT", board: this.byId('pi') });
```

fivdi, 2018-03-08:
There's also an Arduino UNO connected to one of the USB ports on the Pi, correct? I can't see it in the photos.

fivdi, 2018-03-08:
Maybe it would be a good idea to test the components one by one to get them to work before tyring everything at the same time.

For example, if the Arduino UNO is disconnected from the USB port and the following (untested) program is run on the Pi does it display anything on the LCD?
```js
const Raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new Raspi()
});

board.on('ready', () => {
  var count = 0;
  var lcd = new five.LCD({ 
    controller: "PCF8574AT"
  });

  setInterval(() => {
    count += 1;
    lcd.print('Hello' + count);
  }, 1000);
});
```

NishantWioc, 2018-03-09:
Thanks for your quick reply :)
umm  I didn't modified the code. What i can do is download the project once again with clean code. Because it may be possible that i removed something form code during removing lcd code. but that's an another file with lcd code removed

Yes You are right there's an Arduino Uno connected to one of the Usb Port of pi, 
I'll try the same as you suggested then i will tell you the results :)
Thanks once again :D

NishantWioc, 2018-03-09:
# Lcd Prints Nothing
I tried the above code and everything looked fine, i disconnected  Arduino too but lcd prints nothing

#### following is the output
`
1520617959178 Available RaspberryPi-IO
1520617959535 Connected RaspberryPi-IO
1520617959617 Repl Initialized
>>
`
#### Here's the picture
[image]

# It tried something 
`lcd.print('Hello');
	console.log(lcd.print('Hello'))`

## Output 

`1520618340723 Available RaspberryPi-IO
1520618341140 Connected RaspberryPi-IO
1520618341233 Repl Initialized
>> LCD {
  board:
   Board {
     io:
      RaspiIOCore {
        domain: null,
        _events: [Object],
        _eventsCount: 4,
        _maxListeners: undefined,
        name: 'RaspberryPi-IO',
        isReady: [Getter],
        pins: [Getter],
        analogPins: [Getter],
        MODES: [Object],
        HIGH: 1,
        LOW: 0,
        defaultLed: -1,
        SERIAL_PORT_IDs: [Object] },
     timer:
      Timeout {
        '0': null,
        _called: false,
        _idleTimeout: -1,
        _idlePrev: null,
        _idleNext: null,
        _idleStart: 5974,
        _onTimeout: null,
        _timerArgs: undefined,
        _repeat: null,
        _destroyed: false,
        [Symbol(asyncId)]: 11,
        [Symbol(triggerAsyncId)]: 7 },
     isConnected: true,
     isReady: true,
     register: [ [Circular] ],
     occupied: [],
     Drivers: {},
     id: 'DE2690CA-F153-4772-A103-41E63E2F5537',
     debug: true,
     repl: Repl { context: [Object], ready: true, cmd: [REPLServer] },
     sigint: true,
     pins:
      Pins {
        '0': [Object],
        '1': [Object],
        '2': [Object],
        '3': [Object],
        '4': [Object],
        '5': [Object],
        '6': [Object],
        '7': [Object],
        '8': [Object],
        '9': [Object],
        '10': [Object],
        '11': [Object],
        '12': [Object],
        '13': [Object],
        '14': [Object],
        '15': [Object],
        '16': [Object],
        '17': [Object],
        '18': [Object],
        '19': [Object],
        '20': [Object] },
     transport: null,
     port: 'RaspberryPi-IO',
     type: 'OTHER',
     _events: { ready: [Function] },
     _eventsCount: 1,
     MODES: { INPUT: 0, OUTPUT: 1, ANALOG: 2, PWM: 3, SERVO: 4 },
     millis: [Function] },
  io:
   RaspiIOCore {
     domain: null,
     _events:
      { close: [Function: bound ],
        disconnect: [Function: bound ],
        error: [Function: bound ],
        string: [Function: bound ] },
     _eventsCount: 4,
     _maxListeners: undefined,
     name: 'RaspberryPi-IO',
     isReady: [Getter],
     pins: [Getter],
     analogPins: [Getter],
     MODES: { INPUT: 0, OUTPUT: 1, ANALOG: 2, PWM: 3, SERVO: 4 },
     HIGH: 1,
     LOW: 0,
     defaultLed: -1,
     SERIAL_PORT_IDs: { HW_SERIAL0: '/dev/ttyAMA0', DEFAULT: '/dev/ttyAMA0' } },
  id: '0B7F4E4B-BD82-4092-B263-56F9D574D61D',
  custom: {},
  controller: 'PCF8574AT',
  ctype: 'PCF8574AT',
  bitMode: 4,
  lines: 2,
  rows: 2,
  cols: 16,
  dots: '5x8',
  address: { lcd: 63 },
  expander:
   Expander {
     address: 63,
     mask: 0,
     memory: 249,
     io:
      RaspiIOCore {
        domain: null,
        _events: [Object],
        _eventsCount: 4,
        _maxListeners: undefined,
        name: 'RaspberryPi-IO',
        isReady: [Getter],
        pins: [Getter],
        analogPins: [Getter],
        MODES: [Object],
        HIGH: 1,
        LOW: 0,
        defaultLed: -1,
        SERIAL_PORT_IDs: [Object] } } }`

fivdi, 2018-03-09:
So there are no errors but nothing is displayed on the LCD. This is strange. Does adjusting the blue potentiometer on the LCD interface board help? If it doesn't, I can't think of anything else to do other than trying to debug the problem. I'm afraid I don't have an I2C LCD to give it a try.

NishantWioc, 2018-03-13:
I think either i have to adujst the potentiometer  or the lcd is getting the garbage data. For Now i have just removed the lcd code, and the project is working fine. I'll try to debug the problem :) Thanks for your help ;)