rwaldron/johnny-five #1129: Node-Red + Arduino Uno + Johnny five + Windows 10

Jawady, 2016-05-11:
> please help, I have been trying for days and I get this error message only when I use Johhny five node. I tried three boards and flushed them again and again with Firmata . I keep geeting below when I deploy, thx
>
> Welcome to Node-RED
> ===================
>
> 10 May 13:08:45 - [info] Node-RED version: v0.13.4
> 10 May 13:08:45 - [info] Node.js version: v4.4.4
> 10 May 13:08:45 - [info] Windows_NT 10.0.10586 x64 LE
> 10 May 13:08:45 - [info] Loading palette nodes
> 10 May 13:08:51 - [warn] ------------------------------------------
> 10 May 13:08:51 - [warn] Failed to register 2 node types
> 10 May 13:08:51 - [warn] Run with -v for details
> 10 May 13:08:51 - [warn] ------------------------------------------
> 10 May 13:08:51 - [info] Settings file : C:\Users\Haydar\AppData\Roaming\npm\node_modules\node-red\settings.js
> 10 May 13:08:51 - [info] User directory : \Users\Haydar\.node-red
> 10 May 13:08:51 - [info] Flows file : \Users\Haydar\.node-red\flows_DESKTOP-IPFG6GO.json
> 10 May 13:08:51 - [info] Starting flows
> 1462882131089 Device(s) Firmata
> 10 May 13:08:51 - [info] Started flows
> 10 May 13:08:51 - [info] Server now running at http://127.0.0.1:1880/
> 1462882131136 Connected Firmata
> 1462882135574 Device(s) COM3
> 1462882135605 Connected COM3
> Opening \\.\COM3: Access denied
> Port is not open
> Port is not open
> 1462882145637 Device or Firmware Error A timeout occurred while connecting to the Board.
>
> Please check that you've properly flashed the board with the correct firmware.
> See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
>
> If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
> 10 May 13:09:05 - [red] Uncaught Exception:
> 10 May 13:09:05 - Error: Uncaught, unspecified "error" event. ([object Object])
> at Board.emit (events.js:144:17)
> at Board.log (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:642:8)
> at Board.(anonymous function) [as error] (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:653:14)
> at Board.<anonymous> (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:421:14)
> at Timer.listOnTimeout (timers.js:92:15)

henricavalcante, 2016-05-11:
Have you tried to run as administrator?

Jawady, 2016-05-11:
thank you let me check and revert back

Jawady, 2016-05-11:
i wasn't just did, and getting the same 

C:\>node-red

Welcome to Node-RED
===================

11 May 13:59:06 - [info] Node-RED version: v0.13.4
11 May 13:59:06 - [info] Node.js  version: v4.4.4
11 May 13:59:06 - [info] Windows_NT 10.0.10586 x64 LE
11 May 13:59:06 - [info] Loading palette nodes
11 May 13:59:07 - [warn] ------------------------------------------
11 May 13:59:07 - [warn] Failed to register 2 node types
11 May 13:59:07 - [warn] Run with -v for details
11 May 13:59:07 - [warn] ------------------------------------------
11 May 13:59:07 - [info] Settings file  : C:\Users\Haydar\AppData\Roaming\npm\node_modules\node-red\settings.js
11 May 13:59:07 - [info] User directory : \Users\Haydar\.node-red
11 May 13:59:07 - [info] Flows file : \Users\Haydar\.node-red\flows_haydar.json
11 May 13:59:07 - [info] Creating new flow file
11 May 13:59:07 - [info] Server now running at http://127.0.0.1:1880/
11 May 13:59:07 - [info] Starting flows
11 May 13:59:07 - [info] Started flows
11 May 13:59:50 - [info] Stopping flows
11 May 13:59:50 - [info] Stopped flows
11 May 13:59:50 - [info] Starting flows
1462939190482 Device(s) Firmata
11 May 13:59:50 - [info] Started flows
1462939190492 Connected Firmata
1462939194425 Device(s) COM5
1462939194442 Connected COM5
Opening \\.\COM5: Access denied
Port is not open
Port is not open
1462939204443 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
11 May 14:00:04 - [red] Uncaught Exception:
11 May 14:00:04 - Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:642:8)
    at Board.(anonymous function) [as error] (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:653:14)
    at Board.<anonymous> (C:\Users\Haydar\node_modules\node-red-contrib-gpio\node_modules\johnny-five\lib\board.js:421:14)
    at Timer.listOnTimeout (timers.js:92:15)

mwittig, 2016-05-11:
As far as I know it only possible to open the serial port once on WIndows (same on Linux btw). See also <http://stackoverflow.com/questions/12757707/is-it-possible-to-open-a-serial-port-multiple-times>
Maybe the problem is that the port has already been openened by another application or the combination of NodeRed and J5 results in multiple attempts to open the port.

Jawady, 2016-05-11:
thank you Marcus, i can open the port using the Arduino node in NodeRed
same port open no issues, only JF crash.

monteslu, 2016-05-11:
@Jawady 
Are you able to run the johnny-five examples outside of node-red?

Which board are you using?  And what version of node.js ?

Will help if we can isolate the issue.

rwaldron, 2016-05-11:
There was another issue filed about this COM3 port, about 13 hours before this one https://github.com/rwaldron/johnny-five/issues/1131 I'm not sure it's the same bug, but something is clearly amiss...

Jawady, 2016-05-11:
I have run one JF code   outside Node Red only one the Blink and it work
ok, but for information i have onmy three boards one is directly from
Aurduino  website and two others are from kits i bought. all work great  in
Node-red but not JF. the node.js i installed is from the official website i
tried the latest  32bit and 64bit all the same issues. all other insatlled
are from Node-Red instructions  i also tried to isolate using virtual Box
using Windows  10 32bit, 64 bit, windows 8.1 and Linux all the same access
denied with JF.

 once i get this error i have to go to .Node-red file in users  and delete
the config files for Node-red to excute again, then the same thing  happens
when using JF.

Thank you

Jawady, 2016-05-11:
i tried to add the code from 1131 to JF node still no joy. unless i am
missing the point where to add it to ?

new five.Board({
  port: "COM4"
});

Jawady, 2016-05-12:
i have installed a fresh copy of windows 10 32bit on Virtual Box, + Node.Js
32 bit, + NodeRed + JF and uploaded below code still getting the same

[image: Inline images 3]

Jawady, 2016-05-12:
[image: Inline images 1]

rwaldron, 2016-05-13:
I know that many other people are using J5 with Node Red without any trouble. It would be great if someone from that project could chime in with some helpful hints.

rwaldron, 2016-05-13:
> i have installed a fresh copy of windows 10 32bit on Virtual Box,

Does your Virtual Box VM actually have access to the serial ports? I know that one must _explicitly_ expose USB devices to a VM for them to be accessible.

Jawady, 2016-05-14:
Everything work Ok with and without the virtual box, i have loads of nodes
including two Arduino  nodes IBM Watson etc they work great in host and
Virtual Box also this issue I had the same in Linux . Only JF through give
me this error. i am not giving up i want to use everything in JF so
i think its a combination of my Surface 3 and the Arduino Uno. i am waiting
for Mega board I need one anyway.

Jawady, 2016-05-14:
i agree loads of people are using J5 and Node Red its very strange case i
am having

Jawady, 2016-05-18:
just got the Amega board the same error, i am now confident this a surface
3 thing

reconbot, 2016-08-26:
I'm going to close this due to its age, but If this is still an issue. Please comment and we'll reopen. I also recommend opening an issue on node red and we'll do out best to help debug.

closed 2016-08-26 by reconbot

Jawady, 2016-08-27:
Yes sure I bought a Mac Pro and will test JF with it and let you know.

Jawady, 2016-08-28:
hi i am now getting this error on my new mac

1472381228560 Connected Firmata

1472381230209 Device(s) /dev/cu.usbmodem1421

1472381230247 Connected /dev/cu.usbmodem1421

1472381240249 Device or Firmware Error A timeout occurred while connecting
to the Board.

Please check that you've properly flashed the board with the correct
firmware.

See:
https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on
the board, wait approximately 11 seconds for complete reset, then run your
program again.

28 Aug 20:47:20 - [red] Uncaught Exception:

28 Aug 20:47:20 - Error: Uncaught, unspecified "error" event. ([object
Object])

    at Board.emit (events.js:144:17)

    at Board.log
(/Users/haydarjawad/.node-red/node_modules/node-red-contrib-gpio/node_modules/johnny-five/lib/board.js:625:8)

    at Board.(anonymous function) [as error]
(/Users/haydarjawad/.node-red/node_modules/node-red-contrib-gpio/node_modules/johnny-five/lib/board.js:636:14)

    at Board.<anonymous>
(/Users/haydarjawad/.node-red/node_modules/node-red-contrib-gpio/node_modules/johnny-five/lib/board.js:385:14)

    at Timer.listOnTimeout (timers.js:92:15)

gcrev93, 2016-11-21:
I am not using Node Red.. I am actually using J5 with starter-feather (https://github.com/IoTReX/starter-feather) . I am using a Surface Pro , and win 10. I get this:

1479692813166 Device(s) Firmata
1479692813218 Connected Firmata
1479692814365 Device(s) COM3
1479692814369 Connected COM3
1479692814370 Error Opening COM3: Access denied
events.js:146
      throw err;
      ^

soundanalogous, 2016-11-23:
@gcrev93 I believe you should upload [StandardFirmataWiFi](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi) to your feather huzzah board. See the instructions in the [wifiConfig.h file](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h). It looks like you need to configure the board as a TCP server, which is the default configuration in StandardFirmataWiFi. You'll simply need to add your WiFi credentials where indicated in wifiConfig.h.

jourdanrodrigues, 2017-10-05:
Passing for the same problem. Runs just fine on Mac + J5, but Windows gives me access denied on any port (including as administrator).