rwaldron/johnny-five #1130: i2c LCD not working no character

asit569, 2016-05-11:
code runs up but i2c LCD only backlight turns on but no character displays on lcd

changed "1602LCD" TO "PCF8574A"

Errors comming when it sets to 1602LCD but no error after changing it to PCF8574A

lcd = new five.LCD({ 
    controller: "LCM1602",
    board: this.byId('pi')
  });
to:

  lcd = new five.LCD({ 
    controller: "PCF8574A",
    board: this.byId('pi')
  });

rwaldron, 2016-05-11:
Thanks for filing! 

> changed "1602LCD" TO "PCF8574A"

That's good, because `1602LCD` isn't a valid `LCD` controller ;)

Here are the correct I2C/LCD controllers: http://johnny-five.io/api/lcd/#i2c-options

asit569, 2016-05-11:
@rwaldron this issue is still active that i am not getting character on lcd and only backlight on

asit569, 2016-05-11:
@rwaldron in above i told that if i change lcm1602 to PCF8574A then the code error vanish but lcd backlight power on with no values shows in lcd and lsd display is blank

asit569, 2016-05-11:
got resoled by adjusting the potentiometer backside lcd in controller

rwaldron, 2016-05-11:
I misunderstood, I thought you were saying that it worked when you changed to the `PCF8574A` controller.

rwaldron, 2016-05-11:
Can you link me to the _exact_ LCD screen that you're using?

asit569, 2016-05-11:
Got lcd working by adjusting the knob in controller @rwaldron

closed 2016-05-11 by asit569

rwaldron, 2016-05-11:
Oh, that's fantastic!