rwaldron/johnny-five #1131: Not able to connect to correct COM port in Windows

mithunkr, 2016-05-11:
While this worked fine on Mac OSX, in windows, this is not able to connect to the Arduino over the correct COM Port.
My Arduino is connected to COM4 (automatically assigned by windows). However the output i get when executing the small test program is like this:
"1462980829055 Device(s) COM3,COM4
1462980829068 Connected COM3
1462980839069 Device or Firmware Error A timeout occurred while connecting to the Board.

etc...."

Why doesn't it try to connect to the COM4 at all? when the Arduino is connected to COM4, why is it saying 'Connected COM3'? I can see under Arduino IDE menu, that it's connected to COM4.
And I've successfully uploaded the standard Firmata to it as well... 
It works fine on my Mac, though.

rwaldron, 2016-05-11:
Looks like serialport is reporting that COM3 has an Arduino attached. To fix this, you can explicitly specify the port to use: 

```js
new five.Board({
  port: "COM4"
});
```

Please close the issue once you've confirmed this addresses your issue—thanks :)

Jawady, 2016-05-11:
Ok thank you i will try this by adding to JF,

Jawady, 2016-05-11:
Hi mate, i have added the above before the blink code on JF, and below error resulted, it not crashing as before which is a good sign. 

Microsoft Windows [Version 10.0.10586]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\Users\Haydar>cd..

C:\Users>cd..

C:\>node-red

Welcome to Node-RED
===================

12 May 09:16:23 - [info] Node-RED version: v0.13.4
12 May 09:16:23 - [info] Node.js  version: v4.4.4
12 May 09:16:23 - [info] Windows_NT 10.0.10586 x64 LE
12 May 09:16:23 - [info] Loading palette nodes
12 May 09:16:30 - [warn] ------------------------------------------
12 May 09:16:30 - [warn] Failed to register 2 node types
12 May 09:16:30 - [warn] Run with -v for details
12 May 09:16:30 - [warn] ------------------------------------------
12 May 09:16:30 - [info] Settings file  : C:\Users\Haydar\AppData\Roaming\npm\node_modules\node-red\settings.js
12 May 09:16:30 - [info] User directory : \Users\Haydar\.node-red
12 May 09:16:30 - [info] Flows file : \Users\Haydar\.node-red\flows_haydar.json
12 May 09:16:30 - [info] Creating new flow file
12 May 09:16:30 - [info] Starting flows
12 May 09:16:30 - [info] Started flows
12 May 09:16:30 - [info] Server now running at http://127.0.0.1:1880/
12 May 09:17:42 - [info] Stopping flows
12 May 09:17:42 - [info] Stopped flows
12 May 09:17:42 - [info] Starting flows
1463008662919 Device(s) Firmata
12 May 09:17:42 - [info] Started flows
1463008662926 Connected Firmata
12 May 09:17:46 - [error] [nodebot:dfc0412d.9ce78] TypeError: Cannot read property 'Board' of undefined

mithunkr, 2016-05-11:
Great. This worked for me.

closed 2016-05-11 by mithunkr

Jawady, 2016-05-11:
where did you add the  code to ?

new five.Board({
  port: "COM4"
});

Jawady, 2016-05-11:
i added it to the JF node

mithunkr, 2016-05-11:
Yes...
new five.Board({
      port: "COM4"
});

Jawady, 2016-05-11:
thanks mate, strange in my case its still not working. the good thing i learned is it work on Mac OS.

mithunkr, 2016-05-11:
Yeah. But to me it doesn't really seem like an issue with respect to Mac or Windows.
It has something to do with the machine itself, 'cause in the windows laptop somehow COM3 was showing up as connected to Arduino, incorrectly; in Mac there's only one port open for that.
If by any reason the Mac also had more than one port open, then this issue might appear there too.

rwaldron, 2016-05-13:
@Jawady can you post your code?

Jawady, 2016-05-13:
hi mate i did on  (#1129)

Jawady, 2016-05-13:
i also ordered Mega board just in case

FredrikHj, 2019-01-31:
> Yes...
> 
> ```
> new five.Board({
>       port: "COM4"
> });
> ```
> instead of just
> `new five.Board();`

I am was able connecting just thank´s your post :)

maverck-, 2020-04-25:
> Looks like serialport is reporting that COM3 has an Arduino attached. To fix this, you can explicitly specify the port to use:
> 
> ```js
> new five.Board({
>   port: "COM4"
> });
> ```
> 
> Please close the issue once you've confirmed this addresses your issue—thanks :)

That's what I needed! Thanks!