rwaldron/johnny-five #1132: How about make johnny-five inside PhoneGap/Cordova?

iszufu, 2016-05-13:
Have you ever tried to build a PhoneGap or Cordova app for mobile with johnny-five to control an arduino borad by bluetooth serial? If you do have, please send me the link for toturial. But if you haven't, can you please think about it and give me some solutions to achieve this? I would like to use blockly from google to build an mobile app for controlling an arduino based car. 
By the way, I know that cylon.js have done this. But I prefer to use johnny-five, cause I am more familiar with it and I think it should achieve this as well.

monteslu, 2016-05-13:
@iszufu its definitely possible.   I did this a while back for bean-io and bean-serial.  Wrote about it here:  http://iceddev.com/blog/beanbots-rise-of-the-swarm/
The code linked in that post is pretty out of date though.

The key is to get a serial type interface such as https://github.com/monteslu/ble-serial  with a custom firmata sketch on a board such as the Arduino 101.

This is somewhat straight forward using Web Bluetooth on new browsers with https://github.com/sandeepmistry/noble/pull/238

For cordova/phonegap though, we either need a cordova-noble-shim + ble-serial or we need a new ble-serial type of interface using the cordova BLE API.  
@don Would better be able to eloborate on the cordova part.

dtex, 2018-01-05:
@iszufu Assuming @monteslu's advice put you on the right path so I'm going to close this issue. Please re-open if necessary.

closed 2018-01-05 by dtex