rwaldron/johnny-five #1133: Trying to use 6 steppers.

drj-io, 2016-05-13:
https://gist.github.com/drj-io/56db4be7e002e763ff5e0c2a5e256f59

I'm using Big Easy Drivers (sparkfun). I'm able to run the above code with steppers 1-4 or 3-5... any combo of 4.

If I engage 5 or 6, the steppers never run.

I'm using a 10A 12v power supply, so there should be enough juice.

soundanalogous, 2016-05-13:
Are you using an Arduino or something like a Raspberry PI?

drj-io, 2016-05-14:
Arduino.

drj-io, 2016-05-14:
here is the hardware config.  In this picture, the steppers aren't hooked up, but I assure you they've all been tested and work individually.  I've been using J5 with steppers for about a year now and I've never run into anything like this.  But I've also never set up more than 3 steppers.

Any ideas would be great thanks!

[image]

soundanalogous, 2016-05-14:
Firmata (which the J5 implementation for Arduino uses under the hood) supports up to 6 stepper motors simultaneously. However I've never tested with 6 since I've never had that many drivers at one time. The issue could lie in the Firmata firmware or the the firmata.js middle-ware. I'll look over the code for both to see if I see any potential issues. May not get to it until later this weekend.

soundanalogous, 2016-05-14:
Have you tried 5 steppers simultaneously, or the most you can run is 4?

soundanalogous, 2016-05-14:
Also, are you using ConfigurableFirmata, AdvancedFirmata or some other variant?

drj-io, 2016-05-14:
I have tried 4 & 5,  In one case I was able to get 5 working.  We are using advancedFirmata.

drj-io, 2016-05-14:
Also, we did change MAX_STEPPERS in the Firmata code, and in steppers.js J5 lib, but that did not help.

soundanalogous, 2016-05-14:
I don't see anything suspect in the code in either AdvancedFirmata or firmata.js. Have you tried attaching the steppers to different pins on the board to see if that makes any difference? This is probably going to require adding Logging into AdvancedFirmata and firmata.js in order to diagnose. Also if you have access to an oscilloscope that may be useful in monitoring power and signals on the motors to see if that is the source.

soundanalogous, 2016-05-14:
Also have you tried controlling the steppers using firmata.js instead of J5? That can help diagnose if the issue could be in J5.

drj-io, 2016-05-16:
@soundanalogous thanks for your ideas.  We've tried different pins, and haven't had any luck with that. We might want to try a different board, though...  Just to be sure.  

We have swapped out different pins/steppers and can only narrow it down to the number of steppers that are in the mix.  And we can get each stepper to function properly.  

I'll checkout https://www.npmjs.com/package/firmata and see what we can reproduce the same issue. I was not aware of this lib, so it might be something we can also use.  

+ @ludicsound is also helping out, so I'm tagging him.

Thanks again!

drj-io, 2016-06-22:
This is still an issue, though I haven't had a chance to try with Firmata.  We'd like to get this working, but for now we're just using 2 arduino's for our application...

drj-io, 2016-09-06:
I considered several work-arounds on this one, but here's what I decided to do.

Each stepper is set individually, so I didn't need the timing in manipulation to be super precise, at least not yet.  We've got some enhancements that will require this, but for now we just want to get our POC working.  

We had thought about connecting to the board and then trying to close it, alternating pins each time, but we determined that is not a feature that will be supported anytime soon - #617 

We've decided to spawn a separate process script for each stepper.  This node spawn simply loads J5, connects to the board, turns 1 stepper to the precise location, and then does a `process.exit(0)`...  The controlling software goes through each stepper until everything is finished.

the command line script looks something like 

`node arduino-standalone --method=stepper --m1=35 --m2=34 --cw=3600 -rpm=180`

This will work for now for our application, but it would be nice if we could support 6 steppers per board, particularly for some of our more real-time features.

I tested the original script again (https://gist.github.com/drj-io/56db4be7e002e763ff5e0c2a5e256f59) with the latest NPM install and a different computer (OSX -> Arduino) and got the same result.  

Thanks!

techniq, 2016-11-10:
I'm trying to use 3 stepper motors, but am struggling to even get 2 running at the same time.  I'm using a Nano with the A4988 driver.

```js
var five = require("johnny-five");
var board = new five.Board({repl:false});

board.on("ready", function() {

  var stepper1 = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: [4, 5]
  });

  var stepper2 = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: [7, 6]
  });

  stepper1.rpm(180).ccw().step(200, function(){
    console.log("done");

    stepper2.rpm(180).ccw().step(200, function() {
	    console.log("done");
  	});

  });
});
```

If I comment out the other stepper, they work individual, but never together.  I'm using ConfigurableFirmata built from http://firmatabuilder.com/ with StepperFirmata selected.  I haven't tried using AdvancedFirmata, but might since 4 have been shown to work with it (although it is now deprecated in favor of ConfigurableFirmata).  I also might try using a Teensy so see if it helps, but was checking to see if there are any known issues/gotchas with more than 1 stepper.  I saw there were some changes [coming](https://github.com/rwaldron/johnny-five/issues/1218#issuecomment-258685642) to Firmata/J5 but didn't see anything that might fix this.

soundanalogous, 2016-11-10:
Please try using AdvancedFirmata. The stepper code in AdvancedFirmata is equivalent to that in ConfigurableFirmata. I'm wondering if something else in ConfigurableFirmata is causing an issue. Please also report the full list of features you selected when using firmatabuilder.

techniq, 2016-11-10:
Thanks @soundanalogous.  I'm also having trouble with a regular dc motor (using the `Motor` class) for the same project, and it appears to be some kind of firmata compatibility issue with my Arduino Nanos (some from Amazon, and some from Banggood experience the same issues).  I'm going to open a separate issue for this (with better explanation) but until I resolve that issue, consider my report null and void :)

techniq, 2016-11-10:
@soundanalogous Btw, part of my DC Motor / PWM issue might steam from using pins 9 and 10 and `StandardFirmata` and my `ConfigurableFirmata` included `Servo` support, which looks to disable PWM on these pins.  I'm pretty sure I was also having issues with pins 3, 5, 6 with PWM, but I was mostly focused on 9 and 10 due to my breadboard layout.  Once I resolve the PWM/DC motor issue, I'll report back with multiple servo support.

soundanalogous, 2016-11-10:
That's correct you can't reliably use a motor on pin 9 when the Servo library is included.

techniq, 2016-11-15:
I commented out a reference to a [Servo](http://johnny-five.io/api/servo/) I had overlooked and also switched to `AdvancedFirmata`.  I'm not sure if switching to AdvancedFirmata was required, but I now had 3 working steppers and 2 DC motors for my project, so I'm happy.  Thanks

soundanalogous, 2016-11-15:
I would still like to try to figure out why ConfigurableFirmata is not working. Can you create build with [firmatabuilder](http://firmatabuilder.com/) and select only the following features and let me know if that works with 3 stepper motors (I'd test myself but only have 2 stepper motors).

- DigitalInputFirmata
- DigitalOutputFirmata
- AnalogInputFirmata
- AnalogOutputFirmata
- I2CFirmata
- StepperFirmata

techniq, 2016-11-15:
Sure. I won't have the hardware back for a few days but I'll try to report
back by the end of the week.

drj-io, 2017-03-10:
Hi all, I wanted to follow up on this.  I haven't tested lately as we're still using the above workaround. Was wondering if anyone had a chance to look at this.  

I can test next week if needed.

dtex, 2017-07-19:
@drj-io I have a PR against firmata.js and configurableFirmata that is a complete rewrite of stepper. I don't have the Johnny-Five part yet, but would you be willing to test against firmata.js? I can port [your gist](https://gist.github.com/drj-io/56db4be7e002e763ff5e0c2a5e256f59) if necessary.

drj-io, 2017-07-19:
@dtex great news, I will see what I can do to port over to firmata.js to test.  Can you link me to the PRs?

dtex, 2017-07-19:
https://github.com/firmata/ConfigurableFirmata/pull/72
https://github.com/firmata/firmata.js/pull/176
and if necessary for reference
https://github.com/firmata/protocol/pull/90

I'm in the middle of changing references from Stepper2Firmata to AccelStepperFirmata. So if you see stepper2 anywhere replace with accelStepperFirmata

dtex, 2018-08-21:
The new stepper class in Johnny-Five is coming very, very soon so let's remove the stale label and close this properly when the time comes.

dtex, 2019-08-16:
Back off Stale Bot, I'm working on it.

closed 2020-08-19 by stale[bot]