rwaldron/johnny-five #1135: ADXL345 accelerometer -  problems reading data

Jeremboo, 2016-05-15:
Hi,

With @tournevis we use an [ADXL345 accelerometer](https://www.amazon.fr/SODIAL-ADXL345-Digital-Acceleration-dinclinaison/dp/B00KBPR3E0?ie=UTF8&creative=22698&creativeASIN=B00KBPR3E0&hvdev=c&hvnetw=g&hvqmt=&linkCode=df0&ref_=asc_df_B00KBPR3E033810549&tag=googshopfr-21) and we discover a largely known problem about the Z-axis value. One of solutions (except buying another model) is to changing the range of the accelerometer like they say into this two topics :

- [Forum Arduino - Problem with adxl345 3-axis accelerometer](http://forum.arduino.cc/index.php?topic=101845.0)

>  The arduino code I am using (Razor_AHRS) initializes the accelerometer for range +-2g. By switching to +-16 g mode (writing 0x0B instead of 0x08 to register 0x31), I was able to get unclipped reading from Z axis.

- [Avrfreaks - ADXL345 Accelerometer Z-axis not working](http://www.avrfreaks.net/forum/adxl345-accelerometer-z-axis-not-working)

> I changed the range to +- 16 g and that seemed to solve the problem.

And it works fine for us.

Now, we pass to Firmata using jhonny-five and we discover the same trouble using the [ADXL345 example](http://johnny-five.io/examples/accelerometer-adxl345/) :

````
  x            :  -0.22
  y            :  -0.53
  z            :  2.04
--------------------------------------
  x            :  -0.18
  y            :  -0.51
  z            :  2.04
```

The Z-axis never change and it's not possible to calibrate the range. 

Then, when we look `johnny-five/accelerometer.js` (exactly [l.129-130](https://github.com/rwaldron/johnny-five/blob/master/lib/accelerometer.js#L129-L130)) we found mistakes into the bit adresses. According to the [i2cdevlib/ADXL345](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/ADXL345) arduino library, good adresses are : 

```
#define ADXL345_ADDRESS_ALT_LOW     0x53 // alt address pin low (GND)
#define ADXL345_ADDRESS_ALT_HIGH    0x1D // alt address pin high (VCC)
....
#define ADXL345_RA_DATA_FORMAT      0x31
#define ADXL345_RA_DATAX0           0x32
#define ADXL345_RA_DATAX1           0x33
#define ADXL345_RA_DATAY0           0x34
#define ADXL345_RA_DATAY1           0x35
#define ADXL345_RA_DATAZ0           0x36
#define ADXL345_RA_DATAZ1           0x37
....
#define ADXL345_RANGE_2G            0b00
#define ADXL345_RANGE_4G            0b01
#define ADXL345_RANGE_8G            0b10
#define ADXL345_RANGE_16G           0b11
...
```
...and into `johnny-five/accelerometer.js` we have : 

```
ADXL345: {
    ADDRESSES: {
      value: [0x53]
    },
    REGISTER: {
      value: {
        POWER: 0x2D,
        RANGE: 0x31,
        READREGISTER: 0x32
      }
    },
```

So the `RANGE` and `READREGISTER` values seems to be wrong. They shift the values completely. This seems to be true because when we are changing adresses, we already have better results : 
```
 RANGE: 0b11,
 READREGISTER: 0x31
```
```
  x            :  -32.36
  y            :  14.12
  z            :  0
--------------------------------------
  x            :  -31.95
  y            :  21.09
  z            :  -1.02
```

But the Z-axis is locked between `0` and `-1.02` and changes into a very little angle. It is not still usable and we think it's because the range value is at  `+-2g`. And when we are adding into our Firmata arduino code the ADXL345 library to calibrate manually the range, the accelerometer works perfectly. 

So, we think there are two problems into the johnny-five support : 

1. The bit adresse `READREGISTER` is wrong.
2. The bit adresse `RANGE` is wrong and in any case never applied.

Hoping to have helped.

rwaldron, 2016-05-16:
This is all very interesting, a bit much to read through, but interesting. According to the datasheet, 0x32 is the correct register to read from, as 0x31 is the `DATA_FORMAT` register

[image]

> Then, when we look johnny-five/accelerometer.js (exactly l.129-130) we found mistakes into the bit adresses. According to the i2cdevlib/ADXL345 arduino library, good adresses are :

> ```
> #define ADXL345_ADDRESS_ALT_LOW     0x53 // alt address pin low (GND)
> #define ADXL345_ADDRESS_ALT_HIGH    0x1D // alt address pin high (VCC)
> ....
> #define ADXL345_RA_DATA_FORMAT      0x31
> #define ADXL345_RA_DATAX0           0x32
> ```

How is that Johnny-Five is wrong, and this is right, when they literally have the same exact number? 0x32 is the same number in C as it is in JavaScript. In Johnny-Five, `ADXL345_RA_DATAX0` is generically called `READREGISTER`. 

`RANGE: 0x31` is _definitely_ wrong.

ADXL345 controller is one of the oldest controllers here and clearly needs some love. I'll take a look as soon as I have time.

>  And when we are adding into our Firmata arduino code the ADXL345 library to calibrate manually the range, the accelerometer works perfectly.

I don't recommend changing anything at the Firmata level.

rwaldron, 2016-05-16:
Update: the use of `RANGE: 0x31` is correct, I just looked past the bit field table earlier. Range is set in bits 0 and 1 of the `DATA_FORMAT` register and we're just calling that register "RANGE" because... I don't remember why...

What needs to happen is that this controller needs to allow a `range` property in the constructor options: 

```js
Accelerometer({
  controller: "ADXL345", 
  range: 2
});
```

Where range can be one of: 2, 4, 8, 16

[image]

And then this line: 

```js
// Set range (this is 2G range, should be user defined?)
this.io.i2cWrite(address, this.REGISTER.RANGE, 8);
```

Needs to be something like...

```js
var range = ({ 2: 0, 4: 1, 8: 2, 16: 3 })[opts.range || 2];
this.io.i2cWrite(address, this.REGISTER.RANGE, range);
```

rwaldron, 2016-05-17:
@Jeremboo can you test those changes? `npm install rwaldron/johnny-five` to get the changes to try out. 

```js
new five.Accelerometer({
  controller: "ADXL345", 
  range: 2 // 2, 4, 8, 16
});
```

closed 2016-05-17 by rwaldron

Jeremboo, 2016-05-17:
Hi @rwaldron !

Thank you very much for your reactivity and your advices ! With this changes, we will not need to using our dirty tricks for our project ;)

I tried your code and the `range` is updated correctly. So your support works very well !

However, the Z-axis can be read only with the `range: 16`. The quality of my accelerometer can be interrogated...

-----

_Render with `range: 16`:_
```
  x            :  0.59
  y            :  2.5
  z            :  67.89
--------------------------------------
  x            :  0.21
  y            :  1.94
  z            :  69.77
```

Thank you again and sorry for my long and misleading assumption.

rim0621, 2018-09-07:
HI, i using ADXL345 and +-16g.
i know the ranges are -256 ~ +255(10bits).
but i have got over 255 and under -256... 
what is wrong? can you help me? thank you.
'''
  after reading data...
			gxt = HLdata(TempChar[5], TempChar[4]); 
			gyt = HLdata(TempChar[7], TempChar[6]);
			gzt = HLdata(TempChar[9], TempChar[8]);
'''