rwaldron/johnny-five #1137: Adding Tolerance on temperature changes for "thermometer.js" library

alfredoborrero, 2016-05-15:
Line 964 in "thermometer.js" is emitting changes in the temperature whether the change is small or big. 

I would like to see a new variable added the the library called TOLERANCE, then the user can specify what type of change in temperature needs to be emitted to the connected clients. Sometimes the variable "**last**" has a value of 24.14 ºC and "**this.celsius**" has a value of 24.15 ºC ; to be honest with you, that is pretty much the same temperature (not worth emitting to the client) unless I specify a small tolerance, eg:  TOLERANCE = 0.01 

Lets say I am using sensors for Home Automation.  For my particular uses, there is no difference between 24.14 ºC and 24.15 ºC because I will show 24 ºC and  I can convert the number to an integer.  However, the node will be emitting changes that are not needed, and using resources that can be used for other sensors. 
 
I think if we add a TOLERANCE variable, the library will be more flexible for users. I had to modify the library in order to introduce tolerance, then reduced the unnecessary amount of messages emitted from the server to the connected clients.  

```
964)    if (this.celsius !== last) {
965)      last = this.celsius;
966)      this.emit("change", data);
967)    }
```

This tolerance can be specified by the user like this: 
`var temp = new five.Thermometer({ pin: 3, controller: "DS18B20", tol: 0.5  });`
as another field in the opts object

Of course this can be added to any other library library like "**lights.js**"

Let me know your thoughts ...

Regards,

Alfredo Borrero

rwaldron, 2016-06-22:
Hi! Sorry for the delay in response—we've had a crazy few weeks getting ready for some exciting news ;)

I like this idea, and there is already a "threshold" that has this same behavior, but it's not implemented in most component specific classes because early on it wasn't clear to me which value the threshold/tolerance should be applied to. Maybe that doesn't matter? Maybe we just apply it to the "default" value (in the `Thermometer` case, that's `celsius`)

alfredoborrero, 2016-06-23:
Exiting News?I like that .... Are those news related to Johnny Five?

The "celsius" that you are referring to is the one located at line 969 right? 
if (this.celsius !== last) {      last = this.celsius;      this.emit("change", data);}
This one only emit value if the variable "last" is different from "this.celsius" and that scenario can happen for any change, so we can control how big or how small the change is. Are we expecting a future change on this file? I think would be nice then all the files would have a similar structure.  
Regards,
Alfredo Borrero
Date: Wed, 22 Jun 2016 10:58:19 -0700

rwaldron, 2016-08-23:
Sorry I never replied... The exciting news: https://www.sparkfun.com/products/13847 and https://twitter.com/rwaldron/status/766004460506255364 ;)

frxnz, 2016-08-24:
I got busy and sort of dropped the ball on this, but #1078 is probably relevant to this discussion.

dtex, 2018-01-11:
Hi @alfredoborrero ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for tolerance to thermometer.

closed 2018-01-11 by dtex