rwaldron/johnny-five #1138: Raspberry Pi 2 + Raspbian Jessie (2016-05-10) + J5

cpranzl, 2016-05-19:
I'm quite not sure if this is the right place but I'm running in some problems
triying to install J5 in Raspian. So I wrote down my findings, maybe they are
helpfull.

I've downloaded the latest version of Raspbian (2016-05-10), wrote the image
to a 8GB sdcard and put it in my Raspberry Pi 2. Then I tried to follow the
tutorial "Getting started with Raspberry Pi node red and IoT" on GitHub.

https://github.com/rwaldron/johnny-five/wiki/Getting-started-with-Raspberry-Pi-node-red-and-IoT

I followed the tutorial and installed npm with apt-get

    cpranzl@loki:~ $ sudo apt-get update
    cpranzl@loki:~ $ sudo apt-cache show npm | grep Version
    Version: 1.4.21+ds-2
    cpranzl@loki:~ $ sudo apt-get install npm
    cpranzl@loki:~ $ npm --version
    1.4.21

This will correctly lead to npm 1.4.21 but I wanted to use version 2.15.1 of npm so I used

    cpranzl@loki:~ $ sudo npm install -g npm@2.15.1
    /usr/bin/npm -> /usr/lib/node_modules/npm/bin/npm-cli.js
    npm@2.15.1 /usr/lib/node_modules/npm
    cpranzl@loki:~ $ npm --version
    2.15.1

Next step is to check on the installed NodeJS version.

    cpranzl@loki:~ $ nodejs --version
    v.0.10.29

Finally I was able to start installing the johnny-five and raspi-io packages

    cpranzl@loki:~ $ npm i johnny-five raspi-io
    ...
    
    ...
    Disabling serial port login at boot time, you may be asked for your password
    Checking if serial port login is enabled
    Serial port login does not appear to be enabled, skipping
    johnny-five@0.9.49 node_modules/johnny-five
    ├── ease-component@1.0.0
    ├── lodash.debounce@4.0.6
    ├── color-convert@1.2.2
    ├── browser-serialport@2.0.3
    ├── temporal@0.5.0
    ├── lodash.clonedeep@4.3.2 (lodash._baseclone@4.5.7)
    ├── chalk@1.1.3 (escape-string-regexp@1.0.5, supports-color@2.0.0, ansi-styles@2.2.1, strip-ansi@3.0.1, has-ansi@2.0.0)
    ├── nanotimer@0.3.10
    ├── firmata@0.12.0 (es6-shim@0.33.13)
    ├── es6-shim@0.35.1
    └── serialport@3.1.2 (bindings@1.2.1, es6-promise@3.2.1, commander@2.9.0, debug@2.2.0, object.assign@4.0.3, nan@2.3.3)

    raspi-io@6.0.1 node_modules/raspi-io
    ├── raspi-board@3.0.0
    ├── raspi-led@1.3.2 (raspi-peripheral@1.5.2)
    ├── raspi@2.0.0 (raspi-wiringpi@1.0.6, nan@2.3.3)
    ├── raspi-pwm@3.0.0 (raspi-peripheral@1.5.2, nan@2.3.3)
    ├── raspi-gpio@2.2.2 (raspi-peripheral@1.5.2, nan@2.3.3)
    ├── raspi-i2c@3.0.0 (ini-builder@1.0.3, raspi-peripheral@1.5.2, i2c-bus@1.0.3)
    └── raspi-serial@1.3.1 (raspi-peripheral@1.5.2, serialport@2.1.2)

Controlling if all went in the right place

    cpranzl@loki:~ $ ls -al
    total 52
    drwxr-xr-x  5 cpranzl cpranzl 4096 May 19 06:17 .
    drwxr-xr-x  4 root    root    4096 May 17 19:57 ..
    -rw-------  1 cpranzl cpranzl  596 May 19 06:12 .bash_history
    -rw-r--r--  1 cpranzl cpranzl  220 May 17 19:57 .bash_logout
    -rw-r--r--  1 cpranzl cpranzl 3697 May 17 20:03 .bashrc
    drwxr-xr-x  3 cpranzl cpranzl 4096 May 17 20:15 .node-gyp
    drwxr-xr-x  4 cpranzl cpranzl 4096 May 17 20:15 node_modules
    drwxr-xr-x 52 cpranzl cpranzl 4096 May 17 20:19 .npm
    -rw-r--r--  1 cpranzl cpranzl  675 May 17 19:57 .profile
    -rw-r--r--  1 cpranzl cpranzl  795 May 17 20:04 .tmux.conf
    cpranzl@loki:~ $ cd node_modules
    cpranzl@loki:~/node_modules $ ls -al
    total 16
    drwxr-xr-x 4 cpranzl cpranzl 4096 May 17 20:15 .
    drwxr-xr-x 5 cpranzl cpranzl 4096 May 19 06:17 ..
    drwxr-xr-x 4 cpranzl cpranzl 4096 May 17 20:15 johnny-five
    drwxr-xr-x 4 cpranzl cpranzl 4096 May 17 20:15 raspi-io

Looks good, so reverting to my home-directory I started a new file and wrote the first JScript

    cpranzl@loki:~/node_modules $ cd ..
    cpranzl@loki:~ $ nano raspi-io.js

    00 var five = require("johnny-five");
    01 var Raspi = require("raspi-io");
    02 var board = new five.Board({
    03     io: new Raspi()
    04 });
    05
    06 board.on("ready", function() {
    07     var led = new five.Led("P1-13");
    08     led.blink();
    09 });

Then I started the script

    cpranzl@loki:~ $ node raspi-io.js

    /home/cpranzl/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96
    var devices = Symbol('devices');
              ^
    ReferenceError: Symbol is not defined
    at Object.<anonymous> (/home/cpranzl/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/cpranzl/node_modules/raspi-io/lib/index.js:21:17)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)

Where is my error or did i forgot something? Any suggestions?

fivdi, 2016-05-19:
According to the Raspi-io readme, Raspi-io v6.0.1 requires Node.js v0.12 or newer. See [System Requirements](https://github.com/nebrius/raspi-io#system-requirements). I'm not sure of the requirements from a Node-RED perspective.

nebrius, 2016-05-19:
Yep, that error is why Node 0.12 or newer is required. I recommend uninstalling Node from apt-get and following the instructions in the Raspi IO wiki for installing a newer version of Node.

rwaldron, 2016-05-19:
Thanks @fivdi & @nebrius!

closed 2016-05-23 by cpranzl