rwaldron/johnny-five #1139: using Parallax Propeller

martinschuberteu, 2016-05-23:
Hi,
I have a Propeller Hat (https://shop.pimoroni.de/products/propeller-hat) that I'd like to use with johnny-five. Would this require an expander or an I2C backpack, and are there any documents available on how to create one of these? Thank you!

reconbot, 2016-08-01:
It looks like it can be controlled directly via i2c according to it's open source documentation. The [raspi-io](https://github.com/nebrius/raspi-io) docs will show you how to read and write via i2c. The [code for the propeller-hat](https://github.com/pimoroni/propeller-hat/blob/master/software/python/library/p1/io.py) has examples of the device via i2c but as far as I know johnny-five does not yet have an expander class for this particular device.

I would take a look at some of the [existing expanders](http://johnny-five.io/api/expander/) for an example of you might write your own. I think it would be a cool plugin that would help a lot of people.

rwaldron, 2016-08-01:
@reconbot if we can get one of these, we can get support added.

dtex, 2018-01-11:
Hi @martinschuberteu, 

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the Parallax Propeller there.

closed 2018-01-11 by dtex