rwaldron/johnny-five #1140: Arduino Yun & i/o over Wi-fi

gokhandilek, 2016-05-26:
I was wondering if it would be possible to use johny-five api to connect to arduiono yun over wi-fi instead of USB connection?

soundanalogous, 2016-05-26:
You can use the Yun over Ethernet in addition to USB, but I'm not sure about WiFi. I believe you can also run johnny-five direclty on the Yun so you don't even need to tether it.

gokhandilek, 2016-05-28:
I am trying to run john-five. This is the error messsage I am getting:
`node: '/usr/lib/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node' is not an ELF file
/usr/lib/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: File not found
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at bindings (/usr/lib/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/usr/lib/node_modules/johnny-five/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
`

Would you know why this maybe the problem?
I have the johny-five copied and pasted under node_modules. (I could not do nmp install johnny-five as the installation stalls).

rwaldron, 2016-05-30:
The error is coming from node-serialport, which appears to be missing its compiled binary (likely needs to have a cross compiled version in place)

JOSECCZ, 2016-05-30:
I had a similar problem... maybe my solution at
https://github.com/rwaldron/johnny-five/issues/1141
might work for you

gokhandilek, 2017-04-09:
I managed to fix this issue by using the firmatabuilder + yun  + ethernet.
Then join the yun to the local wifi network.

closed 2017-04-09 by gokhandilek