rwaldron/johnny-five #1141: Error using Johnny-five in xubuntu Linux

JOSECCZ, 2016-05-29:
I just purchased your "Learning JavaScript Robotics" book and I am getting the next error at your first example"

Error: /home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)

Can you please help me? I followed all the steps mentioned in the book.
I am using

Xubuntu Linux 14.04LTS  64bit

joseccz@XEnterprise:~/workspaceNODEJS/ArduinoLights$ nodejs -v
v0.10.25

Thanks
Jose

JOSECCZ, 2016-05-29:
I found a solution....
after several hours working on it

First, I uninstalled any package or root level version of nodejs and made the installation using NVM as explained at
https://www.digitalocean.com/community/tutorials/how-to-install-node-js-on-an-ubuntu-14-04-server
 

However, I had to correct NPM as I had the error:

npm WARN enoent ENOENT: no such file or directory, open '/home/joseccz/package.json'
npm WARN joseccz No description
npm WARN joseccz No repository field.
npm WARN joseccz No README data
npm WARN joseccz No license field.

Following this reference: https://www.npmjs.com/package/npm-autoinit

joseccz@XEnterprise:~$ npm install -g npm-autoinit
npm config set onload-script npm-autoinit/autoinit

As the book mentioned I installed johnny-five inside the folder:

joseccz@XEnterprise:~/workspaceNODEJS/ArduinoLights$ npm install johnny-five  (executed two times)

(ArduinoLights is my project folder)

But I noticed that somehow johnny-five was installed in a personal folder /home/joseccz/node_modules/. I deleted the johnny-five from my project folder. Everything is working now (excepting Eclipse which can't find the nodejs executable)

Thanks
Jose

On Sunday, May 29, 2016 6:30 AM, Packt Publishing <errataall@packtpub.com> wrote:

Hi Jose,
Thank you for contacting PacktPub Customer care.
We are sorry to learn about the trouble you experienced following the example files of the book “Learning JavaScript Robotics”.
I have forwarded your query to the author for review and will get back to you on this shortly.
Please mention the case # 164233 in your future correspondence with regards to this query.
 
Regards,
Marimuthu Pandian           
Customer Relations Executive    
Packt Publishing | www.packtpub.com  

On 29/05/2016, 06:15am, wrote:
I just purchased your "Learning JavaScript Robotics" book and I am getting the next error at your first example"
 
Error: /home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
    at Object. (/home/joseccz/workspaceNODEJS/ArduinoLights/node_modules/johnny-five/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
 
Can you please help me? I followed all the steps mentioned in the book.
I am using
 
Xubuntu Linux 14.04LTS  64bit
 
joseccz@XEnterprise:~/workspaceNODEJS/ArduinoLights$ nodejs -v
v0.10.25
 
Thanks
Jose

closed 2016-05-29 by JOSECCZ

rwaldron, 2016-05-30:
@JOSECCZ Thanks for filing an issue and for the follow up/resolution. This issue is actually occurring in https://github.com/EmergingTechnologyAdvisors/node-serialport