rwaldron/johnny-five #1145: Pololu - Dual MC33926 Motor Driver Carrier

martinschuberteu, 2016-06-01:
Hi,
I got a MC33926 connected to an PCA9685 expander and am really struggling to get a motor running. I've tried both 2 (pwm, dir) and 3 (pwm, dir, cdir) pin configurations with different pins and invertPWM:true. I actually only want to use two PWMs that are connected to IN1 and IN2 and according to the truth table in the documentation [https://www.pololu.com/file/download/MC33926.pdf?file_id=0J233](url) they should be set to:
                          IN1            IN2
Forward             High          Low
Reverse             Low           High

(and both low for stop I guess). How can I do this with the motor class? Is there something like a truth table for pwm, dir or pwm, dir, cdir for the motor class? Thank you!

dtex, 2016-06-01:
Hi there,

So you're running Arduino->PCA9685->MC33926? I've never seen that configuration before. First question, how many motors are you trying to control? You might not need the PCA9685.

martinschuberteu, 2016-06-01:
Hi Donovan,
Raspberry PI 3 -> (via I2C) PCA9685 -> (PWM) MC33926 actually. It works perfectly find for e.g. servos and shouldn't be any problem at all. I'm using the PCA9685 because I need more PWM ports than the Raspberry PI has and  I just need to know how to use the motor class with two PWMs to control a motor connected to the MC33926. Is there something like a table that shows the output on pwm, dir or pwm, dir, cdir for the different states in the motor class (stop, forward, reverse, ...)?

dtex, 2016-06-01:
Every motor controller is a snowflake and may or may not have different jumpers available/enabled so a truth table would be hard. This is my first 4-pin input motor controller but I believe we can put a jumper on one or two of those pins and get us down to a configuration that emulates a two or three pin controller.

What pins do you have jumpers on right now?

martinschuberteu, 2016-06-01:
I've tried these to configurations:

#1 jumpers on EN, D1 and D2 and PWM on IN1 and IN2 for pwm and dir
#2 jumpers on EN and D1 and PWM on D2, IN1 and IN2 for pwm, dir and cdir

For #1 (which I would prefer) the motor was running when it should have been stopped, and was stopped when it should run. Reverse actually worked and after the reverse a stop also worked. As I said I've tried different combinations on the pins and also invertPWM:tru, but it never did what it should do. So I was hoping to get some information on the output of the pins for stop, forward, reverse to try and map it with the truth table in the MC33926 documentation.

martinschuberteu, 2016-06-01:
FYI: I was also using this forum entry as reference: [http://letsmakerobots.com/node/39173](url)

dtex, 2016-06-01:
I think you are totally on the right track with #1, invertPWM: true and those jumper settings. The behavior you describe is exactly what I would expect if invertPWM were not set to true. Forgive me for asking again, but can I see you opts object? I confess I'm just looking for a typo.

dtex, 2016-06-01:
Also, I can see it working with the jumpers on D1 and D2 removed.

martinschuberteu, 2016-06-01:
I've removed the jumpers on D1 and D2 and the result was that the motor was stopped all the time. I've also tried it again with #1 and invertPWM: true and here's what I get on OUT1 and OUT2:

initial:             -10.8V
forward 128:    -5.3V
stop:              -10.8V
reverse:          10.8V
stop:              0V
forward 255:  0V  

Also here's the relevant code:

` 
var pwmExpander = new five.Expander({
    controller: "PCA9685"
  });

  var pwmBoard = new five.Board.Virtual({
    io: pwmExpander
  });

  var leftMotor = new five.Motor({
    pins: {
      dir: 8,
      pwm: 9,
      invertPWM: true
    },
    board: pwmBoard
  });

  // motor test
  console.log("motor forward 128");
  leftMotor.forward(128);
  sleep.sleep(5);
  console.log("motor stop");
  leftMotor.stop();
  sleep.sleep(5);
  console.log("motor reverse");
  leftMotor.reverse(255);
  sleep.sleep(5);
  console.log("motor stop")
  leftMotor.stop();
  sleep.sleep(5);
  console.log("motor forward 255");
  leftMotor.forward(255);
`

martinschuberteu, 2016-06-01:
I've just tested it without invertPWM and am getting the same values actually.

dtex, 2016-06-01:
Haha, okay I'm gonna have to draw myself a picture. 

I'll get back to you with some thoughts by tomorrow. Let me know if you figure it out before then.

martinschuberteu, 2016-06-01:
Thank you so much, your help is really appreciated!

dtex, 2016-06-02:
I think I understand it now. Have you tried it with the above code and Jumper on D1 but no jumper on D2? If it still doesn't work I'm just going to order one.

martinschuberteu, 2016-06-02:
I've just tried that and got nothing, but then I tried it again with the following configuration, and it worked!

dir:     IN1
cdir:   IN2
pwm: D2

I'm pretty sure I've tried that before, but who cares if it works now :) Thank you so much!

martinschuberteu, 2016-06-02:
One last question if I may: I' also considering using the feedback port to current, but it would have to be  on a different virtual board. Does johnny-five support having the ports on different boards?

dtex, 2016-06-02:
I'm glad you got it working... I still wonder if it couldn't be done with just two pins...

This was such a brain bender I think we should add an example in case anyone else is having similar troubles. Would you like to be a Johnny-Five contributor? This is a relatively easy first contribution (the hardest part is the Fritzing diagram) and I'll help you with the process.

Yes. The current object is just passed to sensor as sensor opts. It can be on a different virtual board.

martinschuberteu, 2016-06-02:
Sure, I'd love too. I've actually downloaded Fritzing but couldn't find the shapes needed. I'll have a look if I can get the somewhere and then create a new example for Raspberry PI + MC33926  in the motor wiki? 

What would the syntax look like to assign dedicated pins to dedicated boards?
`
  var leftMotor = new five.Motor({
    pins: {
      dir: 8,
      cdir: 9,
      pwm: 10
    },
    board: pwmBoard
  });
`

dtex, 2016-06-02:
You can't split one motor's drive pins between boards. It works for current sensing because current instantiates a sensor object that can take a different board property.

I don't know that I would use a Pi in the example since the real pickle here was the MC33926 jumper and pin configuration. That could be for anybody/any platform. I'd just show it wired to an Uno since that's the most basic option. Oh hey, you could totally do both!

martinschuberteu, 2016-06-02:
Sure, I'll have a look into both :) Thank you again!

martinschuberteu, 2016-06-02:
I've already created /docs/breadboard/motor-Pololu-Dual-MC33926.fzz and /docs/breadboard/motor-Pololu-Dual-MC33926.png and will test this with an Arduino UNO on the weekend and then create /docs/motor-Pololu-Dual-MC33926.md and /eg/motor-Pololu-Dual-MC33926.js and check everything in. Is that it or am I missing something?

[image]

dtex, 2016-06-03:
That's all the hard parts. There are a couple of other small things that you have to do:

Add entries for each example to https://github.com/rwaldron/johnny-five/blob/master/tpl/programs.json#L392

Run ```grunt examples``` That will generate the markdown files and update the readme. 

This process also updates the http://johnny-five.io website

rwaldron, 2016-06-21:
@martinschuberteu how is this coming along? Anything we can iron out together?

martinschuberteu, 2016-06-21:
I have it on my todo list, but didn't get to it yet. I'm hoping to get this done on the weekend though!

rwaldron, 2016-06-21:
@martinschuberteu ok, awesome :)

closed 2019-02-04 by stale[bot]