rwaldron/johnny-five #1147: Unable to get johnny-five nodes working in Node-red

timchoo, 2016-06-03:
I have followed the instructions in the wiki at https://github.com/rwaldron/johnny-five/wiki/Getting-started-with-Raspberry-Pi-node-red-and-IoT to the letter. However, when I go to node-red and deploy a flow with a Johnny-five node it gives me an error "Cannot find module... /build/Release/addon". What am I missing?

rwaldron, 2016-06-03:
Thanks for the report! 

Let's ping @monteslu and @nebrius, as they each know more about these two pieces than I do

monteslu, 2016-06-03:
Was able to reproduce on a node 4.4.5 on a pi3 as well.

It's this issue:  https://github.com/nebrius/raspi-io/issues/40

@nebrius ,  It's been fixed in raspi but I think we need a new release of raspi-io to use it.

timchoo, 2016-06-06:
I managed to get the Johnny-five nodes to work by using --unsafe-perm when installing node-red-contrib-gpio and rapsi-io

timchoo, 2016-06-14:
One side effect is that the Rpi.gpio (python based) nodes no longer work - at least the input node does not. (It will read the initial value and then say its "stopped") I'm guessing Raspi-io is grabbing the io pins and preventing python-rpi.gpio from getting them?

monteslu, 2016-06-14:
@timchoo so the main thing we need to do is make sure raspi-io is installed as the pi user but run as root.

in the case of node-red the root folder will have a `.node-red/node_modules`.  You'll need to copy your raspi-io folder from the pi user into there and run node-red as root.

rwaldron, 2016-06-21:
How's this coming along?

timchoo, 2016-06-21:
After some trial and error, I've documented the series of steps to get Johnny-5 nodes working in Node-red. The steps are found [here](https://timchooblog.wordpress.com/2016/06/14/configuring-the-johnny-five-robotics-library-to-work-in-node-red/). The Python based Rpi.gpio nodes are broken, but they are kinda redundant anyway since the J5 gpio nodes are so much faster. I can live with that.

rwaldron, 2016-06-21:
@timchoo cool, thanks for putting the time into this :) We should link to your article from johnny-five.io. Can you write up a short summary and add it here: https://github.com/j5js/johnny-five.io/tree/master/src/news and then an entry here:  https://github.com/j5js/johnny-five.io/blob/master/src/authors.json

timchoo, 2016-06-30:
Hi

I created the document and submitted a pull request. 

Tim

rwaldron, 2016-08-23:
@timchoo sorry about the delay there, I completely missed the github notification. I've just landed your content and it will be rolled out in the next deployment—thanks again!

closed 2016-08-23 by rwaldron