rwaldron/johnny-five #1148: Reconnection : How can I reallocated the board and pins to make work properly ?

Jeremboo, 2016-06-04:
**Board:** Particle Photon
**Shield:** none
**Hardware:** Motor [Micro buzzer](http://www.ebay.com/itm/121720786154?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT)
**Johnny-Five Version:** 0.9.52
**Node version:** 5.0.0
**OS:** OSX 10.11.3
**Expected result:** The buzzers work even if I reconnect my photon
**Current result:** On reconnection, the buzzers don't work and I have this log : `Component pin: D0 is already in use`.

---

Hi,

I have two micro buzzers connected at my Particle photon board on pins `D0` and `D1` powered by a battery.

When I turn on my board, she connects to my node.js server and I can initiate my Photon and my two buzzers like this : 

```
var leftMotor = { five: false, speed: 0 };
var rightMotor = { five: false, speed: 0 };
var board = new five.Board({
  io: new Particle({
    host: this.IPAddress,
    port: 48879,
  })
});

_initMotor(leftMotor, 'D0');
_initMotor(rightMotor, 'D1');

function _initMotor(motor, pin) {
  motor.five = new five.Motor({ pin });

  motor.five.on('start', () => {
    console.log('start', Date.now());
  });

  motor.five.on('stop', () => {
    console.log('stop', Date.now());
  });

  motor.five.start(0);
}
```
See the complete code here [Bracelet.js](https://github.com/ricochetGobz/ricochet_server/blob/master/app/components/Bracelet.js).

And It works well. I can changing the speed of both buzzer when i want.

## My Problem :

**But now**, I would try to predict the case where my disconnected Photon tries to reconnect (with the same IP). Because I can not restart my node.js server used for other things at the same time.

And in each test, I have the same error and the motors don't reply.

```
 -> bracelet connected at 192.168.1.2
1465033754482 Device(s) particle-io
1465033754484 Connected particle-io
1465033754488 Repl Initialized
>> init motors
1465033754530 Component pin: D0 is already in use
1465033754531 Component pin: D1 is already in use
```

So my question is : **How can I manage reconnection or reallocated the pins to make work the buzzers properly ?**

## Already tried:

- Detect if the IP is already used and : 
  - Do not reconnect the board and motors.  👎*nothing is happening*
  - Do not reconnect the board, just recreate motor `new five.Motor({ pin })`.  👎*log error `Component pin: D0 is already in use` && nothing is happening*
  -  Reconnect the board and : 
    - Nothing else.  👎 *nothing is happening*
    - recreate motor `new five.Motor({ pin })` 👎*log error `Component pin: D0 is already in use` && nothing is happening*
    - `.stop()` and `.start(255)` motor. 👎*nothing is happening*
    - using pins manually  👎*log error `Component pin: D0 is already in use` && nothing is happening*
 ```
        var pin = new five.Pin('D0');
        pin.mode = five.Pin.OUTPUT;
        pin.mode = five.Pin.INPUT;
 ```   
- - delete all variables before recreate board and motors. 👎*log error `Component pin: D0 is already in use` && nothing is happening*

 ```
         board = null;
         leftMotor.five.stop();
         rightMotor.five.stop();
         leftMotor.five = null;
         rightMotor.five = null;

         // or

         delete board;
         delete rightMotor;
         delete leftMotor;
 ```     

--- 

Thanks for reading. My question seem basic and I am sorry to create an issue but I did not found informations about that.

rwaldron, 2016-06-06:
>  nothing is happening

Well, my first thought is that this: 

```js
var leftMotor = { five: false, speed: 0 };
var rightMotor = { five: false, speed: 0 };
var board = new five.Board({
  io: new Particle({
    host: this.IPAddress,
    port: 48879,
  })
});

_initMotor(leftMotor, 'D0');
_initMotor(rightMotor, 'D1');

function _initMotor(motor, pin) {
  motor.five = new five.Motor({ pin });

  motor.five.on('start', () => {
    console.log('start', Date.now());
  });

  motor.five.on('stop', () => {
    console.log('stop', Date.now());
  });

  motor.five.start(0);
}
```

...isn't a valid Johnny-Five program.

Those motors can't be created until after the board and the program have finished initializing: 

```js
var leftMotor = { five: false, speed: 0 };
var rightMotor = { five: false, speed: 0 };
var board = new five.Board({
  io: new Particle({
    host: this.IPAddress,
    port: 48879,
  })
});

_initMotor(leftMotor, 'D0');
_initMotor(rightMotor, 'D1');

function _initMotor(motor, pin) {
  motor.five = new five.Motor({ pin });

  motor.five.on('start', () => {
    console.log('start', Date.now());
  });

  motor.five.on('stop', () => {
    console.log('stop', Date.now());
  });

  motor.five.start(0);
}

board.on("ready", function() {
  _initMotor(leftMotor, 'D0');
  _initMotor(rightMotor, 'D1');
});
```

rwaldron, 2016-06-06:
Ok, I see that _is_ in your full program. I'll need some time to take a closer look.

rwaldron, 2016-06-06:
> So my question is : How can I manage reconnection or reallocated the pins to make work the buzzers properly ?

@BrianGenisio @monteslu what do you think? I haven't encountered such a thing and it's not in Johnny-Five's scope to be concerned with reconnection, etc. I'm thinking that the node.js program should be managed as a child process that restarts when the remote end disconnects?

BrianGenisio, 2016-06-08:
Hmmmm... Yeah, I see disconnects happen, and I agree it is frustrating.  I haven't spent any time trying to detect and recover from it.  I just kill the app and re-start.  

But, it does seem like the [Particle-IO](https://github.com/rwaldron/particle-io) plugin can be smarter.  If it detects a disconnection, it can re-connect and Johnny-Five can be none-the-wiser.  Seems like that might be the best approach?  Perhaps VoodooSpark needs to be updated as well.

mwittig, 2016-06-08:
Perhaps off-topic and pedantic: I think the corrected code example above is still wrong. The two ```_initMotor``` calls before ```function _initMotor``` need to be removed.

Jeremboo, 2016-06-13:
Hi all, sorry for my late answer.

> Ok, I see that is in your full program. I'll need some time to take a closer look. Sorry, long day :\

No problem @rwaldron. If I had make a good example without mistake, there would not been troubles.

> But, it does seem like the Particle-IO plugin can be smarter. 

@BrianGenisio particle-IO, if I'm not mistaken, has not a method for disconnect and seem not recognize a lost connection or a disconnecting message. Maybe should I create an issue on Particle-IO about that for more visibility ? 

@mwittig the corrected code example is not perfect. But in any case it was just an example. My true code is into a es2015 class [here](https://github.com/ricochetGobz/ricochet_server/blob/master/app/components/Bracelet.js) probably with errors too ;).

BrianGenisio, 2016-06-13:
@Jeremboo You are correct.  [particle-io](https://github.com/rwaldron/particle-io/blob/master/lib/particle.js) doesn't try to reconnect or anything.  It isn't terribly straight-forward, since some state needs to be maintained so that it can re-ask for input streams.  But it would make the story much more robust if we were able to come up with a way to do this, which would be nice.  It wouldn't hurt to create an issue in particle-io so we can discuss it more there.  Especially since it isn't J5's responsibility to do this.

closed 2018-08-28 by stale[bot]