rwaldron/johnny-five #1149: arduino board via i2c.

overflowz, 2016-06-05:
Hi, I have a question. Let's say, I have following:
RaspPI (where johnny-five is installed)
Arduino 1 (connected with Serial Port to RaspPI)
Arduino 2 (connected with i2c to Arduino 1).

Both of them are using StandardFirmataPlus sketch.

Question is, how can I control Arduino 2 board from RaspPI :question: 

Regards.

soundanalogous, 2016-06-05:
You cannot use I2C for this. Firmata works over Serial, Ethernet, WiFi and BLE but not over I2C.

overflowz, 2016-06-05:
Hmm, in case of that, is there any way to power up multiple boards and share information with each other? Or do I have to write custom sketch which will respond to i2c commands from firmata protocol?

monteslu, 2016-06-05:
You can plug both in directly to the pi and create a board from each.

overflowz, 2016-06-05:
What if I'll need more than 2 boards? Like 20 or 30 boards together, I'll be limited for ports I guess.

soundanalogous, 2016-06-05:
you can create a custom communication protocol over I2C and use that over Firmata

dtex, 2016-06-05:
Butting in... Do you really need 20-30 boards? If you're trying to control a ton of devices and are just looking for extra pinouts there are expanders that you can chain over I2C to give ~1000 pins.

overflowz, 2016-06-05:
Actually not, maximum I'll need will be like 10 boards (just splitting up for modules, each of them will do different tasks). More like I need more pins rather than different boards (but not shifting ones!).

dtex, 2018-01-05:
It sounds like a conclusion was reached and we just forgot to close the issue. Feel free to re-open if necessary.

closed 2018-01-05 by dtex