rwaldron/johnny-five #1150: Add NW.JS Tutorial to Documentation

GabrielTK, 2016-06-06:
Good Evening,
Sorry, could anyone please add a tutorial on how to use johnny-five with NW.JS?? I Found [this, ](http://tangiblejs.com/posts/nw-js-johnny-five-arduino-wicked-trio) but it doesn't explain how to use it on multipage environments.
Thank you,
Gabriel

noopkat, 2016-06-22:
Hi @GabrielTK !

Could you please clarify "how to use it with multiple pages"? Perhaps with a specific example. I want to understand the difference in use cases against the first article you linked so we can help answer in the right way 😄

GabrielTK, 2016-06-22:
Yes, is exactly that. I'm going to check it there.

noopkat, 2016-06-26:
@GabrielTK hi, sorry for the delay in this response.

I played with a variety of ideas to get this working the way you'd like it to, only to be unsuccessful. I tried assigning the `board` object to a key in the Node `process`, as a Node `global`, and by registering it as an export within an `index.js` `process.mainModule` that is run when `index.html` is run (which I think you also tried?). Unfortunately, after each new page visit, all of these variables were no longer defined.

The one thing I haven't tried, that I'd recommend for you, is to run your app using a frontend framework  (such as ember, react, angular, choo, backbone, etc) with routing support, so that a new page is never technically fetched. Every page will be served within the framework itself, without ever leaving `index.html`.

Another recommendation, which may not be possible for you, is to try creating this app in [electron](http://electron.atom.io/) instead. Electron has a much more formal setup where you can run johnny-five within the background process, which persists state between pages.