rwaldron/johnny-five #1151: unable to read value of digital pin mode is OUTPUT.

overflowz, 2016-06-06:
Well, hi. As docs say, board.digitalRead can read value from any digital pin WHEN IT CHANGES.
While callback is not fired whenever pin mode is set to OUTPUT.

Could someone point me how can I read output mode pin states whenever it changes?

Regards.

rwaldron, 2016-06-06:
That's not possible with the underlying Firmata protocol. If you're writing to the output somewhere else, then it would be inefficient to respond to that state change by reading the actual output itself, it makes more sense to just signal in your program that the output was changed. If that's really out of the question, just use another pin: 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var output = new five.Pin(11);
  var input = new five.Sensor.Digital(7);
  var value = 1;

  setInterval(() => {
    output.write(value ^= 1);
  }, 1000);

  input.on("change", state => console.log(state));
});
```

[image]

closed 2016-06-06 by rwaldron

overflowz, 2016-06-06:
I'm new on this stuff, I apologize. So you mean if I add two wires at the same position (output and input) changing output value also triggers that input value too, and I can read the state, right?

rwaldron, 2016-06-06:
> So you mean if I add two wires at the same position (output and input) changing output value also triggers that input value too, and I can read the state, right?

Yep, that's exactly right :)

rwaldron, 2016-06-06:
No problem :)

overflowz, 2016-06-06:
Instead of opening another topic, I'll write it here anyways. Is it safe to do something like check all output pin states manually within 100ms interval?

overflowz, 2016-06-06:
Well, I'll explain my problem here if you don't mind.
I'm playing with home automation like project, while need real time states.
Let's say, there's a bulb and a switch. When switch is ON, bulb is ON too and vice-versa.
But this bulb also can be changed from the web directly so I would like to attach event directly to OUTPUT (bulb) pin rather than INPUT (switch) one. Is there any other possible ways? I'm kinda limited to additional pins :disappointed:

rwaldron, 2016-06-06:
Before we answer that question... if the bulb can be changed from a switch or from the web, what happens to the switch when the web changes the bulb?

overflowz, 2016-06-06:
switch stays at the same state, it's one-way trigger. I don't know how to explain but I'll try. It has only one state (clicking down) and then jumps back to top.

rwaldron, 2016-06-06:
Ah, I see. That's called a "momentary switch", which is actually falls in the "button" category :)

A few questions: 

- Is the momentary switch/button part of the J5 program? 
- Is the web based control served from a Node.js program that has the J5 code running?

To be more helpful, I need to know more about the hardware you're using and the big picture program you're trying to create. I have to sign off for the evening, but I will check back in the AM

overflowz, 2016-06-07:
Okay! Have nice evening tho! :-)

1) IT SHOULD. I thought a lot and now I know that it should be handled by J5.
2) Yes!

I'm using Raspberry PI, Arduino Due and NodeJS + J5 only.

Things I'm trying to accomplish:
1) I have 1 button and 1 bulb.
2) When user clicks button, I can't catch event of it, thus I can't know if bulb state changed or not.
3) If user triggers that bulb state (on or off) from the web, then J5 comes in play.

I think these steps should not have any problems. now I thought, can OUTPUT pins exist which is not relying on INPUT ones (except sensors of course)? or is it up to me? If there can be OUTPUT pins which won't rely on INPUT ones, how should I check if their state changes? Like a sensor?

I really apologize for poor explanation, my English skills aren't that perfect nor knowledge of this sphere.

Regards.

soundanalogous, 2016-06-07:
You can use the pin state query to get the state of the output pin. This is not event driven so you'd have to poll. I'm also not sure if this is exposed through J5. It's part of firmata.js which J5 uses to communicate with arduino-compatible boards. You'd do something like this:

```javascript
board.queryPinState(pin, function () {
  console.log("state of pin " + pin + " changed to: " + board.pins[pinToQuery].state);
});
```

overflowz, 2016-06-07:
Question is, is it reliable if I'll use poll like every 100ms on 60+ pins?

soundanalogous, 2016-06-07:
You'd have to stagger your polling.

overflowz, 2016-06-07:
That's not a reliable solution then I guess. Cause if I stage it, the last pin would be checked every 100 x 60 ms :disappointed:

soundanalogous, 2016-06-07:
Not 1 ever 100ms, could be one few ms. It depends on how much other work you application is doing and how much you can squeeze in when communication happens at 57600 baud. There is a proposal open however to enable reporting the state of all pins in one shot. It may be some time before that makes it's way into J5 however as it is just a proposal currently.

overflowz, 2016-06-07:
Uhm, and so much data transfers won't cause any troubles to either controllers? or isn't it a bad practice?