rwaldron/johnny-five #1155: L298N not working

illogica, 2016-06-14:
I built a small robot with two 5V DC motors. I want to drive them using a L298N and an Arduino Nano clone ([this one](https://www.amazon.it/gp/product/B015MGHH6Q/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)).
I uploaded the StandardFirmata program in the Nano, using Arduino IDE 1.6.9.
I communicate with the Nano using a JY-MCU bluetooth module (working, node is communicating correctly with the module).
I uploaded the following program:

```
var five = require("johnny-five"), board, motor1, motor2, led;

board = new five.Board();

board.on("ready", function() {
  // Create a new `motor` hardware instance.
  motor1 = new five.Motor({
    pins: {
      pwm:3,
	  dir:4,
	  cdir: 7
    }
  });
  
  motor2 = new five.Motor({
    pins: {
	  pwm:5,
	  dir:8,
	  cdir: 12
	}
  });
	
  motor1.start(124);
  motor2.start(124);

  board.repl.inject({
    motor1: motor1,
    motor2: motor2
  });
  
  // "start" events fire when the motor is started.
  motor1.on("start", function() {
    console.log("start motors", Date.now());

    // Demonstrate motor stop in 2 seconds
    board.wait(5000, function() {
      motor1.stop();
	  motor2.stop();
    });
  });

  // "stop" events fire when the motor is stopped.
  motor1.on("stop", function() {
    console.log(" motors stopped", Date.now());
	motor1.start(200);
  });
});
```
When I launch the program, Node is correctly communicating with the Nano and the console prints meaningful messages, but the PWM outputs don't output  thing and even the direction pins act somewhat randomly.
Out of desperation I uploaded this program into the Arduino:

```
//Code by Reichenstein7 (thejamerson.com), adapted for my application

//Keyboard Controls:
//
// 1 -Motor 1 Left
// 2 -Motor 1 Stop
// 3 -Motor 1 Right
//
// 4 -Motor 2 Left
// 5 -Motor 2 Stop
// 6 -Motor 2 Right

// Declare L298N Dual H-Bridge Motor Controller directly since there is not a library to load.

int SPEED = 255;

// Motor 1
int dir1PinA = 8;
int dir2PinA = 12;
int speedPinA = 5; // Needs to be a PWM pin to be able to control motor speed

// Motor 2
int dir1PinB = 7;
int dir2PinB = 4;
int speedPinB = 3; // Needs to be a PWM pin to be able to control motor speed

void setup() {  // Setup runs once per reset
// initialize serial communication @ 9600 baud:
Serial.begin(9600);

//Define L298N Dual H-Bridge Motor Controller Pins

pinMode(dir1PinA,OUTPUT);
pinMode(dir2PinA,OUTPUT);
pinMode(speedPinA,OUTPUT);
pinMode(dir1PinB,OUTPUT);
pinMode(dir2PinB,OUTPUT);
pinMode(speedPinB,OUTPUT);

}

void loop() {

// Initialize the Serial interface:

if (Serial.available() > 0) {
int inByte = Serial.read();
int speed; // Local variable

switch (inByte) {

//______________Motor 1______________

case '1': // Motor 1 Forward
analogWrite(speedPinA, SPEED);//Sets speed variable via PWM 
digitalWrite(dir1PinA, LOW);
digitalWrite(dir2PinA, HIGH);
Serial.println("Motor 1 Forward"); // Prints out “Motor 1 Forward” on the serial monitor
Serial.println("   "); // Creates a blank line printed on the serial monitor
break;

case '2': // Motor 1 Stop (Freespin)
analogWrite(speedPinA, 0);
digitalWrite(dir1PinA, LOW);
digitalWrite(dir2PinA, HIGH);
Serial.println("Motor 1 Stop");
Serial.println("   ");
break;

case '3': // Motor 1 Reverse
analogWrite(speedPinA, SPEED);
digitalWrite(dir1PinA, HIGH);
digitalWrite(dir2PinA, LOW);
Serial.println("Motor 1 Reverse");
Serial.println("   ");
break;

//______________Motor 2______________

case '4': // Motor 2 Forward
analogWrite(speedPinB, SPEED);
digitalWrite(dir1PinB, LOW);
digitalWrite(dir2PinB, HIGH);
Serial.println("Motor 2 Forward");
Serial.println("   ");
break;

case '5': // Motor 1 Stop (Freespin)
analogWrite(speedPinB, 0);
digitalWrite(dir1PinB, LOW);
digitalWrite(dir2PinB, HIGH);
Serial.println("Motor 2 Stop");
Serial.println("   ");
break;

case '6': // Motor 2 Reverse
analogWrite(speedPinB, SPEED);
digitalWrite(dir1PinB, HIGH);
digitalWrite(dir2PinB, LOW);
Serial.println("Motor 2 Reverse");
Serial.println("   ");
break;

default:
// turn all the connections off if an unmapped key is pressed:
for (int thisPin = 2; thisPin < 11; thisPin++) {
digitalWrite(thisPin, LOW);
}
}
}
}
```

This program works, I can command both motors via the Arduino serial monitor, so the hardware is correctly connected and properly working.

Am I looking at a FIrmata problem or Johnny-Five problem?

dtex, 2016-06-14:
Hi @illogica 

I don't know much about the Nano or Bluetooth modules, but I know a lot about the motor class and it looks to me like you've done everything right with the setup on that. The only difference between your J5 code and that working example is that the working example always sets the PWM value to 0 or 255. I would expect 124 to work as well, but if the motor power is on the threshold of being just barely enough then it might make a difference though the following statement makes me think that's not the culprit here:

> PWM outputs don't output thing and even the direction pins act somewhat randomly.

How are you measuring the pins to know what's happening (a voltage meter)? Sometimes I like to wire up LED's to the motor pins on my micro-controller so I can literally see what's happening on those pins.

illogica, 2016-06-14:
Hi @dtex , thanks for the prompt reponse.
I'm measuring the PWM output using my voltmeter and I see plain 0.0V. Also tried with an Led straight from the PWM output to ground, but it won't turn on. The value 124 is just to be safe, I tried the motors with a .ino program and they start rotating at about 50.
I don't have my robot with me now, but I think I'll try uploading the StandardFirmataPlus tonight, instead of the StandardFirmata, and see if anything changes.

dtex, 2016-06-14:
StandardFirmata should work, but StandardFirmataPlus is the new recommended install for J5. Did you configure the bluetooth module to work at 57,600 baud?

https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-JY-MCU-Bluetooth-Serial-Port-Module

illogica, 2016-06-14:
Yes, I went successfully through the whole AT stuff.
I'm going to try the StandardFirmataPlus now, let's see what happens...

illogica, 2016-06-14:
Another test, even more simple, just one of the two motors:
```
var five = require("johnny-five"), board, motor;

board = new five.Board();

board.on("ready", function() {
  // Create a new `motor` hardware instance.
  motor = new five.Motor({
    pins: {
      pwm:3,
	  dir:4,
	  cdir: 7
    }
  });
  
  board.repl.inject({
    motor: motor
  });
  
  // "start" events fire when the motor is started.
  motor.on("start", function() {
    console.log("motor started", Date.now());
  });

  // "stop" events fire when the motor is stopped.
  motor.on("stop", function() {
    console.log(" motor stopped", Date.now());
  });
});
```
So, I just type "motor.start()" at the command line, and all I get is digital outputs 4,7,8,12 on for less than a second, and then go off again. After this the board will still accept REPL commands, but wit no visible effect.
I also tried with another Nano board (I have 5 of them) and the result is the same.

illogica, 2016-06-14:
Replaced the Nano clone with an original Arduino Uno, I obtain the same non-results.

illogica, 2016-06-15:
I think I made some progress. Using the wired connection the J5 program works. Using the Bluetooth adapter (JY-MCU, HC-05) the program works but only for a fraction of a second. I might try with a different bluetooth module (which I don't have).

rwaldron, 2016-08-23:
> I built a small robot with two 5V DC motors. I want to drive them using a L298N and an Arduino > Nano clone (this one).
> I uploaded the StandardFirmata program in the Nano, using Arduino IDE 1.6.9.
> I [...]

- What is your power source on the Nano side?
- How are you powering the motors?

dtex, 2018-01-16:
Hi @illogica 

I'm going to close this since we haven't heard back. Please feel free to re-open if necessary.

closed 2018-01-16 by dtex