rwaldron/johnny-five #1156: Error when to run with webpack

fortil, 2016-06-15:
This error show to run app

`ERROR in ./~/johnny-five/lib/repl.js
Module not found: Error: Cannot resolve module 'repl' in c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\johnny-five\lib
resolve module repl in c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\johnny-five\lib
  looking for modules in c:\xampp\htdocs\ejemplosFRP\arduino\node_modules
    c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl doesn't exist (module as directory)
    resolve 'file' repl in c:\xampp\htdocs\ejemplosFRP\arduino\node_modules
      resolve file
        c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl doesn't exist
        c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.webpack.js doesn't exist
        c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.web.js doesn't exist
        c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.js doesn't exist
        c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.json doesn't exist
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl]
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl]
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.webpack.js]
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.web.js]
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.js]
[c:\xampp\htdocs\ejemplosFRP\arduino\node_modules\repl.json]
 @ ./~/johnny-five/lib/repl.js 1:11-26`

This is my webpack.config

```
import webpack from 'webpack';
import HtmlWebpackPlugin from 'html-webpack-plugin';

let name = 'arduf';

let vendorModules = /(node_modules|bower_components)/;

export default {
  target: "web",
  entry: {
    app: './app/index.js',
  },

output: {
    path: "./build",
    filename: "[name].js",
  },

module: {
    loaders: [
      {
        test: /\.js/,
        exclude: vendorModules,
        loader: "babel",
        query: {
          optional: [
            "runtime",
            "optimisation.react.constantElements",
          ],
          env: {
            development: {
              plugins: [
                "typecheck",
                "closure-elimination",
              ],
            },
          },
          stage: 0,
        },
      },
      { test: /\.css$/, loader: "style-loader!css-loader" },
      { test: /\.(ttf|eot|svg|woff(2)?)(\?v=[0-9]\.[0-9]\.[0-9]|\?[a-z0-9]+)?$/, loader: 'url-loader' },
      { test: /\.jpg$/, loader: "url-loader?mimetype=image/jpg" },
      { test: /\.bmp$/, loader: "url-loader?mimetype=image/bmp" },
      { test: /\.png$/, loader: "url-loader?mimetype=image/png" },
      { test: /\.scss$/, loaders: ["style", "css", "sass"] },
    ],
  },
externals: ["bindings"],

plugins: [
    new webpack.IgnorePlugin(/^mock-firmata$/),
    new webpack.ContextReplacementPlugin(/bindings$/, /^$/),
    new webpack.optimize.CommonsChunkPlugin('init.js'),
    new HtmlWebpackPlugin({
      title: name,
      minify: process.env.NODE_ENV === 'production' ? {
        removeComments: true,
        removeCommentsFromCDATA: true,
        collapseWhitespace: true,
        conservativeCollapse: false,
        collapseBooleanAttributes: true,
        removeAttributeQuotes: true,
        removeRedundantAttributes: true,
        preventAttributesEscaping: true,
        useShortDoctype: true,
        removeEmptyAttributes: true,
        removeScriptTypeAttributes: true,
        removeStyleLinkTypeAttributes: true,
      } : false,
      template: './app/index.ejs',
    }),
    new webpack.HotModuleReplacementPlugin(),
  ],
};
```

help me

monteslu, 2016-06-15:
@fortil 
try setting `{repl: false}` your five.Board() constructor call.

rwaldron, 2016-06-21:
@bcjordan can you advise @fortil? Thanks!

fortil, 2016-06-23:
for application front can't be used the plugin, only in node.js
I am ending this issue

closed 2016-06-23 by fortil

Bjvanminnen, 2016-06-29:
Note, I was able to work around the repl issue by creating a module that just exports an empty object, and then aliasing repl to that module, ie.

```
// noop.js
module.exports = {}
```
```
// in webpack config
alias: {
    repl: path.resolve(__dirname, 'src/noop'),
}