rwaldron/johnny-five #1157: Edison and arduino sd card slot

wicktus, 2016-06-18:
Hello,
I recently bought an intel edison, since I'm familiar with Javascript I decided to install Johnny-five on it :), works like a charm so far, having a blast with Johnny-five ^ ! But I cannot seem to find a way of writing to an sd card, I'm talking about an sd card inserted into the **arduino shield** ;).
I know that fs = require("fs") allows the use of the node fileSystem api, but I cannot find how to "detect" my sd card reader component. there's no var servo = new Five.servo(9) for the sd card reader that I know of.
Thanks for your help and sorry if i sound noobish I just begun using Johnny-five, GOOD work team ;)

dtex, 2016-06-21:
Which Arduino shield are you using? I suspect that you won't be able to do this without a customized version of firmata.

wicktus, 2016-06-21:
Hello,
I'm using this breakout board :
https://www.adafruit.com/products/2180

dtex, 2016-06-21:
So you're not using a separate data logging Arduino shield? You're just wanting to read the SD card on this breakout board correct?

wicktus, 2016-06-21:
Yes exactly :). But i don't know if it's possible, i don't even if it's possible using Node.js FS or jsonfile

rwaldron, 2016-06-21:
> there's no var servo = new Five.servo(9) for the sd card reader that I know of.

That's correct. SD card interaction is not in scope for Johnny-Five, but it should be easier than it presently is...

I was ~not~ surprised to learn that this process is painful on the Edison, so I spent some time this morning figuring out a reliably way to detect when an sdcard is inserted or removed on the Edison. What I came up with is available as a module called `sdactivity`: 

(On the Edison)

```
npm install sdactivity
```

Then use like: 

```js
var SDActivity = require("sdactivity");

var sd = new SDActivity({
  sdpath: "/media/sdcard"
});

sd.on("mount", function() {
  console.log("Mount");
});

sd.on("unmount", function() {
  console.log("Unmount");
});
```

wicktus, 2016-06-21:
Well thanks that will help ! Will look at datalogging tonight, I still can use traditional Javascript and Node FS/jsonfile for datalogging I hope.

closed 2016-06-21 by rwaldron