rwaldron/johnny-five #1158: Failure to fully turn off RGB Led strip on exit event

JvanderHeide, 2016-06-19:
**Board:** Arduino Uno
**Shield:** n/a
**Hardware you are having an issue with:** RGB Led strip
**Version of Johnny-Five:** 0.9.56
**What your expectations are:** 
When using the following code I expect the RGB Led strip to be off after the emission of the exit event.
```javascript
board.on('ready', function(){	 
  var ledRGB = new five.Led.RGB([ 9, 10, 11 ]);

  this.on('exit', function() {
    ledRGB.off();
  });
```

What the actual outcome is: Steps to reproduce (including code samples):

rwaldron, 2016-06-20:
Thanks for the report! I'll try to take a look at this today

JvanderHeide, 2016-06-29:
@rwaldron I was wondering if you've had the time to take a look at this issue. I understand if you weren't able to. If you have any pointers on resetting the board in any other way that'd be great.

Thank you very much for your work on Johnny-Five!

rwaldron, 2016-06-29:
@JvanderHeide sorry, I've been swamped working on this: https://learn.sparkfun.com/tutorials/experiment-guide-for-the-johnny-five-inventors-kit which just launched today. 

Can you link me to the actual component you're using, described above as "RGB led strip"? Thanks!!

JvanderHeide, 2016-06-29:
Sweet! Looks like a great kit. 

---

If I recall correctly it's [this one](http://www.ledstripkoning.nl/led-strips-rgb/losse-ledstrips-rgb/5-meter-150-leds-rgb-led-strip/). The page is in Dutch however. I'm powering it with a separate 12V transformator. Like this (although I have switched the R and B around (R=9,G=10,B=11)): 

[image]

It works like a charm, with the exception of only the red pin turning off when calling `.off()` at an exit event.

rwaldron, 2016-07-02:
TBH, I suspect the issue isn't a software related one, because all the rgb instance is doing in off() is zeroing the PWM signal. If the red channel is staying on it may be that some power source is causing that? I will have to locate the same or similar components and then try to reproduce

rwaldron, 2016-07-02:
Do you think this is suitable: https://www.amazon.com/uxcell®-Package-L7812CV-Positive-Regulators/dp/B00CMQ4O8Y ?

JvanderHeide, 2016-07-02:
@rwaldron there's no power from the Arduino going to the led strip, the MOSFETs are only controlling the ground. And Red is turning off (the first pin of the rgb configuration), hence my assumption that there's not enough time for the other two pins two be shut down in an exit event. 

.off() works perfectly fine from repl or in any code but the exit event.

JvanderHeide, 2016-07-02:
Just tried without any components, just measuring the pins using my multimeter, and pins 10 and 11 kept giving their voltage at the moment of exit. Only pin 9 got set to zero volts.

rwaldron, 2016-07-02:
Ok, that's very helpful. I believe you're exactly right w/r to "not enough time". Try this: 

```js
// Anywhere inside your board's ready event handler: 
this.repl.inject({
  // Make an accessor named "q"
  get q() {
    rgb.off();
    process.nextTick(() => process.exit());
  }
});
```

I didn't test this exact code, but I've used this same mechanism before. Now you can type `q` + `<enter>` to "clean up" and exit.

JvanderHeide, 2016-07-02:
Thank you!  This works and will do 👍 

Had to rename my ledRGB in the repl, so that it didn't match the global ledRGB (thus no `ledRGB: ledRGB`) and had to add a `setTimeout` of a few ms around the `process.nextTick(() => process.exit());` (timeouts didn't work with the exit event handler).

closed 2016-07-02 by JvanderHeide

rwaldron, 2016-07-05:
Nice! I'm glad we could resolve this :)