rwaldron/johnny-five #1161: GPS : this.io.SERIAL_PORT_IDs error

wicktus, 2016-06-22:
Hello, 
I keep getting this error on my intel edison (arduino shield, gps connected to pin ~6/~5) :

```
**ERROR**: /home/root/node_modules/johnny-five/lib/gps.js:269
  state.portId = opts.serialPort || opts.portId || opts.port || opts.bus || this.io.SERIAL_PORT_IDs.DEFAULT;
                                         ^

TypeError: Cannot read property 'DEFAULT' of undefined
    at GPS.initialize (/home/root/node_modules/johnny-five/lib/gps.js:269:100)
    at new GPS (/home/root/node_modules/johnny-five/lib/gps.js:256:10)
    at Board.<anonymous> (/home/root/.node_app_slot/main.js:35:14)
    at emitNone (events.js:72:20)
    at Board.emit (events.js:166:7)
    at nextTickCallbackWith0Args (node.js:420:9)
    at process._tickDomainCallback (node.js:390:13)
    at Function.Module.runMain (module.js:443:11)
    at startup (node.js:139:18)
    at node.js:968:3

```

Here's the culprit :

 ```
gps = new five.GPS({         
          pins: {rx: 5, tx: 6}
        });
```
GPS used : UBLOX NEO 7N.

I cannot seem to find how it happens, the serial port isn't required in the first place..

Thanks :)

wicktus, 2016-06-22:
After playing with gps.js a little, I think it's related  to **this.io**, i also have a **this.io.serialConfig() is not a function**

here is the log of this.io :
```
Galileo {
  domain: null,
  _events: 
   { close: [Function: bound ],
     disconnect: [Function: bound ],
     error: [Function: bound ],
     string: [Function: bound ] },
  _eventsCount: 4,
  _maxListeners: undefined,
  name: 'Intel Edison',
  isReady: true,
  aref: 5,
  vref: 5,
  pins: 
   [ Pin {
       domain: null,
       _events: {},
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 0,
       supportedModes: [],
       analogChannel: undefined,
       isMiniboard: false,
       isAnalog: false,
       gpio: 0,
       report: 0,
       value: 0 },
     Pin {
       domain: null,
       _events: {},
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 1,
       supportedModes: [],
       analogChannel: undefined,
       isMiniboard: false,
       isAnalog: false,
       gpio: 1,
       report: 0,
       value: 0 },
     Pin {
       domain: null,
       _events: {},
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 2,
       supportedModes: [Object],
       analogChannel: undefined,
       isMiniboard: false,
       isAnalog: false,
       gpio: 2,
       report: 0,
       value: 0 },
     Pin {
       domain: null,
       _events: {},
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 3,
       supportedModes: [Object],
       analogChannel: undefined,
       isMiniboard: false,
       isAnalog: false,
       gpio: 3,
       report: 0,
       value: 0 },
[... 185 lines cut ...]
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 'A4',
       supportedModes: [Object],
       analogChannel: 4,
       isMiniboard: false,
       isAnalog: true,
       gpio: 4,
       report: 0,
       value: 0 },
     Pin {
       domain: null,
       _events: {},
       _eventsCount: 0,
       _maxListeners: undefined,
       addr: 'A5',
       supportedModes: [Object],
       analogChannel: 5,
       isMiniboard: false,
       isAnalog: true,
       gpio: 5,
       report: 0,
       value: 0 } ],
  analogPins: [ 0, 1, 2, 3, 4, 5 ] }
```

dtex, 2016-06-22:
Unfortunately edison-io (aka galileo-io) does not support Serial yet. Serial was only recently added to firmata so it'll be a little while before the other platforms catch up. I'e created an issue over [here](https://github.com/rwaldron/galileo-io/issues/50)

rwaldron, 2016-06-22:
Yeah, It could definitely be implemented, it just hasn't been done yet :(

wicktus, 2016-06-23:
Thanks for the heads-up.

I think that one "workaround" for intel edison users would be to use the intel MRAA for node.js  serial/UART implementation, while waiting for this implementation in J5 in the long term ;). 

Make sure to upgrade mraa to 1.0.0.28 (use the dev branch on opkg) and set max rights for gpio exports 

the code would look like this : 

```
var mraa = require('mraa'); //require mraa
console.log('MRAA Version: ' + mraa.getVersion()); //print out the mraa version in IoT XDK console

var u = new mraa.Uart(0); // here 0 is a port on the groove shield...
var serialPath = u.getDevicePath(); //Default general purpose port "/dev/ttyMFD1" - Intel(R) Edison; 
```
This demo is tailored for the groove base shield but it would be simple to adapt it for non-groove users I think, will have to check

rwaldron, 2016-08-23:
Since @dtex opened https://github.com/rwaldron/galileo-io/issues/50 I'm going to close this—thanks everyone!

closed 2016-08-23 by rwaldron