rwaldron/johnny-five #1162: WEBSITE: add Genuino in supported platform list ?

kosamari, 2016-06-24:
[ not sure if this is a good place to ask about... `http://johnny-five.io/` if not let me know where to move this convo :) ]

This is more of editorial suggestion!
I'd like to suggest listing Genuino in `http://johnny-five.io/platform-support/` along side Arduino.

I was just talking to people wroking on J5 workshop in Japan & noticed they do not refer the board itself as Arduino since product name outside of USA is Genuino.
Given this dual name situation is going on for a while, I think it is entirely possible that there are people who do not know board called Arduino exists, especially people who is getting started now (and have not followed entire drama.)

Proposed Change: 
[image]

Just a suggestion :)

henricavalcante, 2016-06-24:
I think it's a good catch ;)

rwaldron, 2016-08-23:
Can you make a PR: https://github.com/johnny-five-io/johnny-five.io/blob/master/src/platforms-plugins.json

Thanks!

closed 2016-08-23 by rwaldron

kosamari, 2016-08-23:
awesome will do ! thanks @rwaldron