rwaldron/johnny-five #1163: One of the boards to communicate after a few minutes

rodrigoadachi, 2016-06-24:
Hello,
I have two mini Arduino pro USB-connected and communicating to JF5 one nicknamed "A" and the other "B" after a few minutes just one of them (sometimes the The other times B) to communicate without giving errors or reports in board.on ("close", function (), before it was only the board is now B, and always one or the other and it is functioning normally.

Below a piece of my code

```
var five = require('johnny-five');
var ports     = [
	{ id: "A", port: "/dev/ttyUSB0" },
	{ id: "B", port: "/dev/ttyUSB1" }
];
var board = new five.Boards(ports);
board.on("ready", function(boards) {
  var expanderX20 = new five.Board.Virtual( new five.Expander({ address: 0x20, controller: "PCF8574", board: this.byId("A") }) );
  var jfB01E20P00   = new five.Relay({ board: expanderX20, pin: 0 });
  var jfB01E20P01   = new five.Relay({ board: expanderX20, pin: 1 });
  var jfB01E20P02   = new five.Relay({ board: expanderX20, pin: 2 });
  var jfB01E20P03   = new five.Relay({ board: expanderX20, pin: 3 });
  var jfB01D08  = new five.Button({ isPullup: true, invert: true, holdtime: 1000, pin: 8, board: this.byId("A") });
  var jfB01D10  = new five.Button({ isPullup: false, invert: true, holdtime: 1000, pin: 10, board: this.byId("A")});
  var jfB01D11  = new five.Button({ isPullup: false, invert: true, holdtime: 1000, pin: 11, board: this.byId("A") });
  var jfB01D12  = new five.Button({ isPullup: false, invert: true, holdtime: 1000, pin: 12, board: this.byId("A") });
  var jfB01Led  = new five.Led.RGB({ pins: { red: 6, green: 5, blue: 9, board: this.byId("A") }});
  var jfB02A00  = new five.Sensor({ pin: "A0", freq: 250, threshold: 5, board: this.byId("B") });
  var jfB02D04  = new five.Button({ isPullup: false, invert: false, holdtime: 1000, pin: 4, board: this.byId("B") });
  var jfB02D05  = new five.Button({ isPullup: true, invert: false, holdtime: 500, pin: 5, board: this.byId("B") });
  var jfB02D06  = new five.Button({ isPullup: false, invert: false, holdtime: 500, pin: 6, board: this.byId("B") });
  var jfB02D03  = new five.Relay({ pin: 3, board: this.byId("B") });
  var jfB02D07  = new five.Relay({ pin: 7, board: this.byId("B") });

  jfB01E20P00.off();
  jfB01E20P01.off();
  jfB01E20P02.off();
  jfB01E20P03.off();
  jfB01Led.off();
  jfB02D03.off();
  jfB02D07.off();
});

board.on("close", function() {
	console.log('JohnnyFive board closed')
});

```

Excuse the translation, my default language is Portuguese in Brazil

rodrigoadachi, 2016-07-08:
News

if (board[0].isReady){
	console.log('JF board A : ',board[0].isReady);
}

if (board[1].isReady){
	console.log('JF board B : ',board[1].isReady);
}

I put a setInterval monitoring these guys there and when the board to operate they do not realize

rwaldron, 2016-07-08:
Thanks for the report. I will try to assemble something to reproduce this over the weekend.

closed 2018-08-28 by stale[bot]