rwaldron/johnny-five #1165: Johnny FIve and Arduino Uno WIFI

HernanHM, 2016-06-26:
I tried to use J5 and the arduino uno wifi. 

is this board not yet supported?

I followed the steps in this blog (https://wifinodebot.blogspot.com/2016/02/blink-led-over-wifi-with-nodejs-johnny.html?showComment=1466974432844#c5598866979654832275) trying to adapt to this board, but it does not work.

I get error below (I changed the firmata baud rate to 115200)

1466974289262 Device(s) Firmata 
1466974294272 Connected Firmata 
1466974304278 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again. 
events.js:146
throw err;
^

Error: Uncaught, unspecified "error" event. ([object Object])
at Board.emit (events.js:144:17)
at Board.log (/Users/hmillon/Documents/Arduino/JonnyFive_Push/node_modules/johnny-five/lib/board.js:642:8)
at Board.(anonymous function) [as error] (/Users/hmillon/Documents/Arduino/JonnyFive_Push/node_modules/johnny-five/lib/board.js:653:14)
at Board. (/Users/hmillon/Documents/Arduino/JonnyFive_Push/node_modules/johnny-five/lib/board.js:421:14)
at Timer.listOnTimeout (timers.js:92:15)

henricavalcante, 2016-06-27:
Have you tried this before?

https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0

monteslu, 2016-06-27:
It looks like you're getting the network connection but not getting a firmata response.  The blog or that gist should work if everything is connected correctly and the IPs and baud rates are correct.

boneskull, 2016-06-27:
@HernanHM can you assert you are connected to the device's AP?

HernanHM, 2016-06-28:
@henricavalcante, yes actually I'm using it as reference.
@monteslu Yes I changed the baud rates to 115200 in the firmata and uploaded it into the board. IP is correct as well, I can say that becasue I'm able to deploy the sketchs to the board over WIFI.

@boneskull  I'm connected for sure.

closed 2018-08-28 by stale[bot]