rwaldron/johnny-five #1166: New Library Created "Flow"

alfredoborrero, 2016-06-27:
Hi, I needed to add flow meters to my Home Automation, so I decided to create my own library and added to Johnny Five. I know is not elegant, thats why I would like some of you guys to review it and maybe you can add more stuff to it. For now is working fine and its giving me the water usage in real time on my website.

`var flowmeters  = new five.Flow({ pin: 2, board: this.byId("F")}); // Flow Meter Model YF-S201
[image]
`

and then ...

`io.on('connection', function (socket) {`
`    flowmeters.on("open",  function(sensor, data) { `
`         water_usage++;`
`         total_liters = water_usage/(7.5*60);`
`         console.log("Water Usage: " + total_liters.toFixed(2));`
`         socket.emit ('water_changed', total_liters.toFixed(2));`
`    });`
`});`

All I did was save a copy of the existing library called Switch.js and modify some lines. I also mofified the file johnny-five.js and added 

`Flow: require("./flow"), // Added by Alfredo Borrero`
`module.exports.Flows = module.exports.Flow.Collection; // Added by Alfredo Borrero`

However I now it need some modifications like adding Flow Meter model, there are few of them and every model has their own formula to calculate the flow. I also would like to add outputs via emit() like:
- "Pulses"
- "Total Liters"
- "Liters per minute"
- "Gallons per minute"
- "Liters per hour"
- "Gallons per hour"

I know Johnny Five was originally intended for Robots, perhaps someone wants do design a robot to fill up a water tank? or a gas tank? ... Anyways I have been using Johnny Five for a while for my Home Automation and its working very very well, congrats !!!. Every day I add new devices to my house and Johnny Five is handling it very well. I hope some of you guys can add this library to Johnny Five.

Regards,

Alfredo Borrero

[flow.zip](https://github.com/rwaldron/johnny-five/files/334210/flow.zip)

reconbot, 2016-07-14:
Johnny-Five is good to work with any electronics devices. This is a nice little plugin.

reconbot, 2016-07-14:
Btw we have an often overlooked plugin system you can use if you want to share the library. You'd want to change your code to work like [the example](http://johnny-five.io/examples/plugin/).

alfredoborrero, 2016-07-19:
Thanks Francis, I will try to use the plug in. And yes, I would like to share the library with the community, however I don't know how to do it. All I want to do is incorporate this new Flow.js module to the Johnny Five framework, I believe a lot of people will get benefits on this. Any suggestion?

reconbot, 2016-07-19:
If there are a large number of flow valves out there and a lot of people using it it might make it into the core project. 

The rule of thumb is if it can use use a generic device driver then j5 doesn't need to provide anything special for it. This seems to be a composite device with inputs and outputs. If there are several kinds of flow devices with the same interface (even if they have different internal mechanisms) then it might make sense to have a generic flow component with drivers for each device.

Plugins get the benefit of sharing the code quickly and having an independent development cycle. I would start by [making your own github repo](https://help.github.com/articles/create-a-repo/) for the code and then [publish the project on npm](https://docs.npmjs.com/getting-started/creating-node-modules). From there you can get issues and pull requests to improve it. And if it gets used enough it will have a much stronger chance of getting into core.

We don't have a ton of plugins yet, so you'll stand out. I know putting `johnny-five` as a keyword in the package.json helps people find it. As far as I know we don't have a formal keyword or project naming convention for plugins.

rwaldron, 2016-08-23:
Thanks for that guidance @reconbot

closed 2016-08-23 by rwaldron

sapiubelli, 2018-10-23:
Hello, very interesting the project. How could I implement liters per minute ?, I need to measure the flow

saksholm, 2019-09-13:
Any progress? Does anyone release this as a plugin? I definitely need couple of Flows in my HA project. Any response is welcome, have to try this.