rwaldron/johnny-five #1175: PCF8591 expander only emits value for first sensor specified for one of its pins

bblumberg, 2016-07-11:
In playing around with using PCF8591 as an expander, I noticed that in expander.js, line 888 should be changed (I think) from:

if (this.pins[i].report) {
  this.emit("analog-read-" + pin, value);
}
to:
if (this.pins[i].report) {
  this.emit("analog-read-" + i, value);
}

As the code stands, only the first sensor that is associated with a pin on the expander will get notified because the code that sets up the emitters is only called once, and 'pin' will equal whatever the pin is for that call.

rwaldron, 2016-07-11:
Nice catch! Do you want to write a patch for this? 

I'm looking at the tests right now and trying to figure out how this slipped through. I think it's possible that the tests are too "clever"

rwaldron, 2016-07-11:
I see exactly what happened now. In lib/expander.js, I changed this: 

```js
    var callback = this.i2cRead.lastCall.args[2];

    callback([0x00, 0x0f, 0xf0, 0xff]);

    test.equal(spy.callCount, 4);
```

To: 

```js
    var callback = this.i2cRead.lastCall.args[2];
    var emitter = this.sandbox.spy(this.expander, "emit");

    callback([0x00, 0x0f, 0xf0, 0xff]);

    test.equal(spy.callCount, 4);

    console.log(emitter.args);
```

And that outputs: 

```
[ [ 'analog-read-0', 0 ],
  [ 'analog-read-0', 60 ],
  [ 'analog-read-0', 960 ],
  [ 'analog-read-0', 1020 ] ]
```

Which makes sense now: I was testing that the right values made it through, but not that they were emitted to the right objects for the right pin. I'm confident that I only tested `A0` for this expander breakout :(

Let me know what you'd like to do. If you'd like to write the patch and tests, I will give you as much support as needed. If you'd like me to write it, I can do that as well.

bblumberg, 2016-07-11:
Probably best that you do it, you'll do it in a fraction of the time and you'll do a better job of it. 

I am very impressed with Johnny-Five. Nice job!!!

bb

closed 2016-07-11 by rwaldron