rwaldron/johnny-five #1176: Unable to switch an AC Contactor via Relay

criticalboot, 2016-07-12:
Here is the following scenario

**Setup:**
We are using Atmega644 PA-AU and trying to drive AC Contactors via relays. The relays are based on Johnny-five Relay Class through StandardFirmataEthernetPlus.

ATMega644 (StandardFirmataEhternetPlus + UIPEthernet) ==> ENC28J60 ==> network ==> PC ( J5 + NodeJs + Electorn)

**Problem:** 
Please refer the attached schematic as the basis of our setup
[image]

When we are driving just the relay without AC Loads mentioned at the end of the schematic it works fine with out any issues. As soon as we connect this relay to drive an AC contactor the behavior is erratic meaning the relay works for a few times and stops working with the corresponding leds (led1 and led2 ) hanging up. This also is leading to a MCU freeze.

**Observations:**
H/w and S/w disconnect: We are sending an on()/off() via console of Electron to the Relays. When AC is **NOT** connected the Relays status and the corresponding leds status is in sync. Now with AC Contactor connected to the Relays, the whole setup goes out of sync randomly after any number of ons/offs. Any action via browser does not affect the MCUs status. This is leading us to assume that the MCU is freezing.     

As long as the AC load is not connected the system works flawlessly forever. When ever the AC load is connected malfunction occurs. Not sure why. 
 
**Questions:**
1. Is there any thing abnormal in the schematic or connectivity that we have used?
2. Do you think the AC load is causing the MCU freeze up?
3. How do we know/test if J5 is alive?
4. Are we missing some thing here?

NOTE:
The same setup is working perfectly with Arduino UNO+Ehternet shield.

nebrius, 2016-07-12:
Are the relays rated for the load your attaching to them?

rwaldron, 2016-07-13:
Thanks for the report!

>  Any action via browser does not affect the MCUs status. This is leading us to assume that the MCU is freezing.

Browser? I think there is a whole layer I'm missing here :\

Also, looking forward to your answer to @nebrius's question, as it will be helpful to determine if this is software issue or not. 

(Also, thanks for jumping in @nebrius ❤️ )

criticalboot, 2016-07-16:
Thank you for the response. 

@nebrius - Yes, relays are rated at 10A and we are using AC load of less than 1A.

@rwaldron - 

> Browser? I think there is a whole layer I'm missing here :\

We are using Electron browser as the front end. This based on Chrome & NodeJS. J5 is added to NodeJS via NPM. 
J5 objects are being called from the console of the Electron browser. Eg: Motor object can be called via the console and 'external_contactor' can be toggled to switch on/off the motor/device. 

**Setup 1:**
Please refer the image attached to the previous message and the observation still holds good.

MCU = AtMega644PA-AU
Ethernet Module = ENC28J60

**Main issue :** the MCU freezes(?) and LED 1 stops blinking and stays in the same state where it was (on / off) at the time of freeze. When this happens there are no error messages thrown out anywhere giving us no clue as to what is happening.

**Setup 2:**
MCU = AtMega328P-AU
Ethernet Module = Wiz Mini with chip Wiz 5100

The schematic that was attached earlier still holds good except that the MCU and Ethernet module were switched to 328 & Wiz5100.

**Main Issue :** The LED1 stops blinking and remains in the state of on/off, J5 throws two types of errors
1. 'the other party has closed the connection'
2. 'ECONRESET'

The above errors are occuring **ONLY WHEN** the AC load is switched on. But otherwise, this works fine - meaning the relay switches on/off forever without the AC load (in this case the external contactor that we connect to the relay. In another case we connected an incandscent bulb). 

**Setup 3:**
Same as Setup 2 minus the Ethernet module; instead used serial connectivity (TTL to USB). One of the lines in the code attached shows that (commented).

**Main Issue :** When the AC load is switched on LED 1 stops blinking and stays in the same state where it was (on / off) at the time of freeze.

**Setup 4:**
This is the barebones setup without J5. We simply burnt the Arduino Blink sketch on the 328 and tried the same thing. This works absolutely fine without any problem forever with the AC load switched on.

**Main Issue :** None

Not sure what is missing here. Could there be a possible hardware issue?. Or do we suspect the software?. Please help us understand if we are missing something here.

I have a suggestion that we can have a quick Skype call to show you the setup and discuss. Please let us know if that works for you. 

Please find the code below as a sample for you to get a feel of what we are trying to do.
```

'use strict'
let five = require('johnny-five')
let EtherPort = require('etherport')
let Emitter = require('events').EventEmitter
let util = require('util')
let debug = require('debug')('motor')

let Motor = function (opts) {
  Emitter.call(this)
  // debug(opts)
  this.port = opts.port
  this.id = opts.id
  this.name = opts.name
  this.type = opts.type
  this.class = 'motor'
  this.pin_MOT = 7 // PB3 - external contactor on
  this.isEnabled = false

  this.enableAllPins = () => {
    this.external_contactor = new five.Relay({pin: this.pin_MOT, type: 'NO'})
  }
}

Motor.prototype.create = function () {
  let self = this

  let board = new five.Board({
    // port: '/dev/ttyUSB0'
    port: new EtherPort(this.port),
    repl: false,
    debug: false
  })
  board.on('ready', () => {
    // emit board connected signal.
    self.emit('connected', {})

    // enable all Pins
    self.enableAllPins()

    self.isEnabled = true
  })
}

Motor.prototype.onClick = function () {
  if (this.isEnabled) {
    let self = this
    setInterval(function (self) {
      self.external_contactor.toggle()
    }, 1000, self)
  }
}

util.inherits(Motor, Emitter)
module.exports = Motor
```

criticalboot, 2016-07-16:
We again tried the same setup with different AC power supply points. We thought this could be due to some extraneous current leakages in to the circuit. Even this did not work.

nebrius, 2016-07-18:
Can you try connecting a different, smaller AC load to the relay? This _really_ sounds like a hardware problem to me, given that it works without a load attached.

To isolate the issue a little more, can you write a short dummy app in C using the Arduino IDE that, say, closes the relay for 5 second and then opens it again, just to see if it works that way? If it does, then we'll know for certain if it's a hardware or software issue.

criticalboot, 2016-07-20:
Thank you @nebrius for your post and observations.
Please find below **Setup 4** already posted.
> **Setup 4:**
> This is the barebones setup without J5. We simply burnt the Arduino Blink sketch on the 328 and tried the same thing. This works absolutely fine without any problem forever with the AC load switched on.
> 
> **Main Issue :** None

As per **Step 4** this seems to be working fine even with AC Loads. Whenever J5 is used to run the setup we are running into this problem. We are also a bit puzzled about this and not able to figure it out if it is a S/W or H/W problem.

**Here is another setup we tried**
**Setup :** This is exactly same as the previous **Setup 1** (as per the old schematic), the only addition being a relay at the end as shown in the schematic attached below.
 This seems to be working absolutely fine even with any amount of AC Load either driving a 200W incandescent bulb or a contactor.
Not sure how to interpret this outcome.

[image]

**Questions / Issues**

1. The above setup seems to be working with J5. Do you think this is H/W or S/W issue?
2. As per Setup 4, where a simple blink sketch is able to drive the AC loads with out any hitches. Where as the same setup with J5 seems to be failing.

**NOTE**
We have used the standard rated relays throughout our setups. Relay ratings - coil voltage = 5V DC, Load rating = 10A, 230V AC.

Looking forward to hearing your thoughts.

nebrius, 2016-07-20:
Can you try porting the blink sketch to J5 with the same timing and seeing if it's also reliable? It may be possible there is a timing factor at play, either through how often it's being switched, timing relative to startup, or latency in communicating between Node and firmata over serial.

My instinct still says this is hardware related though. I don't know what's going on though, maybe it something with the BJT combined with the initial load surge when the relay is closed causes it to get overloaded or something (I'm totally guessing). If the new dual relay circuit works, can you use that instead?

criticalboot, 2016-08-12:
We have discovered that this was an EMF problem with the relays getting effected switching on any AC load. 
This EMF was causing the micro controller to freeze. So it is neither a J5 problem nor a micro controller problem. We have come across lot of people facing the same problems on various blogs and forums. We have been able to overcome this problem using a Solid State Relay.

But surprisingly this problem did not occur with an Arduino Uno board.

Appreciate your help.

closed 2016-08-12 by criticalboot