rwaldron/johnny-five #1177: Android StandardFirmataWifi with firmata in client mode?

dakoner, 2016-07-14:
Hi,
I would like to know if it is possible to use Johnny Five with an Android (ESP8266) running StandardFirmataWifi where the firmata is in client mode.  I already have server mode (Where the arduino has an IP, runs a firmata that accepts incoming network requests) working with Johnny Five acting as the client.

I would like to invert this: StandardFirmataWifi can act as a client (making an outgoing connection to a Firmata server).  Can Johnny Five act as the server in this case?  I didn't see any documentation on how to achieve this.

soundanalogous, 2016-07-14:
It's not documented, but you need to use the [EtherPort module](https://github.com/rwaldron/etherport) for J5 to act as the server. Use EtherPortClient when J5 is the client. To configure J5 as the server, in wifiConfig.h in StandardFirmataWifi, set the [SERVER_IP address](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h#L111-L114) to the address of the machine running the J5 server.

enso-media, 2017-03-04:
I have an additional question on this topic:
How can i achieve network connections on multiple boards? As far as i know, its not possible to do this in an performance-effective way, as i have to open a separate port for each board?
Are there plans to make firmata and j5 work more like MIDI, using channels or ids to speak to specific boards through a single port?

mwittig, 2017-03-04:
@enso-media Each J5 board has its own configuration and as part of this you can hook a new instance of EtherPort. However, each instance of EtherPort needs to have a different port assigned which is passed to the EtherPort constructor. This way, each board has its own transport endpoint the remote device can connect to. If you want to have remote devices hooked to different boards each remote device (wifiConfig.h as part of StandardFirmataWifi) must be configured to connect to  the transport endpoint of the corresponding board, i.e. you need need to set SERVER_PORT.

Another possible scenario is to have a J5 board connecting to a remote device. For this purpose you can [EtherPortClient](https://www.npmjs.com/package/etherport-client). On the remote device you need to setup the server mode then. This is done commenting out SERVER_IP macro definition as part of wifiConfig.h

enso-media, 2017-03-06:
Yep, i know all of this. I wanted to create a home-automation-like environment, where a single server controls multiple endpoints/boards in a dynamic way.
Sadly J5 and firmate doesnt work this way, as it seems to come from a more "traditional hard-wired" world, where each board needs to be configured and identified by its own connection.
For now i switched over to ESPeasy and the MQTT protocol, where exactly this is possible: I can detect, register and control each board using IDs with a single server.

closed 2018-08-28 by stale[bot]