rwaldron/johnny-five #1178: Arduino code to javascript(johnny-five)

lucked, 2016-07-14:
is possible convert arduino code for johnny-five?
thanks

reconbot, 2016-07-14:
It's often done by hand. It's usually easier to replace pin logic with johnny-five sensors and devices. But you'll have the [board functions](http://johnny-five.io/api/board/#api) (eg `pinMode`, `digitalWrite`, `digitalRead`, etc) available if you want to use those instead.

lyzadanger, 2016-08-23:
Hi, @lucked! There's no automated way to port Arduino Programming Language code to JavaScript/Johnny-Five. Doing it by hand gives you a chance to decide if you'd like to use some of the higher-level (and convenient) Johnny-Five component classes like [Button](http://johnny-five.io/api/button/), [Accelerometer](http://johnny-five.io/api/accelerometer/), [Servo](http://johnny-five.io/api/servo/) (and many, many others) or use some of the lower-level board functions @reconbot mentioned.

Good luck!

closed 2016-08-23 by lyzadanger